DomiStyle/esphome-panasonic-ac #1: Support for Panasonic CZ-TACG1 wifi adaptor

stibbzy, 2020-11-11:
Hi there. Is there any chance that the Panasonic CZ-TACG1 wifi adaptor could be supported with this? It'd be great to be able to get it out of the cloud!

DomiStyle, 2020-11-11:
@stibbzy Unfortunately I don't have a CZ-TACG1 to test with.

From my tests I couldn't get the AC to respond on the CN-CNT connector but that could also very well be because it gets disabled if CN-WLAN is used.

If someone can take a look if the protocol is the same between both modules I'm sure it could work.

stibbzy, 2020-12-25:
Hi me again. I found someone who has done some work decoding the protocol from the CN-CNT connector. You can see their work here: https://github.com/hotswapster/panasonic-aircon-wifi/blob/master/decoding.md

Does this look the same as the data coming directly from the CN-WLAN? Cheers!

DomiStyle, 2020-12-25:
@stibbzy The protocol that's used in that project looks different, you can find all the protocol decoding and raw captures I have done here:

https://github.com/DomiStyle/esphome-panasonic-ac/tree/master/protocol

DomiStyle, 2021-06-19:
@stibbzy Did you ever open up your unit and check if your device has the CN-WLAN connector?

There should be nothing stopping you from disconnecting the wifi adapter on CN-CNT and connecting an ESP to the CN-WLAN connector if there is one on the mainboard, you will have to lay some cables to the outside of course.

It should be on the smaller sub board for the indoor unit:
[image]

stibbzy, 2021-06-20:
Hi @DomiStyle No, I haven't had a chance to open it, but willing to try. Does it involve taking apart the AC, or can I see it when just opening the cover? I'll try to have a look this arvo!

DomiStyle, 2021-06-20:
Which model do you have exactly? I have the service manual for the these indoor units (European/Turkish version):

```
CS-Z20VKEW
CS-Z25VKEW
CS-Z35VKEW
CS-Z42VKEW
CS-Z50VKEW
CS-Z71VKEW
CS-XZ20VKEW
CS-XZ25VKEW
CS-XZ35VKEW
CS-XZ50VKEW
```

If you have a different one you might check in the service manual if that connector is there before disassembling your unit.

There's a disassembly video here: https://www.youtube.com/watch?v=TfQ_1lMjxu8

The sub board hopefully should be inside of the same metal cage on top like the main board is, it's connected via 3 connectors (CN-DATA1, CN-DATA2, CN-DATA3). I do think only the more modern units have them though.

[image]

The main PCB should look like this:

[image]

stibbzy, 2021-06-20:
@DomiStyle Looks like I am out of luck. My model is a CS/CU-Z35VKR, and I can see the wiring diagram on the chassis when I open the dust cover. It looks almost identical to the wiring and electrical diagrams I've attached here. No CNT-WLAN connector unfortunately :(

[image]
[image]

DomiStyle, 2021-06-21:
Well that's unfortunate but it was worth a try.

I don't have time to take on a new project right now but I will see if I can get my hands on a CZ-TACG1 soon. The protocol looks a lot simpler than the one DNSK-P11 uses, so it might not be that hard to integrate.

stibbzy, 2021-06-23:
Thanks anyway @DomiStyle Your efforts are appreciated! If there is anything else I can do, just let me know :)

AndreKuester, 2021-07-15:
I do have a CZ-TACG1 connected to my CS-Z25UFEAW. Does this esphome solution replace the CZ-TACG1, meaning the ESP board will be connected to the same port the CZ-TACG1 is currently using?

DomiStyle, 2021-07-15:
@AndreKuester This project does not support the protocol the CZ-TACG1 uses (yet). Only the DNSK-P11 is supported for now.

However, I ordered a CZ-TACG1 today so maybe I can get it working as well.

DomiStyle, 2021-07-19:
I got my CZ-TACG1 today and did some digging and I can confirm that it matches the protocol info posted [here](https://github.com/hotswapster/panasonic-aircon-wifi/blob/master/decoding.md). Assuming nothing goes wrong it shouldn't be that hard to implement.

I can't get my timer LED to stop blinking but that might just be because it's looking for the other wifi adapter. Commands are working fine.

It also seems like Comfort Cloud hasn't gotten any faster over the last year.

AndreKuester, 2021-07-19:
That’s great to hear. This means your solution could work for me as well. Just one question - does this work in combination with the CZ-TACG1 by soldering it to some point on the AC board or would I unplug the CZ-TACG1 and plug-in the ESP board to the existing connectors?

DomiStyle, 2021-07-19:
@AndreKuester You can just unplug your CZ-TACG1 and plug in an ESP with logic converter. If you do not want to break any of the original connectors there's a list of connectors [here](https://github.com/Egyras/HeishaMon#where-to-get-connectors).

stibbzy, 2021-07-20:
@AndreKuester @DomiStyle This is fantastic news! So now, I should be good to go with following the instructions and possibly obtaining a different connector. Thanks so much for the effort here!

DomiStyle, 2021-07-25:
Initial support for the CZ-TACG1 is available in this dev branch: https://github.com/DomiStyle/esphome-panasonic-ac/tree/dev-cztacg1

No instructions available yet, pinout for the CN-CNT connector (top to bottom):

1: +5V
2: TX
3: RX
4: +12V (do not connect)
5: GND

Needs a logic converter just like the other wifi unit (3.3V to 5V).

Working:
* Full status reporting
* Temperature, Mode, Fan mode, Swing mode control

Not working yet:
* Preset (Powerful/Quiet)
* Manual swing modes
* nanoeX toggle

brywithawhy, 2021-07-26:
Thanks @DomiStyle
I've just tested this and confirmed working. I LOVE how instant this now is.
NOTE: Saying the pinout from top to bottom could fry someone's board, because my connector on my unit is upside down. Maybe better to say from lock tab facing up, left to right. Which will be 5v, RX, TX, 12v(nc), GND

Powerful mode seems to work for me (shows up as boost).
My unit doesn't have horizontal swing or nanoex so perhaps detecting this and excluding it from the component would be handy. 

Not directly related to this however for my unit (CS-RZ25WKRW), the Eco preset seems to control the Quiet mode. This is the same when using the comfort cloud integration from HACS. Setting it doesn't make the eco light come on, but the compressor winds down and the fan speed seems to mimic quiet mode. Using Eco in the Panasonic app turns the eco light on but not quiet mode.

I'm getting the following errors in the logs however it doesn't seem to cause any problems.

`[10:00:39][W][esppac:438]: Received unknown vertical swing mode
[10:00:39][W][esppac:484]: Received unknown mild dry value`

Will continue testing while working from home today and post my findings

DomiStyle, 2021-07-26:
> NOTE: Saying the pinout from top to bottom could fry someone's board, because my connector on my unit is upside down. Maybe better to say from lock tab facing up, left to right. Which will be 5v, RX, TX, 12v(nc), GND

Good to know, will add a picture in the readme to make it more clear.

> Powerful mode seems to work for me (shows up as boost).
> Not directly related to this however for my unit (CS-RZ25WKRW), the Eco preset seems to control the Quiet mode. However, the nanoex toggle (once implemented) will probably toggle your eco mode.

I currently mapped powerful to boost and quiet to eco. I didn't know there's a separate eco mode on older units.

> My unit doesn't have horizontal swing or nanoex so perhaps detecting this and excluding it from the component would be handy.

Depending on what the AC sends I might be able to detect this.

Can you enable verbose logging and post the RX/TX debug lines when you see warnings like unknown vertical swing mode?
Also, can you enable eco mode via IR/cloud and once again post the RX/TX debug lines when it's enabled and when it's disabled?

brywithawhy, 2021-07-26:
>  I didn't know there's a separate eco mode on older units.
This unit was installed new about 3 months ago

This snippet is 2 polls with eco Enabled. Interestingly, the "Received unknown dry/mild state" warning disappeared after doing so.

`
[17:11:52][V][esppac:268]: Polling AC

[17:11:52][V][esppac:259]: TX: 70 0A 00 00 00 00 00 00 00 00 00 00 86
[17:11:52][V][esppac:259]: RX: 70 20 44 2A 80 A0 30 00 00 40 40 00 4C 2C 00 00 00 00 16 10 FF 17 10 FF 80 80 FF 80 C2 01 0B C1 38 14 15
[17:11:52][W][esppac:438]: Received unknown vertical swing mode
[17:11:52][D][climate:369]: 'lounge_ac' - Sending state:
[17:11:52][D][climate:372]:   Mode: HEAT
[17:11:52][D][climate:374]:   Action: HEATING
[17:11:52][D][climate:377]:   Fan Mode: AUTO
[17:11:52][D][climate:383]:   Preset: NONE
[17:11:52][D][climate:389]:   Swing Mode: OFF
[17:11:52][D][climate:392]:   Current Temperature: 22.00°C
[17:11:52][D][climate:398]:   Target Temperature: 21.00°C
[17:11:57][V][esppac:268]: Polling AC
[17:11:57][V][esppac:259]: TX: 70 0A 00 00 00 00 00 00 00 00 00 00 86
[17:11:57][V][esppac:259]: RX: 70 20 44 2A 80 A0 30 00 00 40 40 00 4C 2C 00 00 00 00 16 10 FF 17 10 FF 80 80 FF 80 EC 01 0C 80 85 71 81
[17:11:57][W][esppac:438]: Received unknown vertical swing mode
[17:11:57][D][climate:369]: 'lounge_ac' - Sending state:
[17:11:57][D][climate:372]:   Mode: HEAT
[17:11:57][D][climate:374]:   Action: HEATING
[17:11:57][D][climate:377]:   Fan Mode: AUTO
[17:11:57][D][climate:383]:   Preset: NONE
[17:11:57][D][climate:389]:   Swing Mode: OFF
[17:11:57][D][climate:392]:   Current Temperature: 22.00°C
[17:11:57][D][climate:398]:   Target Temperature: 21.00°C
`

DomiStyle, 2021-07-26:
> This unit was installed new about 3 months ago

And yours did not come with the newer wifi unit? Strange.

Can you do one more with eco off? The eco setting might be stored in the dry/mild state byte.

brywithawhy, 2021-07-26:
> And yours did not come with the newer wifi unit? Strange.

Don't know if it is country specific, but the CZ-TACG1 is the only wifi unit I can seem to find in Australia for panasonic units 

Eco off:

`
[17:19:33][V][esppac:268]: Polling AC

[17:19:33][V][esppac:259]: TX: 70 0A 00 00 00 00 00 00 00 00 00 00 86
[17:19:33][V][esppac:259]: RX: 70 20 04 2E 80 40 F0 00 00 40 00 00 00 2E 00 00 00 00 17 10 FF 17 10 FF 80 80 FF 80 33 00 01 C0 00 00 61
[17:19:33][W][esppac:438]: Received unknown vertical swing mode
[17:19:33][D][climate:369]: 'lounge_ac' - Sending state:
[17:19:33][D][climate:372]:   Mode: HEAT_COOL
[17:19:33][D][climate:374]:   Action: COOLING
[17:19:33][D][climate:377]:   Fan Mode: MEDIUM
[17:19:33][D][climate:383]:   Preset: NONE
[17:19:33][D][climate:389]:   Swing Mode: VERTICAL
[17:19:33][D][climate:392]:   Current Temperature: 23.00°C
[17:19:33][D][climate:398]:   Target Temperature: 23.00°C
[17:19:38][V][esppac:268]: Polling AC
[17:19:38][V][esppac:259]: TX: 70 0A 00 00 00 00 00 00 00 00 00 00 86
[17:19:38][V][esppac:259]: RX: 70 20 04 2E 80 40 F0 00 00 40 00 00 00 2E 00 00 00 00 17 10 FF 17 10 FF 80 80 FF 80 33 00 01 C1 38 14 14
[17:19:38][W][esppac:438]: Received unknown vertical swing mode
[17:19:38][D][climate:369]: 'lounge_ac' - Sending state:
[17:19:38][D][climate:372]:   Mode: HEAT_COOL
[17:19:38][D][climate:374]:   Action: COOLING
[17:19:38][D][climate:377]:   Fan Mode: MEDIUM
[17:19:38][D][climate:383]:   Preset: NONE
[17:19:38][D][climate:389]:   Swing Mode: VERTICAL
[17:19:38][D][climate:392]:   Current Temperature: 23.00°C
[17:19:38][D][climate:398]:   Target Temperature: 23.00°C
`

The dry mild state error isn't coming back. I'm wondering if it is a residual incorrect setting from the previous integration from HACS

DomiStyle, 2021-07-26:
Thanks, will check these out once I get time.

> The dry mild state error isn't coming back. I'm wondering if it is a residual incorrect setting from the previous integration from HACS

The Panasonic AC protocols are very prone to setting incorrect byte values. My powerful/quiet mode is currently broken and can't be set anymore until I reboot the unit, for example. Not the first time something like this has happened during testing.

DomiStyle, 2021-07-26:
@brywithawhy Can you enable eco mode together with powerful or quiet via the IR remote or is it a toggle?

brywithawhy, 2021-07-26:
> @brywithawhy Can you enable eco mode together with powerful or quiet via the IR remote or is it a toggle?

You can enable Eco/Powerfull with Quiet. One button toggles between eco/powerfull. Fan speed cycles through auto, 1, 2, 3, 4, 5, quiet.

Below is eco and quiet together:
```
[19:26:22][V][esppac:268]: Polling AC
[19:26:22][V][esppac:259]: TX: 70 0A 00 00 00 00 00 00 00 00 00 00 86
[19:26:22][V][esppac:259]: RX: 70 20 04 2E 80 A0 F0 04 00 40 40 00 4C 30 00 00 00 00 14 0E FF 14 0E FF 80 80 FF 80 AC 00 06 C1 38 14 AE
[19:26:22][W][esppac:438]: Received unknown vertical swing mode
[19:26:22][D][climate:369]: 'lounge_ac' - Sending state:
[19:26:22][D][climate:372]:   Mode: HEAT_COOL
[19:26:22][D][climate:374]:   Action: HEATING
[19:26:22][D][climate:377]:   Fan Mode: AUTO
[19:26:22][D][climate:383]:   Preset: ECO
[19:26:22][D][climate:389]:   Swing Mode: VERTICAL
[19:26:22][D][climate:392]:   Current Temperature: 20.00°C
[19:26:22][D][climate:398]:   Target Temperature: 23.00°C
[19:26:27][V][esppac:268]: Polling AC
[19:26:27][V][esppac:259]: TX: 70 0A 00 00 00 00 00 00 00 00 00 00 86
[19:26:27][V][esppac:259]: RX: 70 20 04 2E 80 A0 F0 04 00 40 40 00 4C 30 00 00 00 00 14 0E FF 14 0E FF 80 80 FF 80 AC 00 06 80 85 71 45
[19:26:27][W][esppac:438]: Received unknown vertical swing mode
[19:26:27][D][climate:369]: 'lounge_ac' - Sending state:
[19:26:27][D][climate:372]:   Mode: HEAT_COOL
[19:26:27][D][climate:374]:   Action: HEATING
[19:26:27][D][climate:377]:   Fan Mode: AUTO
[19:26:27][D][climate:383]:   Preset: ECO
[19:26:27][D][climate:389]:   Swing Mode: VERTICAL
[19:26:27][D][climate:392]:   Current Temperature: 20.00°C
[19:26:27][D][climate:398]:   Target Temperature: 23.00°C
```

DomiStyle, 2021-07-26:
Eco mode, powerful/quiet should work now.
Horizontal swing should show as unsupported now.

You can enable the new sensors and disable nanoeX as shown in the ac.yaml.example.

Only manual swing sensors are missing, the rest should work.

brywithawhy, 2021-07-26:
Hmmm doesn't seem to work.

I thought Quiet should go under fan speeds rather than presets since you can't have high fan speed and quiet mode at the same time. The preset menu should have Powerfull or eco in the preset menu as you cycle between those options on the remote. Is this possible?
There's also a mild dry switch now but I don't have any such feature on my air con. What is this for?

I can't get the eco light to come on the unit using any combination of settings. Here is a post of the log with Heat/Cool enabled with Eco

```
[07:55:57][V][esppac:287]: Polling AC
[07:55:57][V][esppac:297]: TX: 70 0A 00 00 00 00 00 00 00 00 00 00 86
[07:55:57][V][esppac:297]: RX: 70 20 04 2E 80 A0 30 00 00 40 00 00 4C 33 00 00 00 00 14 07 FF 17 07 FF 80 80 FF 80 E6 03 16 80 85 71 04
[07:55:57][D][climate:369]: 'lounge_ac' - Sending state:
[07:55:57][D][climate:372]:   Mode: HEAT_COOL
[07:55:57][D][climate:374]:   Action: HEATING
[07:55:57][D][climate:377]:   Fan Mode: AUTO
[07:55:57][D][climate:386]:   Custom Preset: None
[07:55:57][D][climate:389]:   Swing Mode: OFF
[07:55:57][D][climate:392]:   Current Temperature: 20.00°C
[07:55:57][D][climate:398]:   Target Temperature: 23.00°C
```

And this is with the eco toggle off and eco mode on using the IR remote

```
[07:57:32][V][esppac:287]: Polling AC
[07:57:32][V][esppac:297]: TX: 70 0A 00 00 00 00 00 00 00 00 00 00 86
[07:57:33][V][esppac:297]: RX: 70 20 04 2E 80 A0 F0 00 00 40 40 00 4C 33 00 00 00 00 14 07 FF 16 07 FF 80 80 FF 80 C5 03 15 C0 00 00 DD
[07:57:33][D][climate:369]: 'lounge_ac' - Sending state:
[07:57:33][D][climate:372]:   Mode: HEAT_COOL
[07:57:33][D][climate:374]:   Action: HEATING
[07:57:33][D][climate:377]:   Fan Mode: AUTO
[07:57:33][D][climate:386]:   Custom Preset: None
[07:57:33][D][climate:389]:   Swing Mode: VERTICAL
[07:57:33][D][climate:392]:   Current Temperature: 20.00°C
[07:57:33][D][climate:398]:   Target Temperature: 23.00°C
```

The Quiet preset works but as mentioned, when using the remote this is an option cycling through the fan speed buttons. The fan speeds also have 'middle' and 'focus' in the menu now which don't make a lot of sense to me.

[image]

Let me know if you need anything further

brywithawhy, 2021-07-26:
On another note, the comfort cloud integration reported back power usage stats too. Would be cool if this was available as well.

DomiStyle, 2021-07-27:
> I thought Quiet should go under fan speeds rather than presets since you can't have high fan speed and quiet mode at the same time.

Quiet/powerful is a toggle switch on most ACs I have seen so far. When you switch back from Powerful/Quiet to Normal mode it should use your previous set fan speed since it's not overwritten. Eco mode is not available on my unit.

Just for reference, this is what my remote looks like: https://user-images.githubusercontent.com/975131/122646463-90e43b80-d11f-11eb-9d27-977b41d4bbbc.jpg

> There's also a mild dry switch now but I don't have any such feature on my air con. What is this for?

You can disable it by removing the sensor and the corresponding function call in your config (same for nanoeX). It's a feature on mine to reduce how much humidity it removes from the room.

> I can't get the eco light to come on the unit using any combination of settings. Here is a post of the log with Heat/Cool enabled with Eco

Can you post the log as you go back and forth on the eco mode switch? Make sure you have this line not commented:
> ac->set_eco_switch(id(ac01_eco_switch));

> The fan speeds also have 'middle' and 'focus' in the menu now which don't make a lot of sense to me.

Those are the presets Home Assistant provides and yes, they are non-descriptive.

However, since ESPHome 1.20.1 released today I can finally use custom presets and rename them to something more understandable. A verson which implements this should be done in a few hours.

> On another note, the comfort cloud integration reported back power usage stats too. Would be cool if this was available as well.

I use external methods of monitoring energy usage so I'm not interested to implement this for now (same with the timer and sleep features). But feel free to create a feature request for it.

brywithawhy, 2021-07-27:
> Make sure you have this line not commented:
> 
> > ac->set_eco_switch(id(ac01_eco_switch));

I missed this in the config file previously - sorry!. I just uncommented it and reuploaded. Now when I toggle the Eco switch, it turns itself back off straight away.

Here is the logs from toggling eco on and off with my remote. It contains setting powerfull too as I have to cycle through powerful first to get to eco. Everything else is on Auto for reference.

```
[17:27:26][V][esppac:287]: Polling AC
[17:27:26][V][esppac:297]: TX: 70 0A 00 00 00 00 00 00 00 00 00 00 86
[17:27:27][V][esppac:297]: RX: 70 20 04 2E 80 A0 F0 00 00 40 00 00 4C 2A 00 00 00 00 16 10 FF 17 10 FF 80 80 FF 80 14 02 0C C1 38 14 7F
[17:27:27][D][climate:369]: 'lounge_ac' - Sending state:
[17:27:27][D][climate:372]:   Mode: HEAT_COOL
[17:27:27][D][climate:374]:   Action: COOLING
[17:27:27][D][climate:377]:   Fan Mode: AUTO
[17:27:27][D][climate:386]:   Custom Preset: None
[17:27:27][D][climate:389]:   Swing Mode: VERTICAL
[17:27:27][D][climate:392]:   Current Temperature: 22.00°C
[17:27:27][D][climate:398]:   Target Temperature: 23.00°C
[17:27:31][V][esppac:287]: Polling AC
[17:27:32][V][esppac:297]: TX: 70 0A 00 00 00 00 00 00 00 00 00 00 86
[17:27:32][V][esppac:297]: RX: 70 20 04 2E 80 A0 F0 02 00 40 00 00 4C 32 00 00 00 00 16 10 FF 17 10 FF 80 80 FF 80 3A 02 0D 80 85 71 E5
[17:27:32][D][climate:369]: 'lounge_ac' - Sending state:
[17:27:32][D][climate:372]:   Mode: HEAT_COOL
[17:27:32][D][climate:374]:   Action: COOLING
[17:27:32][D][climate:377]:   Fan Mode: AUTO
[17:27:32][D][climate:386]:   Custom Preset: Powerful
[17:27:32][D][climate:389]:   Swing Mode: VERTICAL
[17:27:32][D][climate:392]:   Current Temperature: 22.00°C
[17:27:32][D][climate:398]:   Target Temperature: 23.00°C
[17:27:37][V][esppac:287]: Polling AC
[17:27:37][V][esppac:297]: TX: 70 0A 00 00 00 00 00 00 00 00 00 00 86
[17:27:37][V][esppac:297]: RX: 70 20 04 2E 80 A0 F0 00 00 40 40 00 0C 2A 00 00 00 00 16 10 FF 17 10 FF 80 80 FF 80 58 02 0E C0 00 00 86
[17:27:37][D][climate:369]: 'lounge_ac' - Sending state:
[17:27:37][D][climate:372]:   Mode: HEAT_COOL
[17:27:37][D][climate:374]:   Action: COOLING
[17:27:37][D][climate:377]:   Fan Mode: AUTO
[17:27:37][D][climate:386]:   Custom Preset: None
[17:27:37][D][climate:389]:   Swing Mode: VERTICAL
[17:27:37][D][climate:392]:   Current Temperature: 22.00°C
[17:27:37][D][climate:398]:   Target Temperature: 23.00°C
[17:27:42][V][esppac:287]: Polling AC
[17:27:42][V][esppac:297]: TX: 70 0A 00 00 00 00 00 00 00 00 00 00 86
[17:27:42][V][esppac:297]: RX: 70 20 04 2E 80 A0 F0 00 00 40 00 00 0C 2A 00 00 00 00 16 10 FF 17 10 FF 80 80 FF 80 70 02 0E C1 38 14 61
[17:27:42][D][climate:369]: 'lounge_ac' - Sending state:
[17:27:42][D][climate:372]:   Mode: HEAT_COOL
[17:27:42][D][climate:374]:   Action: COOLING
[17:27:42][D][climate:377]:   Fan Mode: AUTO
[17:27:42][D][climate:386]:   Custom Preset: None
[17:27:42][D][climate:389]:   Swing Mode: VERTICAL
[17:27:42][D][climate:392]:   Current Temperature: 22.00°C
[17:27:42][D][climate:398]:   Target Temperature: 23.00°C
```

Here's a pic of my remote for reference

[image]

> I use external methods of monitoring energy usage so I'm not interested to implement this for now (same with the timer and sleep features). But feel free to create a feature request for it.

Happy to prioritise getting everything working first, but it would be good to add this later. Where I live we have a silly "demand usage" metering scheme, so I've got home assistant to prevent things turning on consecutively so we get a cheaper bill. I'll add a feature request later.

daggy72, 2021-07-27:
great work guys and in particular @DomiStyle and @brywithawhy !!!

I have 8 Panasonic Etherea units (CS-ZxxTKEW) in my home and office which are all connected via CZ-TACG1-modules to Panasonic Comfort Cloud (PCC). Which is turning out to be quite problematic lately. About 5 "server unreachables" in the last month. 

Since I am fasing out all cloud-based solutions for local HA-controlled alternatives the aircons were high on my wish list. 

Is there anything I can do to help/support? Where shall I start?

by the way: the HACS-PCC integration never managed to visualise the NanoeX function my aircon has. Not even the original PanasonicCC App supports it, so I am hoping for this project to finally get me my nanoex support ;-)

DomiStyle, 2021-07-27:
>Now when I toggle the Eco switch, it turns itself back off straight away.

Wrong byte, should hopefully work now.
You can also try turning it on/off via the remote and the button in Home Assistant should update automatically after 5 seconds or so.

> Here's a pic of my remote for reference
Looks like a few things are switched around there. Does the AC remember your previous fan power when cycling through 
Powerful/Eco/Normal or does it reset to normal speed?

> About 5 "server unreachables" in the last month.

That's the reason I wrote this, their app and server is horrible.

> Is there anything I can do to help/support? Where shall I start?

You can try switching one of your units over to an ESP and help with testing. Also, you can post a picture of your remote so we can see the layout for your AC.

> Not even the original PanasonicCC App supports it, so I am hoping for this project to finally get me my nanoex support

It does! My AC supports nanoeX and I can toggle it from my library just like normal. Probably just their laziness to not support it for CZ-TACG1 since it is supported on DNSK-P11.

brywithawhy, 2021-07-27:
> Quiet/powerful is a toggle switch on most ACs I have seen so far. When you switch back from Powerful/Quiet to Normal mode it should use your previous set fan speed since it's not overwritten

Also you're right. Was just trying different things and even though I use the fan speed button to toggle quiet, the quiet mode turns off when changing to powerfull. However I can run Eco and Quiet mode at the same time.

> Does the AC remember your previous fan power when cycling through
> Powerful/Eco/Normal or does it reset to normal speed?

Seems to, but I'm not certain if that is the remote remembering it and sending the last speed when I turn off or if the aircon itself is.

Will test and report back shortly.

brywithawhy, 2021-07-27:
Confirmed - Eco mode is now working.

brywithawhy, 2021-07-27:
> Does the AC remember your previous fan power when cycling through
> Powerful/Eco/Normal or does it reset to normal speed?

I can't think of an effective way to test this, however I suspect the remote remembers the last setting and re-sends it when turning it off.

DomiStyle, 2021-07-27:
> I can't think of an effective way to test this, however I suspect the remote remembers the last setting and re-sends it when turning it off.

You can try from Home assistant: Set the fan speed to high then set the preset to powerful and then back to none/normal again.
Now you can check what fan mode is showing.

> However I can run Eco and Quiet mode at the same time.

Yes, so Quiet/Powerful being a preset and Eco being a switch should work for both cases.

brywithawhy, 2021-07-27:
> You can try from Home assistant: Set the fan speed to high then set the preset to powerful and then back to none/normal again.
> Now you can check what fan mode is showing.

I'm embarrassed at how much sense that made. Sleep deprived over here 😫 
Yes - fan speed setting is persistant between preset changes for both powerfull and quiet. Awesome!

One thing I'm noticing is that the connection to the esp32 sometimes times out but it reconnects straight away. This is causing the entities to become unavailable momentarily. Could this be changed to say, have 3-5 consecutive timeouts before declaring it as unavailable?

[image]

DomiStyle, 2021-07-27:
> Yes - fan speed setting is persistant between preset changes for both powerfull and quiet. Awesome!

Sounds good, will leave it as is for now. Should work for the majority of ACs.

> One thing I'm noticing is that the connection to the esp32 sometimes times out but it reconnects straight away.

Spotty wifi connection? I don't think you can choose any sort of timeout in the ESPHome API or in the Home Assistant ESPHome component.

Your best chance might be adding a IoT SSID with a less used frequency for IoT devices. These devices don't need a high bandwidth so you can reduce the channel width to a minimum to reduce interference at the cost of speed. Other than that maybe adding a wifi repeater or another access point could help.

brywithawhy, 2021-07-27:
> Sounds good, will leave it as is for now. Should work for the majority of ACs.

Great, thanks for all your help getting this working. Let me know if you need any further testing.

> Spotty wifi connection?

I've got a pretty decent mesh network with unifi hardware, and all my smart devices are on a dedicated 2.4ghz IoT SSID. The nearest access point is literally a metre away from the esp32. Never mind though, not a big deal. I'm getting a pretty crappy ping to it where other devices are fine so I'll look into the esp32 side of things to rectify that.

Thanks again

DomiStyle, 2021-07-29:
Does anybody here have a picture of the original CZ-TACG1 installed in the AC I can use for the readme?
And maybe one of the installed ESP with logic converter?

Also, is the wifi LED on the unit blinking, lit or off?

crispie, 2021-07-29:
[image]

brywithawhy, 2021-07-29:
Here's mine. I don't have a neat little pocket for it so mine is stuck to the side.
All 3 lights are on and Link blinks when commands are sent like an activity LED on a network card.
Also my current ESP32 is just hanging from the wire until the proper JST PA connectors arrive. I've opened the CZ-TACG1 case and unclipped the connector in order to use it until then.

[image]

[image]

[image]

DomiStyle, 2021-07-30:
Thanks for the pictures, hopefully I will get everything cleaned up with instructions over the weekend and push the new version to master.

There is still a pending bug in Home Assistant that prevents the custom fan speeds from working but it should be done soon.

@crispie 
> I get a python syntax error when doing esphome ac.yaml run. See below output. Any idea?

Interesting, never saw that error before.

What is the output of `esphome version`?
Can you post the entire log?
What does your folder structure look like and did you make any changes to the ac.yaml?
Did you pull the dev branch?

brywithawhy, 2021-07-30:
Looking forward to it. Any progress on manual swing positions working?

DomiStyle, 2021-07-30:
@brywithawhy Done already, I'm just waiting for the Home Assistant update.

crispie, 2021-07-30:
@DomiStyle 

> What is the output of `esphome version`?
bash-5.1# esphome version                                                                                                                                                                                                                                                                                                                                                                         
Version: 1.20.1  

> What is the output of `python --version` and `python3 --version`?
bash-5.1# python --version                                                                                                                                                                                                                                                                                                                                                                        
Python 3.9.5 

bash-5.1# python3 --version                                                                                                                                                                                                                                                                                                                                                                       
Python 3.9.5  

> Can you post the entire log?
[33mWARNING Calling ESPHome with the configuration before the command is deprecated and will be removed in the future. [[0m                                                                                                                                                                                                                                                                      
[[33mWARNING Please instead use:[[0m                                                                                                                                                                                                                                                                                                                                                              
[[33mWARNING    esphome -v run ac.yaml[[0m                                                                                                                                                                                                                                                                                                                                                        
[[32mINFO Reading configuration ac.yaml...[[0m                                                                                                                                                                                                                                                                                                                                                    
[[32mINFO Generating C++ source...[[0m                                                                                                                                                                                                                                                                                                                                                            
[[36mDEBUG Running to_code in esphome.components.async_tcp (num 17)[[0m                                                                                                                                                                                                                                                                                                                           
[[36mDEBUG Adding: // async_tcp:[[0m                                                                                                                                                                                                                                                                                                                                                              
[[36mDEBUG Adding library: <esphome.core.Library object at 0x7f687614f3d0>[[0m                                                                                                                                                                                                                                                                                                                    
[[36mDEBUG  -> finished[[0m                                                                                                                                                                                                                                                                                                                                                                       
[[36mDEBUG Running to_code in esphome.core.config (num 0)[[0m                                                                                                                                                                                                                                                                                                                                     
[[36mDEBUG Adding: // esphome:[[0m                                                                                                                                                                                                                                        [... cut ...]
                                                                                                                                                                                                                                                                                                                                                                             
> What does your folder structure look like and did you make any changes to the ac.yaml?
esphome-panasonic-ac directory under /config (for homeassistant), ac-yaml.example changed to contain correct wifi network/pw and copied to /config (same level as esphome-panasonic-ac

> Did you pull the dev branch?
yes

DomiStyle, 2021-07-30:
@crispie No idea unfortunately, maybe you can use ESPHome from Docker instead so there are no dependency issues?

@brywithawhy Everything is in the repo now, it needs the latest Home Assistant Release candidate. Instructions on how to add the swing selector(s) are [here](https://github.com/DomiStyle/esphome-panasonic-ac/tree/dev-cztacg1#adding-manual-swing-selection-to-home-assistant).

jesserockz, 2021-07-31:
Hello, Just found this repo after searching around for a few hours today (Not specifically ESPHome based, but anything around `CZ-TACG1`) 
I have wired a d1-mini to my panasonic CS-E18NKR CN-CNT port but I am getting no response at all coming in on uart after the poll is sent out.

Any ideas? I have tried swapping RX/TX. Perhaps my AC unit just doesn't use the port :cry: 

<details>
<summary>
Logs
</summary>

```text
[21:41:04][V][esppac:279]: Polling AC
[21:41:04][VV][uart_esp8266:111]:     Wrote 0b01110000 (0x70)
[21:41:04][VV][uart_esp8266:111]:     Wrote 0b00001010 (0x0A)
[21:41:04][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[21:41:04][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[21:41:04][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[21:41:04][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[21:41:04][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[21:41:04][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[21:41:04][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[21:41:04][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[21:41:04][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[21:41:04][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[21:41:04][VV][uart_esp8266:111]:     Wrote 0b10000110 (0x86)
[21:41:04][V][esppac:272]: TX: 70 0A 00 00 00 00 00 00 00 00 00 00 86
[21:41:09][V][esppac:279]: Polling AC
[21:41:09][VV][uart_esp8266:111]:     Wrote 0b01110000 (0x70)
[21:41:09][VV][uart_esp8266:111]:     Wrote 0b00001010 (0x0A)
[21:41:09][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[21:41:09][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[21:41:09][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[21:41:09][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[21:41:09][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[21:41:09][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[21:41:09][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[21:41:09][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[21:41:09][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[21:41:09][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[21:41:09][VV][uart_esp8266:111]:     Wrote 0b10000110 (0x86)
[21:41:09][V][esppac:272]: TX: 70 0A 00 00 00 00 00 00 00 00 00 00 86
```
</details>

DomiStyle, 2021-07-31:
@jesserockz Did you have a CZ-TACG1 plugged in there before?

I'm not sure if the port needs to actually be enabled somehow since I didn't check the protocol during pairing but I was able to just swap the CZ-TACG1 out for an ESP after pairing and the AC responded fine. I'm even using both wifi adapters at the same time right now.
I did have some issues at first because of a broken channel on the logic converter but it worked after changing to channel 2 & 3 on the logic converter.

Unfortunately the CZ-TACG1 is harder to debug than the DNSK-P11 because it doesn't push any information and only responds to polling. The polling seems to be working fine in your log though.

I have a second AC I can test that never had a CZ-TACG1 connected to it to see if it works there out of the box.

Is the D1-Mini an ESP32 or an ESP8266?

jesserockz, 2021-07-31:
No I have not had a CZ-TACG1 in there and dont have one to try. 
It's an ESP8266. I did not use a logic converter as the 8266 can handle the 5V uart lines (The reason I am not testing with an ESP32)

DomiStyle, 2021-07-31:
Did you disable UART logging or are you using software serial for the 8266?

Is your CN-CNT connector exposed like in the picture [here](https://user-images.githubusercontent.com/54029411/127578086-814ecb7a-ccd4-4f08-9bc1-0b9e6a32307d.jpg) or is it on the mainboard? If it's exposed I don't see a reason why it shouldn't be supported.

I tried connecting the ESP to a fresh AC and everything worked so I assume there is no packet that needs to be sent to activate the port.

I will test it on an 8266 next and see if it works aswell.

jesserockz, 2021-07-31:
> Did you disable UART logging or are you using software serial for the 8266?
Yes, haha (I am an ESPHome maintainer 😉)

> Is your CN-CNT connector exposed like in the picture [here](https://user-images.githubusercontent.com/54029411/127578086-814ecb7a-ccd4-4f08-9bc1-0b9e6a32307d.jpg) or is it on the mainboard? If it's exposed I don't see a reason why it shouldn't be supported.

No, it's on the mainboard and I have to pull the cover off to get to it. I think my model (NKR) just doesn't use it, might be too old. I don't know when it was installed in this house as we got the house last year.

DomiStyle, 2021-07-31:
> Yes, haha (I am an ESPHome maintainer wink)

I knew that name sounded familiar. ;)

It's strange that the connector is not on the outside like it usually is but I definitely have heard from people who also had to manually add the cable from the mainboard. Unfortunately it's not possible to see if the connector is actually "alive" unless you poll.

Panasonic does indeed not list the NKR model as supported on their website. so it is possible that it's too old.

I tried with an ESP8266 and I also can't get the AC to respond with it for some reason. Everything looks normal with a logic analyzer.

It might be worth doing a final test with an ESP32 and a logic converter or a USB to UART bridge? Sending `0x70 0x0A 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x86` should work.

For reference, here's a picture of the connector: https://raw.githubusercontent.com/DomiStyle/esphome-panasonic-ac/dev-cztacg1/images/cztacg1/connector.jpg
RX/TX are from the ESP side.

jesserockz, 2021-07-31:
I'll set up an ESP32 tomorrow then and see if I get anything. 

I just checked before, and both tx/rx are at 5V compared to gnd all the time (disconnected from ESP), so perhaps this port is not intended for use on this model.

DomiStyle, 2021-07-31:
> I just checked before, and both tx/rx are at 5V compared to gnd all the time (disconnected from ESP)

That's actually a good sign, the serial idle state is high. If 12V is present too it seems like a normal CN-CNT connector.

jesserockz, 2021-07-31:
Yeah it does have 12V too

DomiStyle, 2021-07-31:
Sounds good then.

I measured the RX/TX pin on the ESP8266 and they are only 3.3V at idle. That might cause the AC to interpret it as constant low. At least mine doesn't respond either.

jesserockz, 2021-08-01:
Couldn't find my level converter today but the bus pirate gets a response back. Will report back once I sort out the circuitry and get it working (or not)

jesserockz, 2021-08-01:
So partially working. Level converter was not needed in the end, and actually didn't make a difference. The RX pin on the ESP8266 Hardware Serial was blocking the AC from sending anything. I could not see anything coming via the bus pirate tapped into that line, but as soon as I disconnected the 8266 RX line the poll response came in. In the end I got it working with SoftwareSerial which does not have the same properties.

A few things to note from my unit:
nanoe-g doesn't report or control, always 0x00 (No packet change when toggling from remote)
econavi does't report or control, always 0x00 (No packet change when toggling from remote)
current_temp doesn't report

DomiStyle, 2021-08-01:
> In the end I got it working with SoftwareSerial which does not have the same properties.

Strange, but at least it works now.

> nanoe-g doesn't report or control, always 0x00 (No packet change when toggling from remote)
econavi does't report or control, always 0x00 (No packet change when toggling from remote)
current_temp doesn't report

That's strange but maybe a limitation of the older unit. None of these are regularly needed so it's not that bad I think.

brywithawhy, 2021-08-01:
> @brywithawhy Everything is in the repo now, it needs the latest Home Assistant Release candidate. Instructions on how to add the swing selector(s) are [here](https://github.com/DomiStyle/esphome-panasonic-ac/tree/dev-cztacg1#adding-manual-swing-selection-to-home-assistant).

Thanks for the update.
I've just updated to the latest HA release candidate and your latest code.
Happy to report that all functions are now working perfectly.
The connectors I ordered arrived yesterday too so looking forward to tidying up the setup and deploying across the 3 other units in the house :)

Let me know if I can help with anything further

[42 more replies]

closed 2021-08-18 by DomiStyle