DomiStyle/esphome-panasonic-ac #2: More clear instructions

War3333, 2021-04-24:
Sorry but it's not really clear in the instructions if i should connect RX cable to RX ESP32 or RX cable to TX on ESP32.

From the color code it seems that the correct RX to RX but i think should be more clear, maybe put it in the table.

On a side note on the first boot it always reboot and give gibberish values, it's normal?

Francesco

DomiStyle, 2021-04-24:
You're right, I will add that detail to the readme.

The connector description is from the side of the ESP. RX goes to logic level converter goes to GPIO16 (RX).

Do you have an example of the gibberish being logged? Via serial it is certainly possible to receive gibberish during the ESP bootup but over the network it should not happen. It might also be crashing on first boot for some reason and the gibberish you're seeing is the crash dump.

Does your unit support external temperature? I have seen mine report 255 before, but I'm not sure what triggers that behaviour. I might just filter out all values that are not plausible and stop reporting outside temperature if none is available.

leon-v, 2021-04-24:
I found this too..I had to research some to find out. I can't remember the answer, I just wanted you to know you are not alone in this.

War3333, 2021-04-24:
It has external temperature readings and it's working fine, only when the unit is off show 126°.

I have installed it on a Panasonic Etherea CS-Z25VKEW (not sure of the outside unit)
I want also to install on a Panasonic Etherea CS-Z35VKEW that i have in another room

For the gibberish i will try to log something. For now an image.
[image]

DomiStyle, 2021-05-04:
@War3333 I added more instructions to the readme and a fix for the invalid temperature readings, can you give it a try?

War3333, 2021-05-04:
Not really any changes.

I looked up at the code a bit:

```
void PanasonicAC::setup()
{
  // Initialize times
  initTime = millis();
  lastPacketSent = millis();

  this->fan_mode = climate::CLIMATE_FAN_OFF;
  this->swing_mode = climate::CLIMATE_SWING_OFF;
  this->target_temperature = (int8_t) 0;
  this->publish_state(); // Post dummy state so Home Assistant doesn't disconnect

  ESP_LOGI(TAG, "Panasonic AC component v%s starting...", ESPPAC_VERSION);
}
```

publish_state send a dummy report with a NULL, and it works ;) 

For the external temperature you added a correction in the last commit but reverted it in the latest, So I copy and pasted and it manually and it works .

Francesco

DomiStyle, 2021-05-05:
It shouldn't be necessary to post a dummy state on start, the value will get overwritten by the data from the AC once the next query runs.

If it reverts back to a strange value after a while it might be because the packet format changed and the value I'm reading isn't actually the target temperature.

Can you enable verbose logging and post a debug log with the latest version? Wait until the first query is sent, then change the target temperature from the IR remote and check if that corrects the temperature.

I also added the threshold check to all temperature values now so values over 100 shouldn't be reported anymore.

sygys, 2021-07-15:
Is there a way you could provide a link to the bi directional logic converter so i know i buy the right one. When searching i find a ton of different ones.

Also could you add a layout to the tutorial that shows graphically which cable goes to which port on which device.

Thanks allot! I have been searching for this for 6 months now i really hope i get this working! Im using panasonic cloud comfort at the moment from sockless. while it does work great, the problem is that everytime panasonic changes something to the api or the app it stops working.... i need a more stable solution.

Will this ESP solution also let the AC show up as a climate device in Home Assistant?

sygys, 2021-07-15:
By the way i have a panasonic cs-tz35tkew does anyone know if this solution works for the wifi module on that unit? I cant see any type numbers on the wifi module.

DomiStyle, 2021-07-15:
@sygys Doesn't really matter much as long as it can handle 3.3V and 5V, I used these: https://www.amazon.de/dp/B08Y1GYXP4.

> Also could you add a layout to the tutorial that shows graphically which cable goes to which port on which device.

I don't have time right now to add a grahical layout but I clarified the connection table a little bit more so every pin should be clear.

> Will this ESP solution also let the AC show up as a climate device in Home Assistant?

Yes, once you add the ESPHome integration in Home Assistant the climate device will automatically show up.

> By the way i have a panasonic cs-tz35tkew does anyone know if this solution works for the wifi module on that unit? I cant see any type numbers on the wifi module.

This library only works with the DNSK-P11 that is mounted on the far left side of the unit. It is not compatible with the CZ-TACG1 that is mounted on the top right, which is accessible without disassembly.

If you have a DNSK-P11 it's a drop-in replacement.
If you currently have a CZ-TACG1 installed you will have to look up the service manual on the Panasonic website and check if there's a CN-WLAN connector on your circuit board.

It should look like this:

[image]

sygys, 2021-07-15:
Just opened up the AC unit and thank god it has the DNSK-P11 module. but its on the bottom right of the AC. 

https://photos.app.goo.gl/9ghx5W495yUQ5oPt5

I will let you know if it works. Then you can add the Panasonic cs-tz35tkew to the list of working AC's

DomiStyle, 2021-07-15:
@sygys That's strange, never have seen it mounted there. Hopefully the cable is still long enough so you can route it outside like [this](https://raw.githubusercontent.com/DomiStyle/esphome-panasonic-ac/master/images/installed_controller.jpg) for easier access.

No matter, the DNSK-P11 definitely works with this library.
Happy tinkering!

sygys, 2021-07-15:
Thanks for the help! I dont have that space above to put in the esp board so i have to fix it somewhere else. I guess i will 3d print a case for the espboard with the same size as the wifi module and put it in the same place as it is now.

I ordered the bi directional board from your link and wanted to start programming the ESP board.

In your tutorial you write something about pull the repository. I dont think i have done this before. Normally you only need a single yaml file to write the esp board. Could you tell me where the rest of files go exactly? Is it a custom component? and what should the folder be named?

sorry for all the questions...

DomiStyle, 2021-07-15:
@sygys No worries.

This is a custom component for ESPHome so you have to get the repository files in order to build it. To do that you go into your ESPHome folder where you want your configs to be and run `git clone https://github.com/DomiStyle/esphome-panasonic-ac.git`. (If you do not have git installed you can also grab the files from the Download button on Github)

That will clone the repo into your current directory in the folder `esphome-panasonic-ac`. Now copy the `ac.yaml` in this folder to your config folder (one folder above) and edit it to your liking. If you have multiple units you can copy it multiple times and number them `ac01.yaml` and so on.

Once that is done you can deploy it on your device with `esphome ac.yaml run`. That will compile everything and upload it to your ESP32.

sygys, 2021-07-15:
So if im correct it should look like this: https://photos.app.goo.gl/EmVFZTSMwf2K6MdA8

DomiStyle, 2021-07-15:
Yes, that's correct, the includes is the folder name. So you can change either.

sygys, 2021-07-15:
The very last question for today (and then ill be out of your hair)

what does "  use_address: ac01.example.org" do? Do i need to change this? And if so to what.

DomiStyle, 2021-07-15:
It's the name of the node in the network, you can remove it entirely if you do not need the node to have a certain name or set it to <nodename>.<domain> or just <nodename>.

sygys, 2021-07-15:
hmmm i get allot of errors while compiling :(

`Compiling /data/airco_slaapkamer_esp/.pioenvs/airco_slaapkamer_esp/src/esphome/components/logger/logger.cpp.o
src/esphome-panasonic-ac-master/esppac.cpp: In member function 'virtual esphome::climate::ClimateTraits PanasonicAC::traits()':
src/esphome-panasonic-ac-master/esppac.cpp:14:10: error: 'class esphome::climate::ClimateTraits' has no member named 'set_supports_heat_cool_mode'
   traits.set_supports_heat_cool_mode(true);
          ^
src/esphome-panasonic-ac-master/esppac.cpp: In member function 'virtual void PanasonicAC::control(const esphome::climate::ClimateCall&)':
src/esphome-panasonic-ac-master/esppac.cpp:126:12: error: 'CLIMATE_MODE_HEAT_COOL' is not a member of 'esphome::climate'
       case climate::CLIMATE_MODE_HEAT_COOL:
            ^
src/esphome-panasonic-ac-master/esppac.cpp: In member function 'void PanasonicAC::determine_mode(byte)':
src/esphome-panasonic-ac-master/esppac.cpp:340:20: error: 'CLIMATE_MODE_HEAT_COOL' is not a member of 'esphome::climate'
       this->mode = climate::CLIMATE_MODE_HEAT_COOL;
                    ^
src/esphome-panasonic-ac-master/esppac.cpp: In member function 'void PanasonicAC::determine_action()':
src/esphome-panasonic-ac-master/esppac.cpp:441:70: error: 'CLIMATE_MODE_HEAT_COOL' is not a member of 'esphome::climate'
   else if((this->mode == climate::CLIMATE_MODE_COOL || this->mode == climate::CLIMATE_MODE_HEAT_COOL) && this->current_temperature + ESPPAC_TEMPERATURE_TOLERANCE >= this->target_temperature)
                                                                      ^
src/esphome-panasonic-ac-master/esppac.cpp:445:70: error: 'CLIMATE_MODE_HEAT_COOL' is not a member of 'esphome::climate'
   else if((this->mode == climate::CLIMATE_MODE_HEAT || this->mode == climate::CLIMATE_MODE_HEAT_COOL) && this->current_temperature - ESPPAC_TEMPERATURE_TOLERANCE <= this->target_temperature)
                                                                      ^
Compiling /data/airco_slaapkamer_esp/.pioenvs/airco_slaapkamer_esp/src/esphome/components/ota/ota_component.cpp.o
*** [/data/airco_slaapkamer_esp/.pioenvs/airco_slaapkamer_esp/src/esphome-panasonic-ac-master/esppac.cpp.o] Error 1
========================= [FAILED] Took 12.67 seconds =========================`

DomiStyle, 2021-07-15:
What's your ESPHome version?

sygys, 2021-07-15:
1.18.0

sygys, 2021-07-15:
That seemed to fix it

DomiStyle, 2021-07-15:
Good to hear, I forgot to update the minimum version to 1.19.1.

Yes, the card will update once it's connected to the AC.

sygys, 2021-07-24:
@DomiStyle Could you please verify this is correct before i start connecting this in the near future?
[image]

sygys, 2021-07-24:
I dont really understand how the ESP board gets power this way

DomiStyle, 2021-07-24:
@sygys Connect the 5V from the connector through the HV pin to the 5V/VIN pin on the ESP32.

Other than that it looks fine.

sygys, 2021-07-24:
[image]

So like this? 

Feel free to use this on your page. So it can help more people.

Just opened the wifi module case. The ESP will fit in there so i think ill modify it a bit and put all this in that case.

DomiStyle, 2021-07-24:
No, LV stays at 3V3 but VIN goes to HV so it is connected directly to the 5V of the AC.

Thanks, I will update the readme once support for the other wifi module is done.

sygys, 2021-07-24:
[image]

Could you please verify i did it right this time lol

DomiStyle, 2021-07-24:
Looks good. You could switch the ESP ground to the LV side and vice versa since that's nicer but it won't change anything.

sygys, 2021-07-25:
Thanks for everything so far! Still waiting for my logic converters to arrive. Cant wait to test this .

closed 2021-08-18 by DomiStyle