DomiStyle/esphome-panasonic-ac #4: Support for Presets like ECO and POWERFULL, and more swing modes

Expaso, 2021-06-19:
Hi @DomiStyle!

To start with: Thank you big time for this one! It works like a charm!
I wanted to do this myself for over a year or so, but never found enough time to really pull it off! Thanks again!

Currently, the biggest differences between the HA integration using Comfort Cloud, and this ESPHome solution is the way presets are handled:

This is the HA Comfort Cloud integration:
[image]

And this is the ESPHome Version:

[image]

Biggest differences are in the Preset modes, and Swing Modes.

Now, something tells me this is related to the following pull requests:

ESPHome Rework Climate Traits: https://github.com/esphome/esphome/pull/1941
ESPHome Climate: https://github.com/esphome/aioesphomeapi/pull/24#pullrequestreview-687339203

ESPHome Add preset, custom_preset and custom_fan_mode support to climate: https://github.com/esphome/esphome/pull/1471
Home Assistant Add esphome preset, custom preset and custom fan mode support: https://github.com/home-assistant/core/pull/48116

DomiStyle, 2021-06-19:
Hey, glad it's useful to more than one person. :)

I didn't know ESPHome and HA already support custom presets and fan mode, when I wrote this neither was supported so I put the powerful/quiet mode into the fan speed setting instead (focus/diffuse).

Can you go into more details on what's missing exactly? My AC remote doesn't have any Boost/Eco mode on it. The swing manual modes are exposed as sensors, can they be included in the climate instance now?

Just for reference, this is what my remote looks like and how I have included everything into Home Assistant:

[image]

[image]

Expaso, 2021-06-19:
Yeah it's super useful! 
I rather don't want all my data to end-up in the panasonic cloud. No sir, thank you very much :)  Their cloud is an unreliable piece of junk.

Let alone the possibility to add extra moisture/temp/PIR sensors or displays and switches to the ESP module.
Turn off airco when nobody in the room? Check! Turn on dry-mode when humidity is too high? Check! This is wat a real smart device looks like!

And so, on topic:  I have the same remote as you. It's the POWERFUL/QUIET button that switches BOOST/ECO. Same thing.

The pull requests I mentioned are hot right now, so I think we should wait a little until that dust settles.
From there, it should be possible to implement these functions right within the climate component. The ESPHome part seems to be in place now, the HA part is still in draft.

DomiStyle, 2021-06-19:
Yeah, their app is really sluggish compared to real local control. I was hoping for a local control interface when I bought this AC but of course everything has to be connected to the cloud nowadays.

I added the HA pull request to my subscriptions and will keep a look at it. Will add the custom options as soon as they are documented.

In the mean time you can use focus/diffuse and the vertical & horizontal swing sensors.
Focus/diffuse should toggle the boost/eco mode just like it toggles powerful/quiet for me if they are identical.

I don't think I ever documented it properly but the swing sensors can be added like this in HA:
Configuration -> Helpers -> Add -> Dropdown:

Name: Horizontal swing
Icon: mdi:swap-horizontal

Options:
* left
* left_center
* center
* right_center
* right

Name: Vertical swing
Icon: mdi:swap-vertical
Options:
* down
* down_center
* center
* up_center
* up

For the ESPHome part, it's already in the example yaml:

```
text_sensor:
  - platform: homeassistant
    id: ac01_vertical_swing
    name: ac01_vertical_swing
    entity_id: input_select.ac01_vertical_swing
    on_value:
      - homeassistant.service:
          service: input_select.select_option
          data_template:
            entity_id: input_select.ac01_vertical_swing
            option: "{{ swing }}"
          variables:
            swing: |-
              return id(ac01_vertical_swing).state;
```

Adjust entity IDs and repeat for horizontal swing.

Expaso, 2021-06-22:
Eey! Thanks! That helps tremendously!

closed 2021-08-18 by DomiStyle

Expaso, 2021-08-22:
Allright! 
@DomiStyle I really appreciate you pulled off this big change this soon!! 😀 Thank you!