DomiStyle/esphome-panasonic-ac #5: 🎉 Tested and Confirmed: Working on an ESP8266 NodeMCUv2 🎉

Expaso, 2021-06-22:
Perhaps not the correct place to share this information, but nice to hear anyway!

I still had some ESP8266 NodeMCUv2's laying around that I wanted to designate to this task. 
But.. as you already suggested, these only have 1 full UART (the second one is TX only).

This is fixable in the ESP Config, by moving the Log component to UART-1 (instead of the default UART-0):

```
# Enable logging
logger:
  hardware_uart: UART1 #Uses D4 (GPIO2) as LOG output!
```

And so we freed up UART-0 for our purpose:

```
uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  id: ac_uart
  baud_rate: 9600
  parity: EVEN
```

Yes, we loose logging on the default USB port, but hey, who cares! We have logging over WIFI.
(and a fallback on Pin GPIO2 if we would ever need it).

**No level shifter**

Next, I wanted to skip the level shifter. I don't think it's really needed, and here's why: 

1) The ESP is 5v tolerant on it's IO pins.
2) Even if it's not (since the internet is 50/50 on this one) we can fix this.

Please note! This trick works because the Panasonic AC has resistors in series with it's TX and RX lines!

- For TX line on the ESP, we don't have to change anything. We can drive the RX line of the AC directly. This line is pulled-up by a resistor in the AC, and the ESP will simply pull it down hard when sending signals. The ESP's internal clamping diodes will make sure the voltage on this line never reache a too high value (also because of the series resistor on the line at AC side).
If you are really worried about your $3 module, add a 3v3 zener-diode to GND for this task.

- For the RX line on the ESP, we have to solder a 1k resistor between GND and RX on the ESP Side. The TX line of the AC also has a series resistor, but.. there is also a pull-up resistor on the NodeMCU (because this line is also still attached to the usb2serial chip of the NodeMCU module. The combination leads to the situation whereby the voltage won't go low enough when the AC pulls this line low.
To counter this effect, adding a 1K resistor to GND will shift this line's voltage more towards ground (like a voltage divider). Now, this line won't go higher then 3.3V. This line won't go to GND completely (because of the series resistor) so to be even more on the safe side around this low-voltage, you can go as low as 500Ohm's for this resistor if your NodeMCU is very picky.

P.S. 

For the fanboys who read the ESP's 8266 datasheet:
Yes, you can move the hardware UART to pin GPIO13 and 15 instead of GPIO3 and GPIO1. But.. Your ESP won't boot, because 

[image]
https://randomnerdtutorials.com/esp8266-pinout-reference-gpios/

So, connecting your AC (with idle-high tx and rx lines) to this pin will prevent booting your ESP. It's a pity, because from an electrical perspective, this would be the better option.

I also tried D4 and D5 pins with a soft-uart, but couldn't get it to work.

DomiStyle, 2021-06-23:
Good to know that it works, logging via USB isn't needed anyway most of the time.

> I also tried D4 and D5 pins with a soft-uart, but couldn't get it to work.

I'm surprised that it didn't work since the AC uses a fairly low baudrate of 9600, nothing a ESP8266 should struggle with.

The ESP8266 and ESP32 are both 5V tolerant, but it's out of spec so I still recommend a logic converter since it's not really complicated to connect through it and they are cheap.

Expaso, 2021-06-24:
Agreed, if you have a logic converter, that is the best option. 
Problem was, I only had 2 left, and I needed to replace 4 wifi units.

I will retry the soft uart option using a scope, see if I can find anything weird.

Could you clarify out of spec? I always learned that TTL 5v and 3v3 were digitally compatible: https://learn.sparkfun.com/tutorials/logic-levels/all

DomiStyle, 2021-06-29:
Out of spec for the ESP8266 data sheet, which lists a working voltage of 2.5V to 3.6V. Doesn't really matter if it works though.

Expaso, 2021-06-29:
Ah you mean that.

For the VSS, the nodemcu has ofcourse its own 3.3v regulator., so you will always stay in-spec there. 

For the IO, above measures will make sure you also stay within specs, because of the 1k resistor on the RX line (forming a voltage divider) and the 3.3v zener on the TX line.
No 5v will ever reach 3.6v or higher on any pin if you take these measures.

Indeed, working out of specs is asking for trouble, so I value your caution.

foxy82, 2021-08-12:
Is it possible to consolidate this info into README.md? Or perhaps better an esp8266.md doc linked from README rather than in the issue tracker? I'm going to look to try this as soon as I can.

foxy82, 2021-10-31:
This worked for me I just needed a 470 ohm resistor instead of the 1k mentioned (I didn't have any 500 ohms to hand).

Sashik259, 2022-03-17:
> 

> I confirm that it works fine as sad and it's possible to leave UART0 to debug logging. I've used GPIO13 for RX and GPIO2 for TX transmission

Do you use logic converter? I use Wemos D1 mini with logic converter and GPO1 for TX and GPIO3 for RX, but it doesnt work (no reply for commands, no status receiving from AC). I've opened an issue [here](https://github.com/DomiStyle/esphome-panasonic-ac/issues/54). 

Can you help me to fix it?

kostrum, 2023-06-04:
Hi
@Expaso , could you share your full script for this integration for nodeMCU. Trying to follow your steps but still can't understand why it doesn't work...

Expaso, 2023-06-05:
Yeah sure!

```
substitutions:
  devicename: airco-zolderkamer
  use_address: $devicename.local
  human_devicename: Airco Zolderkamer
 
esp8266:
  board: nodemcuv2

esphome:
  name: $devicename
  
packages:
  device_base: !include common/device_base.yaml
  wifi: !include common/wifi.yaml
  sensor_base: !include common/sensor_base.yaml

  # Enable logging
logger:
  hardware_uart: UART1 #Uses D4 (GPIO2) as LOG output!

uart:
  rx_pin: GPIO3 
  tx_pin: GPIO1 
  id: ac_uart
  baud_rate: 9600
  parity: EVEN

# Pull in the AC external component
external_components:
  source: github://DomiStyle/esphome-panasonic-ac
  components: [panasonic_ac]

climate:
  - platform: panasonic_ac
    name: ${human_devicename}
    # For CZ-TACG1
    # type: cnt

    # For DNSK-P11
    type: wlan

    horizontal_swing_select:
      name: ${human_devicename} Horizontal Swing Mode
    vertical_swing_select:
      name: ${human_devicename} Vertical Swing Mode
    outside_temperature:
      name: ${human_devicename} Outside Temperature

    # Enable as needed
    #eco_switch:
    #  name: Panasonic AC Eco Switch
    nanoex_switch:
      name: ${human_devicename} NanoeX Switch
    #mild_dry_switch:
    #  name: Panasonic AC Mild Dry Switch

    # Useful when the ac does not report a current temperature (CZ-TACG1 only)
    # current_temperature_sensor: temperature_sensor_id

```

Pay special attention to the logger! I use the hardware uart of that on, and move the logger out of the way.

Mrsash2020, 2024-07-05:
Hello everyone. Like you I picked the easy option but `not....

I went the route of buying [this](https://www.aliexpress.com/item/1005005777409596.html?spm=a2g0o.detail.pcDetailTopMoreOtherSeller.2.1056vJqovJqoYl&gps-id=pcDetailTopMoreOtherSeller&scm=1007.40000.327270.0&scm_id=1007.40000.327270.0&scm-url=1007.40000.327270.0&pvid=d8ab699e-a984-4299-a213-fc10f30934c5&_t=gps-id:pcDetailTopMoreOtherSeller,scm-url:1007.40000.327270.0,pvid:d8ab699e-a984-4299-a213-fc10f30934c5,tpp_buckets:668%232846%238109%231935&pdp_npi=4%40dis%21AUD%216.90%216.90%21%21%2133.34%2133.34%21%402103243417201452472976308e9728%2112000034316083453%21rec%21AU%212721663486%21&utparam-url=scene%3ApcDetailTopMoreOtherSeller%7Cquery_from%3A) for all my Aircons. All work ok but I didnt see a Esphome Yaml on the Esphome website for it. After searching quite a while I got here. Basically wondering if I can use it with the right cable of course(based on what I have read) to control my Panasonic AC.

The AC is a Split Panasonic CS-E9NKR. Yes its old.

The other question is regarding the code. Its different to what I see here. I sorta need to understand it I guess but have no knowledge of coding. I was trying to amend it but failed since it does not even recognise panasoinic_ac. 

`esphome:
  name: living-ac
  friendly_name: Living AC

esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxx"

ota:
  - platform: esphome
    password: "xxxxxxxxxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Living-Ac Fallback Hotspot"
    password: "xxxxxxxx"

captive_portal:

text_sensor:
  - platform: wifi_info
    ip_address:
      name: "living-ac IP Address"
 
binary_sensor:
  - platform: status
    name: "living-ac Connect"
 
remote_receiver:
  id: ir_receiver
  pin:
    number: GPIO3
    inverted: True
    mode:
      input: true
      pullup: true
  tolerance: 55%
  dump: all
 
remote_transmitter:
  id: ir_transmit
  pin: GPIO4
  carrier_duty_percent: 50%
 
climate:
  - platform: fujitsu_general   # adjust to match your AC unit!
    name: "living-ac"
    id: climate_ac
    transmitter_id: ir_transmit
    receiver_id: ir_receiver
    sensor: temp
 
sensor:
  - platform: homeassistant
    name: "living-ac-Temp"
    entity_id: sensor.living_aq_temperature
    id: temp
    internal: true
 
  - platform: wifi_signal
    name: "living-ac WiFi Strength"
    update_interval: 60s
 
  - platform: uptime
    name:  "living-ac Uptime"
 
switch:
  - platform: restart
    name: "living-ac reboot" `

Hope anyone is able to assist. thanks.