DomiStyle/esphome-panasonic-ac #7: Compile error

brywithawhy, 2021-07-22:
This is my first ESP32 board so please bear with me. Using Windows command line with python, I get this compile error:

`c:\users\bryan\Desktop\esphome-panasonic-ac-master\ac.yaml: In lambda function:
c:\users\bryan\Desktop\esphome-panasonic-ac-master\ac.yaml:80:21: error: expected type-specifier before 'PanasonicAC'
       auto ac = new PanasonicAC(id(ac_uart));
                     ^
c:\users\bryan\Desktop\esphome-panasonic-ac-master\ac.yaml:88:17: error: could not convert '{ac}' from '<brace-enclosed initializer list>' to 'std::vector<esphome::climate::Climate*>'
       return {ac};
                 ^
Compiling .pioenvs\ac01\lib11d\WiFi\WiFiMulti.cpp.o
Compiling .pioenvs\ac01\lib11d\WiFi\WiFiSTA.cpp.o
*** [.pioenvs\ac01\src\main.cpp.o] Error 1
============================================= [FAILED] Took 16.43 seconds =============================================`

ESPHome version 1.20.0
Python 3.8

Any ideas?

DomiStyle, 2021-07-22:
Can you post your full log and folder structure?

brywithawhy, 2021-07-22:
If you mean the full compile log I've pasted it it below.
Folder structure is as per your repo. I just downloaded the zip, extracted it to my desktop, but had to comment out the includes for it to compile. I didn't think this would be an issue since it everything ran within that folder but correct me if I'm wrong.

`C:\Users\bryan\AppData\Local\Programs\Python\Python38-32>esphome run c:\users\bryan\Desktop\esphome-panasonic-ac-master\ac.yaml
←[32mINFO Reading configuration c:\users\bryan\Desktop\esphome-panasonic-ac-master\ac.yaml...←[0m
←[32mINFO Generating C++ source...←[0m
←[32mINFO Core config or version changed, cleaning build files...←[0m
←[32mINFO Compiling app...←[0m
←[32mINFO Running:  platformio run -d 'c:\users\bryan\Desktop\esphome-panasonic-ac-master\ac01'←[0m
Processing ac01 (board: nodemcu-32s; framework: arduino; platform: platformio/espressif32@3.2.0)
------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/nodemcu-32s.html
PLATFORM: Espressif 32 (3.2.0) > NodeMCU-32S
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10006.210326 (1.0.6)
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing esphome/AsyncTCP-esphome @ 1.2.2
Library Manager: AsyncTCP-esphome @ 1.2.2 has been installed!
Library Manager: Installing Hash
Warning! Could not find the package with 'Hash' requirements for your system 'windows_x86'
Found 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <Update> 1.0
|-- <WiFi> 1.0
Building in release mode
Compiling .pioenvs\ac01\src\esphome\components\api\api_connection.cpp.o
Compiling .pioenvs\ac01\src\esphome\components\api\api_pb2.cpp.o
Compiling .pioenvs\ac01\src\esphome\components\api\api_pb2_service.cpp.o
Compiling .pioenvs\ac01\src\esphome\components\api\api_server.cpp.o
Compiling .pioenvs\ac01\src\esphome\components\api\list_entities.cpp.o
Compiling .pioenvs\ac01\src\esphome\components\api\proto.cpp.o
Compiling .pioenvs\ac01\src\esphome\components\api\subscribe_state.cpp.o
Compiling .pioenvs\ac01\src\esphome\components\api\user_services.cpp.o
Compiling .pioenvs\ac01\src\esphome\components\api\util.cpp.o
Compiling .pioenvs\ac01\src\esphome\components\climate\climate.cpp.o
Compiling .pioenvs\ac01\src\esphome\components\climate\climate_mode.cpp.o
Compiling .pioenvs\ac01\src\esphome\components\climate\climate_traits.cpp.o
Compiling .pioenvs\ac01\src\esphome\components\homeassistant\text_sensor\homeassistant_text_sensor.cpp.o
Compiling .pioenvs\ac01\src\esphome\components\logger\logger.cpp.o
Compiling .pioenvs\ac01\src\esphome\components\ota\ota_component.cpp.o
src\esphome\components\climate\climate.cpp: In constructor 'esphome::climate::Climate::Climate(const string&)':
src\esphome\components\climate\climate.cpp:431:58: warning: 'esphome::climate::Climate::away' is deprecated: away is deprecated, use preset instead [-Wdeprecated-declarations]
 Climate::Climate(const std::string &name) : Nameable(name) {}
                                                          ^
In file included from src\esphome\components\climate\climate.cpp:1:0:
src\esphome\components\climate\climate.h:196:18: note: declared here
   bool away{false};
                  ^
src\esphome\components\climate\climate.cpp:431:58: warning: 'esphome::climate::Climate::away' is deprecated: away is deprecated, use preset instead [-Wdeprecated-declarations]
 Climate::Climate(const std::string &name) : Nameable(name) {}
                                                          ^
In file included from src\esphome\components\climate\climate.cpp:1:0:
src\esphome\components\climate\climate.h:196:18: note: declared here
   bool away{false};
                  ^
Compiling .pioenvs\ac01\src\esphome\components\sensor\automation.cpp.o
Compiling .pioenvs\ac01\src\esphome\components\sensor\filter.cpp.o
Compiling .pioenvs\ac01\src\esphome\components\sensor\sensor.cpp.o
Compiling .pioenvs\ac01\src\esphome\components\switch\automation.cpp.o
Compiling .pioenvs\ac01\src\esphome\components\switch\switch.cpp.o
[... 13 lines cut ...]
Compiling .pioenvs\ac01\src\esphome\core\helpers.cpp.o
Compiling .pioenvs\ac01\src\esphome\core\log.cpp.o
Compiling .pioenvs\ac01\src\esphome\core\preferences.cpp.o
Compiling .pioenvs\ac01\src\esphome\core\scheduler.cpp.o
Compiling .pioenvs\ac01\src\esphome\core\util.cpp.o
Compiling .pioenvs\ac01\src\main.cpp.o
Generating partitions .pioenvs\ac01\partitions.bin
Compiling .pioenvs\ac01\lib310\AsyncTCP-esphome\AsyncTCP.cpp.o
Compiling .pioenvs\ac01\lib11d\WiFi\ETH.cpp.o
Compiling .pioenvs\ac01\lib11d\WiFi\WiFi.cpp.o
Compiling .pioenvs\ac01\lib11d\WiFi\WiFiAP.cpp.o
Compiling .pioenvs\ac01\lib11d\WiFi\WiFiClient.cpp.o
Compiling .pioenvs\ac01\lib11d\WiFi\WiFiGeneric.cpp.o
c:\users\bryan\Desktop\esphome-panasonic-ac-master\ac.yaml: In lambda function:
c:\users\bryan\Desktop\esphome-panasonic-ac-master\ac.yaml:80:21: error: expected type-specifier before 'PanasonicAC'
[... cut ...]
                     ^
c:\users\bryan\Desktop\esphome-panasonic-ac-master\ac.yaml:88:17: error: could not convert '{ac}' from '<brace-enclosed initializer list>' to 'std::vector<esphome::climate::Climate*>'
       return {ac};
                 ^
Compiling .pioenvs\ac01\lib11d\WiFi\WiFiMulti.cpp.o
Compiling .pioenvs\ac01\lib11d\WiFi\WiFiSTA.cpp.o
*** [.pioenvs\ac01\src\main.cpp.o] Error 1
============================================= [FAILED] Took 16.43 seconds =============================================
`

DomiStyle, 2021-07-22:
You cannot comment out the includes since it will not compile the custom component anymore.

Instead, clone the repo to esphome-panasonic-ac and copy the ac.yaml one folder above so your directory structure looks like this:

* ac.yaml
* esphome-panasonic-ac
  * esppac.cpp

brywithawhy, 2021-07-22:
Yep my bad. Thought it was a Windows issue but reading further that fixed the problem.
Uploaded successfully, now going to try replacing the CZ-TACG on my unit.

closed 2021-07-22 by brywithawhy

DomiStyle, 2021-07-22:
Not sure if you're aware but the CZ-TACG is not supported (yet).

brywithawhy, 2021-07-22:
Yep, just getting ready for when it is because I'm tired of Comfort Cloud timing out and causing my automations to fail. Saw there was progress on the other feature request so will wait patiently and hope it is supported soon :)

DomiStyle, 2021-07-22:
Good to hear there are testers for when it's ready. Should hopefully be done in 1-2 weeks.