DomiStyle/esphome-panasonic-ac #9: Feature Request: Power Usage stats for CZ-TACG1

brywithawhy, 2021-07-27:
Requesting this functionality which is available in the cloud based comfort cloud integration. Monitoring and having automations etc will assist in better managing energy usage.

brywithawhy, 2021-08-01:
I'd like to propose packet 29 as being the current wattage used by the unit (or at least involved with it).

Below is a collection of captures from my unit being off, to it being turned on heat with powerful mode, then turned off again. You can see the values increase significantly, then drop off.

In the off state, it still shows to draw around 30-40 watts, however I've just learned that this is most likely the crankcase heater since it is quite cold here at the moment. (Never knew about this, so I'll be turning the main breaker off when I go away).

I'm running my other unit still connected to comfort cloud in the same manner to compare wattages but the sensor takes ages to update, so I'll post when it comes through.

If it were implemented as a sensor, it should probably have some sort of averaging included over say 10 or so packets to avoid spikes or dips.

```
70 20 00 2E 80 A0 10 04 00 40 00 00 00 2E 00 00 00 00 10 08 FF 10 08 FF 80 80 FF 80 1F 00 01 80 85 71 5D
70 20 00 2E 80 A0 10 04 00 40 00 00 00 2E 00 00 00 00 10 08 FF 10 08 FF 80 80 FF 80 1F 00 01 C1 38 14 C6
70 20 00 2E 80 A0 10 04 00 40 00 00 00 2E 00 00 00 00 10 08 FF 10 08 FF 80 80 FF 80 1F 00 01 C0 00 00 13
70 20 00 2E 80 A0 10 04 00 40 00 00 00 2E 00 00 00 00 10 08 FF 10 08 FF 80 80 FF 80 29 00 01 C1 38 14 BC
70 20 44 2E 80 A0 10 02 00 40 00 00 4C 3A 00 00 00 00 10 08 FF 10 08 FF 80 80 FF 80 7D 00 04 C0 00 00 18
70 20 44 2E 80 A0 10 02 00 40 00 00 4C 3A 00 00 00 00 14 08 FF 16 08 FF 80 80 FF 80 9F 02 0F 80 85 71 29
70 20 44 2E 80 A0 10 02 00 40 00 00 4C 3A 00 00 00 00 14 08 FF 16 08 FF 80 80 FF 80 0E 03 11 C0 00 00 6D
70 20 44 2E 80 A0 10 02 00 40 00 00 4C 3A 00 00 00 00 14 08 FF 17 08 FF 80 80 FF 80 2D 03 12 C1 38 14 FF
70 20 44 2E 80 A0 10 02 00 40 00 00 4C 3A 00 00 00 00 14 08 FF 17 08 FF 80 80 FF 80 4E 03 13 C0 00 00 2A
70 20 44 2E 80 A0 10 02 00 40 00 00 4C 3A 00 00 00 00 16 09 FF 19 09 FF 80 80 FF 80 CA 03 16 C1 38 14 58
70 20 44 2E 80 A0 10 02 00 40 00 00 4C 3A 00 00 00 00 15 09 FF 18 09 FF 80 80 FF 80 BA 03 15 80 85 71 02
70 20 44 2E 80 A0 10 02 00 40 00 00 4C 3A 00 00 00 00 13 09 FF 16 09 FF 80 80 FF 80 EB 03 16 C0 00 00 8A
70 20 44 2E 80 A0 10 02 00 40 00 00 4C 3A 00 00 00 00 13 09 FF 16 09 FF 80 80 FF 80 D9 03 16 80 85 71 E6
70 20 44 2E 80 A0 10 02 00 40 00 00 4C 3A 00 00 00 00 13 09 FF 16 09 FF 80 80 FF 80 EB 03 16 C0 00 00 8A
70 20 40 2E 80 A0 10 00 00 40 00 00 00 2E 00 00 00 00 13 09 FF 16 09 FF 80 80 FF 80 0B 00 00 C1 38 14 94
70 20 40 2E 80 A0 10 00 00 40 00 00 00 2E 00 00 00 00 14 09 FF 17 09 FF 80 80 FF 80 0B 00 00 C0 00 00 DF
70 20 40 2E 80 A0 10 00 00 40 00 00 00 2E 00 00 00 00 17 09 FF 19 09 FF 80 80 FF 80 0B 00 00 80 85 71 24
```

brywithawhy, 2021-08-01:
@DomiStyle from memory you said you had a separate method for measuring the energy use, perhaps you could compare that data against this packet?

DomiStyle, 2021-08-02:
@brywithawhy Doubtful, the AC is probably not measuring the power usage as accurately as my smart meter.

The correct method would be checking what Comfort Cloud displays and what is sent by the normal wifi module via serial. That way you can compare numbers.

brywithawhy, 2021-08-02:
Is there a way I can connect both at the same time?
Also is there an easy way I could make the 29th packet a sensor in HA so I can graph its data?
The problem is the comfort cloud app takes ages to display usage so if i could record it in HA and then compare with CC after a period of time that would help

DomiStyle, 2021-08-02:
Nope, needs to be done with a logic analyzer and the Panasonic module.

I can create a test branch with a sensor exposed for byte 29 once everything is merged into master.

brywithawhy, 2021-08-23:
I've just built an energy sensor and clipped it around my aircons power. After some further research into this and I think I have it figured out.

Byte 28 and 29 (zero indexed) are responsible for energy calculations.

Byte 28 is a temperature value
Byte 29 is how many times the temperature exceeds 255 watts

With my aircon at idle, it burns around 38 watts, and bytes 28 and 29 are [26 00]

I powered up my air con on heat and powerful mode and watch as the power usage ramps up.
251 Watts recorded as E0 00
497 Watts recorded as D7 01
873 Watts recorded as 56 03

These hex values don't completely agree with what the CT sensor recorded but they were within around 20-30 watts which I put down to noise/error on the CT clamp I'm using as no load jumps around from 5 to 15 watts.

Keen for others to try and hopefully implement this as a sensor!

blacklizard, 2021-08-29:
@brywithawhy 

i just tested your suggestion by using byte 28 and 29, but with that its showing extra 30W if compared with my ct clamp. I noticed that byte 30 also changes, so this is what i did, by doing this, the output is around +-4W if compared with CT clamp. byte_30 is some sort of offset, i cant be sure

`(byte_28 + (byte_29*255)) - byte_30 `

Could you try this?

blacklizard, 2021-08-29:
@brywithawhy 

I've implement the calculation and exposed it as a sensor. It's available in my fork.
https://github.com/blacklizard/esphome-panasonic-ac/tree/experimental/power_consumption 

You just need to enable it in the config file under `climate`. Please test

```yaml
climate:
  - platform: panasonic_ac 
    ...
    current_power_consumption:
      name: Panasonic AC Power Consumption
    ...
```

brywithawhy, 2021-08-29:
Hi @blacklizard,
Thanks for looking into that. I need to get a more accurate CT sensor to test this properly. The one I have is a home made kit I use to send data to HA and it isn't accurate enough yet, reading about 20 watts with no load.
I've ordered a clamp meter and will report back in a few days unless someone else can test in the mean time.

DomiStyle, 2021-08-31:
I installed your branch on my unit and it shows 9W idle and 42W in fan only mode. Seems a tad bit high for fan only but I didn't measure yet.

blacklizard, 2021-08-31:
@DomiStyle Mine was 6W idle, but it is inconsistence. As of now, based on my tested the error rate is not linear, using my multimeter and clamp meter, the error rate jumps between -+ 4W to +-20W. Currently I'm quite caught up with work, will measure again when I have the time

blacklizard, 2021-09-09:
@DomiStyle @brywithawhy 

I've spend past days to figure out the value, but it seems the values return by the AC is just an estimation. Sometimes the returned value has error up to 50% at day time if compared to night time. I validated this against reading from SDM230 meter. To monitor the usage, Its just wise to use a proper kwh meter.

I also came across this.
[image]

But, the estimation is close enough to measurement from SDM with variation of +-50W. Here is the comparison graph. Blue is from AC, orange is from SDM

[image]

brywithawhy, 2021-09-09:
@blacklizard that is pretty darn close. For me, I want to create automations when power usage is high so this is accurate enough for my liking,.

I haven't tested your branch yet as I'm still waiting for my meter to arrive. We are in lockdown here with so many delays so who knows when.

brywithawhy, 2021-09-29:
Finally received the clamp meter today and installed @blacklizard 's fork.
Can confirm the data is relatively close, within 10-20 watts. I put this down to the power factor and voltage fluctuations under load. On Powerful heat mode the sensor reported 864w. I was drawing 4.04 amps off the clamp meter. Another sensor that reports voltage showed 236vac at the time. With a power factor of say 0.9 for an air con, this gives the following:

Air con sensor 864W / 236v / 0.9 = 4.06A (0.02A difference)
236v * 4.04A * 0.9 = 858W (6W difference)

There's also minute differences between two identical ac units
[image]

As we have a silly demand usage tariff which charges you an extra daily rate against your highest half hourly usage, my use case is to have automations avoid the concurrent use of high energy devices. This sensor is very much accurate enough for me.

DomiStyle, 2021-10-02:
@blacklizard Feel free to put in a pull request if you think this feature is ready.

blacklizard, 2021-10-03:
@DomiStyle created the PR. Please check

DomiStyle, 2021-10-04:
@blacklizard Thanks!

Closed via https://github.com/DomiStyle/esphome-panasonic-ac/pull/29

closed 2021-10-04 by DomiStyle