DomiStyle/esphome-panasonic-ac #10: Swing mode

sygys, 2021-07-31:
Just installed the mod like in your tutorial. I can start and stop the AC and set the fan speed. But the angle of the fan cant be set. The lid is almost closed when on full speed. making the AC blow to the ceiling. Is this a software problem?

This is the esp version:
[image]

And it should be like the cloudcomfort integration like this:
[image]

sygys, 2021-07-31:
the swing mode only have automatic ones and no fixed positions also the automatic ones go to far in which it closes the lit completely which doesnt seem like a good idea while the fan spins at max speed and the air cant go out.

DomiStyle, 2021-07-31:
The fixed position can be added to Home Assistant as seen [here](https://github.com/DomiStyle/esphome-panasonic-ac/tree/dev-cztacg1#adding-manual-swing-selection-to-home-assistant).
They are not present in the climate entity itself because it's not supported (yet).

I don't think I have any control over how far the auto mode closes the lid. Does the auto mode from the IR remote do something different?

sygys, 2021-07-31:
Your goddamn awsome! Thats for sure! its works absolutely fantastic. And best of all... SUPER FAST!

sygys, 2021-07-31:
And for the rest. the auto modes are not very interesting for me. I never use them. And now that i have control over the swingmode manually i dont care at all. This is fantastic how it is now.

Boost and eco modes are those difuse and focus options?

DomiStyle, 2021-07-31:
Are you using master or dev branch? Which wifi module/connector?

sygys, 2021-07-31:
i use  the DNSK-P11 

Not sure what you mean by master or dev branch

DomiStyle, 2021-07-31:
There is a newer version of this library here: https://github.com/DomiStyle/esphome-panasonic-ac/tree/dev-cztacg1

It uses fan speed 1-5 and automatic.  Powerful and quiet are available as presets there. Eco mode is not supported by the DNSK-P11 but will be soon.

Currently it needs the latest RC of Home Assistant or fan speeds won't show up.

sygys, 2021-07-31:
So i put these files over the previous ones in home assistant, compile it and upload it to the esp32 and this will work?

DomiStyle, 2021-07-31:
Ideally you delete the old one and copy over the new one. Also, you need to make some adjustments to your yaml file, but there are instructions for that in the readme.

sygys, 2021-07-31:
Thanks for all the work done. You cant imagine how glad i am you made all this. Truly local AC control is amazing.

sygys, 2021-07-31:
Just upgraded to the latest version but it seems im missing most of the fan speeds now. and powerfull and normal are one and the same speed. both are the powerful option.

Also i wonder what eco does. When using the panasonic comfort cloud app in the past it set the fan to the lowest and cooled on a minimum. When i turn the switch in HA i dont see or hear anything happening. It could be that the fan speed isnt affected anymore by the eco preset, or it isnt working for me atm. Not really a way to test if its working right now.

DomiStyle, 2021-07-31:
It should have 5 fan speeds + Auto with the latest HA release candidate. With HA 2021.7 and below the fan speed selector is missing.

They only work when none of the presets (powerful/quiet) are enabled.

Eco mode is not supported yet.

sygys, 2021-07-31:
Upgrading for 2021.7.2 to 2021.7.4 now... maybe this will fix it. i dont see any fan speeds

DomiStyle, 2021-07-31:
You need to update to the release candidate/beta, which is 2021.8.0b5 right now or wait until it makes its way into the stable branch for Home Assistant (which should be soon).

sygys, 2021-07-31:
Ok thanks. then i will wait until the new release.

9th of august my second AC from panasonic is comming in the livingroom.  :)

DomiStyle, 2021-07-31:
If you have some time you could help with getting eco mode supported since my unit doesn't have eco mode.

* Enable verbose logging
* Enable eco mode
* Wait 1 minute
* Disable eco mode
* Wait 1 minute
* Enable eco mode
* Wait 1 minute

Then post the logs here.

sygys, 2021-07-31:
Offcourse i would do this for you. But you need to help me. I have no idea what verbose logging is

DomiStyle, 2021-07-31:
Yep, from your remote.

Just replace the line

> level: DEBUG

with

> level: VERBOSE

Reflash and you should now see more details in your ESPHome logs.

Can you post a picture of your remote? It should either have eco mode or nanoeX mode afaik.

sygys, 2021-07-31:
Ok please give me a moment i will do all this. My unit does not support nanoeX

sygys, 2021-07-31:
By the way my remote also shows quiet. So quiet is eco... that why when setting your climate to quiet it looks like eco. And the powerfull option in your esp is boost or powerful in mine as well. I can log this but im 99% sure that your presets aply to mine as well

sygys, 2021-07-31:
[image]

DomiStyle, 2021-07-31:
Yes, it looks like yours supports neither eco nor nanoeX. So the presets should be sufficient for you once the next HA version releases.

There's a strange terminology issue going on at Panasonic where they don't know what their modes are called either.

sygys, 2021-07-31:
[20:36:28][V][esppac:226]: Polling AC
[20:36:28][V][esppac:272]: TX: 5A 11 10 09 00 38 01 01 30 01 11 00 80 00 00 B0 00 02 31 00 00 A0 00 00 A1 00 00 A5 00 00 A4 00 00 B2 00 02 35 00 02 33 00 02 34 00 02 32 00 00 BB 00 00 BE 00 02 20 00 02 21 00 00 86 00 47
[20:36:28][V][esppac:272]: RX: 5A 11 10 89 00 76 00 01 30 01 11 00 80 01 30 00 B0 01 42 02 31 01 22 00 A0 01 41 00 A1 01 42 00 A5 01 43 00 A4 01 42 00 B2 01 43 02 35 01 41 02 33 01 41 02 34 01 41 02 32 01 42 00 BB 01 14 00 BE 01 14 02 20 01 41 02 21 01 41 00 86 2E 2A 00 00 0B 01 01 48 30 30 30 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 AF
[20:36:28][D][esppac:495]: Received query response
[20:36:28][V][sensor:035]: 'ac01_outside_temperature': Received new state 20.000000
[20:36:28][D][sensor:127]: 'ac01_outside_temperature': Sending state 20.00000 °C with 0 decimals of accuracy
[20:36:28][D][climate:369]: 'ac01' - Sending state:
[20:36:28][D][climate:372]:   Mode: COOL
[20:36:28][D][climate:380]:   Custom Fan Mode: Automatic
[20:36:28][D][climate:386]:   Custom Preset: Quiet
[20:36:28][D][climate:389]:   Swing Mode: OFF
[20:36:28][D][climate:392]:   Current Temperature: 20.00°C
[20:36:28][D][climate:398]:   Target Temperature: 17.00°C

sygys, 2021-07-31:
this is what i get when i turn on quiet on the AC

DomiStyle, 2021-07-31:
Thanks, will check out tomorrow if there is anything other than normal quiet mode set.

Can you post the error?

sygys, 2021-07-31:
wait its working now... i guess i didnt look good enough.

sygys, 2021-07-31:
[20:42:50][V][esppac:226]: Polling AC
[20:42:50][V][esppac:272]: TX: 5A 1D 10 09 00 38 01 01 30 01 11 00 80 00 00 B0 00 02 31 00 00 A0 00 00 A1 00 00 A5 00 00 A4 00 00 B2 00 02 35 00 02 33 00 02 34 00 02 32 00 00 BB 00 00 BE 00 02 20 00 02 21 00 00 86 00 3B
[20:42:50][V][esppac:272]: RX: 5A 1D 10 89 00 76 00 01 30 01 11 00 80 01 30 00 B0 01 42 02 31 01 22 00 A0 01 41 00 A1 01 42 00 A5 01 43 00 A4 01 42 00 B2 01 42 02 35 01 41 02 33 01 41 02 34 01 41 02 32 01 42 00 BB 01 14 00 BE 01 14 02 20 01 41 02 21 01 41 00 86 2E 2A 00 00 0B 01 01 48 30 30 30 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 A4
[20:42:50][D][esppac:495]: Received query response
[20:42:50][D][climate:369]: 'ac01' - Sending state:
[20:42:50][D][climate:372]:   Mode: COOL
[20:42:50][D][climate:380]:   Custom Fan Mode: Automatic
[20:42:50][D][climate:386]:   Custom Preset: Powerful
[20:42:50][D][climate:389]:   Swing Mode: OFF
[20:42:50][D][climate:392]:   Current Temperature: 20.00°C
[20:42:50][D][climate:398]:   Target Temperature: 17.00°C

this is powerful

sygys, 2021-07-31:
[20:43:21][V][esppac:272]: RX: 5A 1D 10 0A 00 09 00 01 30 01 01 00 A0 01 35 5D
[20:43:21][V][esppac:540]: Received report
[20:43:21][V][esppac:272]: TX: 5A 1D 10 8A 00 04 00 01 30 01 B9
[20:43:21][V][esppac:588]: Received fan speed
[20:43:21][V][esppac:591]: Received preset
[20:43:21][W][esppac:391]: Received unknown fan power
[20:43:21][D][climate:369]: 'ac01' - Sending state:
[20:43:21][D][climate:372]:   Mode: COOL
[20:43:21][D][climate:380]:   Custom Fan Mode: 4
[20:43:21][D][climate:386]:   Custom Preset: Normal
[20:43:21][D][climate:389]:   Swing Mode: OFF
[20:43:21][D][climate:392]:   Current Temperature: 20.00°C
[20:43:21][D][climate:398]:   Target Temperature: 17.00°C

This is normal with 4 bars on fan power. which gives a yellow warning: Received unknown fan power

sygys, 2021-07-31:
After that no errors while polling:

[20:45:21][V][esppac:226]: Polling AC
[20:45:21][V][esppac:272]: TX: 5A 21 10 09 00 38 01 01 30 01 11 00 80 00 00 B0 00 02 31 00 00 A0 00 00 A1 00 00 A5 00 00 A4 00 00 B2 00 02 35 00 02 33 00 02 34 00 02 32 00 00 BB 00 00 BE 00 02 20 00 02 21 00 00 86 00 37
[20:45:22][V][esppac:272]: RX: 5A 21 10 89 00 76 00 01 30 01 11 00 80 01 30 00 B0 01 42 02 31 01 22 00 A0 01 35 00 A1 01 42 00 A5 01 43 00 A4 01 42 00 B2 01 41 02 35 01 41 02 33 01 41 02 34 01 41 02 32 01 42 00 BB 01 12 00 BE 01 14 02 20 01 41 02 21 01 41 00 86 2E 2A 00 00 0B 01 01 48 30 30 30 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 AF
[20:45:22][D][esppac:495]: Received query response
[20:45:22][D][climate:369]: 'ac01' - Sending state:
[20:45:22][D][climate:372]:   Mode: COOL
[20:45:22][D][climate:380]:   Custom Fan Mode: 4
[20:45:22][D][climate:386]:   Custom Preset: Normal
[20:45:22][D][climate:389]:   Swing Mode: OFF
[20:45:22][D][climate:392]:   Current Temperature: 18.00°C
[20:45:22][D][climate:398]:   Target Temperature: 17.00°C

sygys, 2021-07-31:
DomiStyle do you plan to add the swingmodes (angle of the blade) to the climate entity? If i understand correctly this isnt yet supported by ESPhome? But will it be in the future?

DomiStyle, 2021-07-31:
There is a pull request active at https://github.com/esphome/esphome/pull/1968 for adding this feature.
However, there are is no API for this in Home Assistant yet so it probably will take some time.

Of course, I will implement it once it's available.

sygys, 2021-07-31:
That would be great. because as it is right now you cant put a swingmode change and a preset change after eachother in an automation. The preset will be skipped. i think because the esp is still dealing with the swingmode change when the preset change is fired. If i dont put in a delay of a few seconds the preset isnt changed

closed 2021-08-18 by DomiStyle

Driekes, 2022-06-17:
@DomiStyle I just installed this, but I do have swing modes Auto / Up / UpMid / Mid / DownMid / Down in the original Pana integration.

In your software I have: off / both / vertical / horizontal. 

If I change it in the panasonic app, I see these changes in your software as follows:

Auto = vertical
The rest = OFF. Any way to map them correctly? I see the issue on ESPHOME#1968 linked above, but it is closed.

DomiStyle, 2022-06-17:
@Driekes The manual swing modes are controlled via the select element, not on the AC entity in Home Assistant. The selection you mentioned is for the automatic modes.

Driekes, 2022-06-17:
@DomiStyle I now see there is a seperate select. entity in HA that I can use to control the swing. I missed that.

The default panasonic allows me to control this under the climate entity, so I just have to adjust the dashboard a bit. Sorry I missed this!

I guess there is no way, or at least not without forking your code to merge this into the climate entity again? I only have vertical swing mode anyway on this model.

Driekes, 2022-06-17:
@DomiStyle thanks for the info. I dug a bit deeper and indeed understand the linked issue better (had to do with IR setting rather than the climate component. And indeed there seems no option (yet) to override this behaviour in esphome.