DomiStyle/esphome-panasonic-ac #11: Ac going through 3 states when automation is running

sygys, 2021-08-01:
I have an automation to start the ac on dry when the humidity is high in the room. But for some reason when that hits the state of the ac is changed multiple.times in a second. In which it chooses heat_cool most of the time

[image]

I'm not sure what goes wrong

sygys, 2021-08-01:
So i understand the first 2... When its turned on it uses its last state. but i dont understand why it then uses heat/cool i think it has something to do with the swing modus. When this runs to fast after a state change the code in the esp/ac is messed up and it uses heat/cool.

This is the automation that ran:

`      - service: climate.turn_on
        target:
          entity_id: climate.airco_slaapkamer_esp
      - service: climate.set_temperature
        data:
          hvac_mode: dry
          temperature: 16
        target:
          entity_id: climate.airco_slaapkamer_esp
      - service: input_select.select_option
        target:
          entity_id: input_select.airco_status
        data:
          option: ontvochtigen
      - service: telegram_bot.send_message
        data:
          title: '*Ter info*'
          message: 'Airco op ontvochtigen'
      - service: input_select.select_option
        target:
          entity_id: input_select.ac01_vertical_swing
        data_template:
          option: "{{states('input_select.airco_slaapkamer_swingmode')}}"`

DomiStyle, 2021-08-01:
Can you post the verbose log when the automation runs?

sygys, 2021-08-01:
Ill have to run this manually because this is part of a 4 way choose automation that triggers when the humidity is above a certain value. I think i can let it run by breathing into the humidity sensor.

sygys, 2021-08-01:
[18:24:44][D][esppac:495]: Received query response
[18:24:44][W][esppac:100]: Received out of range outside temperature
[18:24:44][D][climate:369]: 'ac01' - Sending state:
[18:24:44][D][climate:372]:   Mode: OFF
[18:24:44][D][climate:380]:   Custom Fan Mode: 4
[18:24:44][D][climate:386]:   Custom Preset: Normal
[18:24:44][D][climate:389]:   Swing Mode: OFF
[18:24:44][D][climate:392]:   Current Temperature: 18.00°C
[18:24:44][D][climate:398]:   Target Temperature: 16.00°C
[18:25:00][D][climate:009]: 'ac01' - Setting
[18:25:00][D][climate:013]:   Mode: HEAT_COOL
[18:25:00][V][esppac:012]: AC control request
[18:25:00][V][esppac:073]: Requested mode change
[18:25:00][D][climate:369]: 'ac01' - Sending state:
[18:25:00][D][climate:372]:   Mode: HEAT_COOL
[18:25:00][D][climate:380]:   Custom Fan Mode: 4
[18:25:00][D][climate:386]:   Custom Preset: Normal
[18:25:00][D][climate:389]:   Swing Mode: OFF
[18:25:00][D][climate:392]:   Current Temperature: 18.00°C
[18:25:00][D][climate:398]:   Target Temperature: 16.00°C
[18:25:00][V][esppac:272]: TX: 5A 85 10 08 00 0D 01 01 30 01 02 00 B0 01 41 00 80 01 30 24
[18:25:00][D][climate:009]: 'ac01' - Setting
[18:25:00][D][climate:013]:   Mode: DRY
[18:25:00][D][climate:038]:   Target Temperature: 16.00
[18:25:00][V][esppac:012]: AC control request
[18:25:00][V][esppac:073]: Requested mode change
[18:25:00][D][climate:369]: 'ac01' - Sending state:
[18:25:00][D][climate:372]:   Mode: DRY
[18:25:00][D][climate:380]:   Custom Fan Mode: 4
[18:25:00][D][climate:386]:   Custom Preset: Normal
[18:25:00][D][climate:389]:   Swing Mode: OFF
[18:25:00][D][climate:392]:   Current Temperature: 18.00°C
[18:25:00][D][climate:398]:   Target Temperature: 16.00°C
[18:25:00][V][esppac:111]: Requested temperature change
[18:25:00][V][esppac:272]: TX: 5A 86 10 08 00 11 01 01 30 01 03 00 B0 01 44 00 80 01 30 00 31 01 20 C9
[18:25:00][V][esppac:272]: RX: 5A 85 10 88 00 0B 00 01 30 01 02 00 B0 00 00 80 00 1A
[18:25:00][W][esppac:289]: Correcting shifted tx counter
[18:25:00][V][esppac:536]: Received command ack
[18:25:00][V][esppac:272]: RX: 5A 4A 10 0A 00 0D 00 01 30 01 02 00 80 01 30 00 B0 01 41 5E
[18:25:00][W][esppac:297]: Correcting shifted rx counter
[18:25:00][V][esppac:540]: Received report
[18:25:00][V][esppac:272]: TX: 5A 4A 10 8A 00 04 00 01 30 01 8C
[18:25:00][V][esppac:568]: Received power mode on
[18:25:00][D][climate:369]: 'ac01' - Sending state:
[18:25:00][D][climate:372]:   Mode: HEAT_COOL
[18:25:00][D][climate:380]:   Custom Fan Mode: 4
[18:25:00][D][climate:386]:   Custom Preset: Normal
[18:25:00][D][climate:389]:   Swing Mode: OFF
[18:25:00][D][climate:392]:   Current Temperature: 18.00°C
[18:25:00][D][climate:398]:   Target Temperature: 16.00°C

sygys, 2021-08-01:
im almost certain that when i remove:

 - service: input_select.select_option
    target:
      entity_id: input_select.ac01_vertical_swing
    data_template:
      option: up

From the automation this isnt happening but i have to check.

sygys, 2021-08-01:
This is the log with the one removed:

[18:30:00][V][esppac:111]: Requested temperature change
[18:30:00][V][esppac:272]: TX: 5A 92 10 08 00 11 01 01 30 01 03 00 B0 01 44 00 80 01 30 00 31 01 20 BD
[18:30:00][V][esppac:272]: RX: 5A 91 10 88 00 0B 00 01 30 01 02 00 B0 00 00 80 00 0E
[18:30:00][W][esppac:289]: Correcting shifted tx counter
[18:30:00][V][esppac:536]: Received command ack
[18:30:00][V][esppac:272]: RX: 5A 4C 10 0A 00 09 00 01 30 01 01 00 80 01 30 53
[18:30:00][W][esppac:297]: Correcting shifted rx counter
[18:30:00][V][esppac:540]: Received report
[18:30:00][V][esppac:272]: TX: 5A 4C 10 8A 00 04 00 01 30 01 8A
[18:30:00][V][esppac:568]: Received power mode on
[18:30:00][D][climate:369]: 'ac01' - Sending state:
[18:30:00][D][climate:372]:   Mode: DRY
[18:30:00][D][climate:380]:   Custom Fan Mode: 4
[18:30:00][D][climate:386]:   Custom Preset: Normal
[18:30:00][D][climate:389]:   Swing Mode: OFF
[18:30:00][D][climate:392]:   Current Temperature: 17.00°C
[18:30:00][D][climate:398]:   Target Temperature: 16.00°C

And it goes to dry like it should. the swingmodus is messing things up when to close to the rest of the automation. When putting in a delay of 5sec this wont happen.

sygys, 2021-08-01:
Wait eventually it still went back to heat/cool out of itself. the webpage reloaded so the log was gone when this happened.

jmichiel, 2023-02-15:
I see something similar happening on my setup: switching between HEAT and HEAT_COOL seemingly by itself...
I do have an automation (via Node-RED) that will automatically tunr off the A/C when the room temperature is more than 0.5° above the target temperature. Seems to happen on those moments. I'll investigate further!

sygys, 2023-02-15:
You can fix this by sending one command at a time and put in delays between them.