DomiStyle/esphome-panasonic-ac #12: Pre-installed units

sygys, 2021-08-02:
Im planning on making a few pre-installed wifi controllers. I designed a circuitboard with the size of the ESP in which the esp32 is placed on the board and the logic converter underneath. Im currently in contact with my Panasonic AC mechanic and asked him if he can sell me the cable from the AC motherboard to the wifi module. If he can provide those for me i can make a preinstalled circuitboard with a 3d printed case and a cable that can be pressed into the motherboard of the AC. Without soldering. The design of the circuitboard is so compact that it can be housed in a case that is as big as the standard wifi module. So it can be mounted the same way.

If people are interested in such a plug and play module please let me know. Ill build 2 for my self as a test and if its a success i could make more.

HA-TB303, 2021-08-02:
Been there, done that :)
[image]
[image]

HA-TB303, 2021-08-02:
[image]

HA-TB303, 2021-08-02:
[image]

DomiStyle, 2021-08-02:
That's pretty cool. Maybe you guys can do a combo one with both connectors onboard? I would order a few of those too.

Could probably add a DIP switch as well to switch between protocols.

HA-TB303, 2021-08-02:
There is not really enough space for both connectors.
I wanted this to be a 100% drop in replacement for the original PCB.

HA-TB303, 2021-08-02:
I'm always open for suggestions by the way,. This is a first revision and it's a near perfect fit for the original enclosure.

DomiStyle, 2021-08-02:
Are the files for the PCB open source? Maybe it can be done with an adapter cable or 2 variants.

HA-TB303, 2021-08-02:
I'm afraid not. There was a lot of work for Paul and me and we would like to sell these with a few $ margin to recuperate the cost.

All of our through hole designs are open source, but this is SMD and that makes it hard to source and solder yourself, so I just order them assembled, except for the pin header.

HA-TB303, 2021-08-02:
We're Open to designing a different version for the different header, as long as I can test it on my own Panasonic airconditioning unit.

DomiStyle, 2021-08-02:
I think I can see the CN-CNT connector above your wifi module so you should be able to test just fine.

You can even connect both at the same time.

HA-TB303, 2021-08-02:
Can you explain te difference? Why has my unit have both options and are there different panasonic units that only have CN-CNT?

I do not have the CN-CNT cable, and therefor not the connector..... :(
What is the pin spacing on the CN-CNT pcb connector? Getting the original connectors seems impossible, so it's probably just going to be a "naked" header like in my design anyway.

Are you aiming at people that have the Panasonic airconditioning unit, but no wifi module? Then those people would need a proper cable as well. That is going to be difficult.

I already had the DNSK-P11 wifi modules, but just wanted to get rid of the cloud ;)

DomiStyle, 2021-08-02:
> Can you explain te difference? Why has my unit have both options and are there different panasonic units that only have CN-CNT?

CN-CNT is the older connector that (almost) every Panasonic AC and many other Panasonic products have. CN-WLAN (the one the DNSK-P11 uses) is newer but not present on every AC.
Also, CN-CNT is reachable without disassembly while CN-WLAN is only reachable with disassembly on many units.

> I do not have the CN-CNT cable, and therefor not the connector..... :(

I assembled the cable for the CN-CNT connector by myself since they are available from RS-Online and Conrad:
https://github.com/Egyras/HeishaMon#where-to-get-connectors

There are also some PCB designs linked there.

> Getting the original connectors seems impossible, so it's probably just going to be a "naked" header like in my design anyway.

I see, so you desoldered the connector from your DNSK-P11?

> Then those people would need a proper cable as well. That is going to be difficult.

See above, building the CN-CNT cables is easy.

> I already had the DNSK-P11 wifi modules, but just wanted to get rid of the cloud ;)

No worries, the older CZ-TACG1 module wasn't even supported until recently. The DNSK-P11 has a much more complicated but newer (and better) protocol.

Here's some comparison shots (ignore the wire mess):

[image]
[image]
[image]
[image]

sygys, 2021-08-03:
@HA-TB303 thanks for sharing this. I wonder why this was never posted here before because this could help allot of people. For example I would really like to know what print connector you used to connect the connector of the ac to the print. It's smaller then the ones I have

sygys, 2021-08-03:
@DomiStyle is it correct that you soldered wires to the back of the DNSK-P11 in which you can use both the official as the esp32 for sending commands

HA-TB303, 2021-08-03:
> @HA-TB303 thanks for sharing this. I wonder why this was never posted here before because this could help allot of people. For example I would really like to know what print connector you used to connect the connector of the ac to the print. It's smaller then the ones I have. Did you design the board yourself and let it make by a PCB company?

Well, I literally just got this first iteration in from the mail today , so any sooner was really not possible ;)

I did not use a connector, I used a simple header as you can see in the pictures.

DomiStyle, 2021-08-03:
> is it correct that you soldered wires to the back of the DNSK-P11 in which you can use both the official as the esp32 for sending commands? Or are these for reverse engineering the code

It's for the logic analyzer to reverse engineer the protocol. I just have two ESPs, one connected to CN-CNT and one connected to CN-WLAN for debugging.

> I did not use a connector, I used a simple header as you can see in the pictures.

Ah, now I see it. Shame that you couldn't find the connector they used for the CN-WLAN.

sygys, 2021-08-03:
I understand that you are not gonna share the print layout. But could you provide me a link the the right headers? The normal ones that I have do not fit the connector. Thanks!

HA-TB303, 2021-08-03:
It's just a simple 1.27mm pitch header: https://s.click.aliexpress.com/e/_9HEwKi

> Ah, now I see it. Shame that you couldn't find the connector they used for the CN-WLAN.

The header works absolutely fine.  No requirement for using the propriety Panasonic header.

HA-TB303, 2021-08-03:
Hopefully today we'll post a small article about this design on our website www.espthings.io. I have 12 to maximum 18 modules available and I only need 2 myself.

I am willing to sell with a small margin + post en packaging of your choice from Belgium or the Netherlands, just to get recuperate the expenses we made. No profit intended!

DomiStyle, 2021-08-03:
Looking forward to the article. I would take 4 if they're not too expensive.

HA-TB303, 2021-08-03:
I'll put them aside for you for now, no strings...

HA-TB303, 2021-08-03:
https://www.espthings.io/index.php/2021/08/03/esp32-based-panasonic-airconditioning-wifi-adapter/?preview_id=1004&preview_nonce=6177296814&post_format=standard&_thumbnail_id=-1&preview=true

sygys, 2021-08-03:
Nice article. Where are you from. I'm from the Netherlands living near Eindhoven. Maybe I could pick one up.

HA-TB303, 2021-08-04:
> Nice article. Where are you from. I'm from the Netherlands living near Eindhoven. Maybe I could pick one up.
> 

Thanks, contact me via the contact form. I live fairly nearby (25-30 minute drive) but I don't want to address public ;)

> In a few days my circuit board will be ready. It's not as advanced as yours. I designed it so I could put an esp32 and a logic converter on it in which the board connects the right pins to the right components like this
> [image]
> 

> It will be as big as an esp32 which fits in the housing but I will have to wait 2 months before the 1.27mm connector pins arrive from china lol. A friend of mine is making these circuitboards himself
>

Great work!
 
> By the way not to be an ass but you should give a big discount to Dominik. He is the one who made all this possible ;)

Selling at cost is no problem, but I won't be giving them away. All the time and effort that went into this is fine and a hobby, just like it is for Dominik.

sygys, 2021-08-09:
@HA-TB303 I send you a message on your website. It took a while could you please confirm you received it? Thanks

sygys, 2021-08-11:
I made a case for the unit HA-TB303 is selling. It's a perfect fit
[image]
[image]

Still working on the logo though. And want to buy some blue filament to make it the color of HA just for fun

HA-TB303, 2021-08-13:
Great work!

daggy72, 2021-08-18:
great work!

HA-TB303, 2021-10-03:
All 18 modules left are sold. I won't be making another batch soon, unless there are buyers for another 20 pcs, so you're welcome the reserve one (or more), but I can't guaranty I will in fact order another batch.  https://www.espthings.io/p11

DomiStyle, 2021-10-04:
> Can I just plug in a esphone-panasonic-ac device into the CN-CNT and keep the DNSK-P11 unit or is it only possible to connect to one or the other at a time?

Yes, of course the official unit will still connect to the cloud so you might want to remove it if you care about that.

> Are there any advantages or disadvantes to using the CN-CNT vs. CN-WLAN?

There are some features missing on the CN-WLAN side of the protocol but it depends on what your AC supports.

HA-TB303, 2021-11-20:
Hi, as we have some people interested we have ordered another small batch. If anybody is interested in our DROP-IN DNSK-P11 replacement module, please read our article and let us know. https://espthings.io/p11
[image]

sygys, 2021-11-20:
I want to reserve 2 more for replacement when another fails. you guys ever considered using an ESP module with wifi antenna?

this would give the units a far better wifi  connection.  I use an ESP32 with wifi antenna and the reception becomes 50% better then the regular ones.

HA-TB303, 2021-11-20:
Hi,

Mine is located quit far away from the access-point and I have not experienced any problems with the reception. As I have had no complains at all about reception, there are no plans to change the current design.

Cheers,

Michel

sygys, 2021-11-20:
I dont have reception problems with the 2 im using. They are pretty solid. I dont really mind to have thicker PCB because i designed my own enclosure in which they easily fit. But i have been spending too much money this month on smart home stuff and my girlfriend is going crazy if i spend more. So i have to wait for my allowance at the end of the month :P

HA-TB303, 2021-11-20:
Lol, I've been having that issue quit a lot as well :P

sygys, 2021-11-20:
I read that there will be coming zigbee esp's next year i would really love to use these in the AC's. My zigbee network is allot more stable then wifi.

HA-TB303, 2021-11-20:
We will be looking into that as well as I am a Zigbee fan myself, but then Zigbee2Mqtt, ESPhome or ZHA has to support the ESP32-H2 and create a Panasonic Airco module. 

I don't see that happening soon, but If it happens, we will be making a module for it. I also wonder IF ESPhome is planning to support Zigbee in the future at all... I certainly hope so.

We are also currently looking into a version for the CZ-TACG1. Either a 'one for all' or a separate module.

Cheers,

Michel

HA-TB303, 2022-02-08:
The second bath is sold and we are no longer offering these interfaces.

sygys, 2022-06-10:
Our MailScanner believes that the attachment to this message sent to you
   
    From: ***@***.***
 Subject: Re: [DomiStyle/esphome-panasonic-ac] Pre-installed units (#12)

is Unsolicited Commercial Email (spam). Unless you are sure that this message
is incorrectly thought to be spam, please delete this message without opening
it. Opening spam messages might allow the spammer to verify your email
address.

If you believe that this message has been incorrectly marked as spam, please
forward this email to postmaster.

Date: 20220610
 pts rule name              description
---- ---------------------- --------------------------------------------------
-0.0 NO_RELAYS              Informational: message was not relayed via SMTP
 0.0 HTML_MESSAGE           BODY: HTML included in message
 1.4 HTML_IMAGE_ONLY_28     BODY: HTML: images with 2400-2800 bytes of
                            words
-1.9 BAYES_00               BODY: Bayes spam probability is 0 to 1%
                            [score: 0.0000]
 0.7 MIME_HTML_ONLY         BODY: Message only has text/html MIME parts
 1.9 MISSING_MIMEOLE        Message has X-MSMail-Priority, but no X-MimeOLE
 0.4 HTML_MIME_NO_HTML_TAG  HTML-only message, but there is no HTML
                            tag
-0.0 NO_RECEIVED            Informational: message has no Received headers

HA-TB303, 2022-06-10:
I'm not sure why you would want to share your mailserver/spamfilter issues here? Your spamfilter is obviously wrong. ;)

sygys, 2022-06-10:
No I thought I mailed you to order 2. But I mailed it to GitHub :)

HA-TB303, 2022-07-23:
The above version is no longer available.

We have a new version in the making that also includes IO for a DS18B20 1 witre temperature sensor like this:
[image]

And this is it:

[image]

If there is enough people interested, we could run another small batch.

Cheers,

Michel

stmps, 2025-01-31:
> If there is enough people interested, we could run another small batch.

Hey @HA-TB303 ! This is quite an old thread, but are you still producing these? :)

HA-TB303, 2025-01-31:
No, sorry. On our website is a DIY alternative.

HA-TB303, 2025-03-14:
We have just updated the DIY version to esp32-C3 Super Mini here: [https://www.ingeniousmakers.com/p11](https://www.ingeniousmakers.com/p11)