DomiStyle/esphome-panasonic-ac #14: Change/feature request (UI)

HA-TB303, 2021-08-04:
HI, I noticed the Panasonic (cloud based) integration and this one differ quit substantially.
It would be great if the integration could be more in line with the "official" integration.
Some functions are missing, some are substantially different:

Cloud based integration -- VS -- esphome integration:

Practically the same and all good:
[image]

No ECO mode, or just a different name?
[image]

No fan mode?
[image]

Positioning the fan is not possible? / my unit only has a vertical swing.
[image]

Separate switches and sensors, for stuff that might be able to live in the climate component? (I really don't know it that is possible.....
[image]

DomiStyle, 2021-08-04:
Please see the readme on the dev branch: https://github.com/DomiStyle/esphome-panasonic-ac/tree/dev-cztacg1
Things have changed quite a bit. All of the options are identical to the Comfort Cloud options but they might look slightly different depending on the exact model since Panasonic likes to shift stuff around.

> No ECO mode, or just a different name?

Eco mode is exposed as sensor, see https://github.com/DomiStyle/esphome-panasonic-ac/tree/dev-cztacg1#setting-supported-features

> No fan mode?

Fan modes require the dev branch, ESPHome 1.20.0+ and Home Assistant 2021.8.0+ (released today).

> Positioning the fan is not possible? / my unit only has a vertical swing.

See https://github.com/DomiStyle/esphome-panasonic-ac/tree/dev-cztacg1#adding-manual-swing-selection-to-home-assistant

They will be exposed as select soon instead of input_select which makes them auto discoverable in Home Assistant. However, this change has not made it into the latest ESPHome version so far.

> Separate switches and sensors, for stuff that might be able to live in the climate component? (I really don't know it that is possible.....

It's not possible to modify the ESPHome climate component, they have to be exposed as switches and sensors.

sygys, 2021-08-09:
I contacted esphome on the matter of the swing mode I left I understand correct esphome allready supports swing mode and this is up to the home assistant part of esp to support it. It would be really great of we had swing mode in a different way then a text sensor. This really is a pain when combining it with other commands in an automation. The ac is going all over the place when you change the swing mode and then put in another command just behind that one. Cool heat auto and dry all can happen then. I can imagine this could eventually even crash the AC.

closed 2021-08-18 by DomiStyle