DomiStyle/esphome-panasonic-ac #16: ESPHome external Components

jesserockz, 2021-08-04:
This is great and I hope someday that you decide to get this merged into ESPHome directly. Until then it would be great if you could structure the code as a standard component so that people can use it with [external_components](https://esphome.io/components/external_components.html)

This would allow people to update a lot easier as that is handled by ESPHome itself.

DomiStyle, 2021-08-05:
Sure thing! Will take a look after v2 is released.

Is there an API documentation for the Python interface like [there is for the C++ part](https://esphome.io/api/namespaces.html)? Is it possible to add all the sensor/switches/selects into one `climate` block or do I have to resort to individual `sensor`, `select` and `switch` blocks like I'm doing now?

I wrote this with https://github.com/geoffdavis/esphome-mitsubishiheatpump as reference, which has now also moved to `external_components`. They don't use any components except climate though.

jesserockz, 2021-08-05:
It is possible to include other types inside the climate yaml, and then you can check if the user specified that and ignore those parts in C++

DomiStyle, 2021-08-05:
Sounds good, do you know an existing component I can use as reference which does that?

jesserockz, 2021-08-05:
Not off the top of my head.
But basically the same way any other component has sub-components. Perhaps DHT11 is one of the simplest...

jesserockz, 2021-08-06:
Here you go @DomiStyle #17 :stuck_out_tongue_closed_eyes:

DomiStyle, 2021-08-09:
Closed via https://github.com/DomiStyle/esphome-panasonic-ac/pull/17

Thanks @jesserockz

PR #17 (2021-08-08): Rework code for external_components
I got bored and decided to do this...haha

Because I pretty much moves the files, and reformatted them, and changed quite a lot, its hard to get a real diff.

But this code is working for my CN-CNT based AC. 

I have updated the yaml example with how to use it. (The repo/branch in yaml will need to be updated if you decide to merge this in)

Closes #16
closed 2021-08-09 by DomiStyle