DomiStyle/esphome-panasonic-ac #18: Updates online

sygys, 2021-08-10:
I looked at the code of the example.yaml and came across a new line:

```
external_components:
  source: github://DomiStyle/esphome-panasonic-ac@dev-cztacg1
  components: [panasonic_ac]
```

Does this mean it will automatically download the latest files from github instead of needing to download the files and put them in the esphome directory?

DomiStyle, 2021-08-10:
Yes, the new component will auto update when compiling. Also all sensors are now exposed automatically.

sygys, 2021-08-15:
I changed the code of my yaml file to:

[image]

But as you can see i get an error. I cannot install the new component while this error presists. What am i missing?

sygys, 2021-08-15:
if i remove the space in front of the "type: wlan" line i get the error on that line saying "platform not found: climate.Panasonic_ac"

sygys, 2021-08-18:
Never mind old version of ESPhome....

closed 2021-08-18 by sygys