DomiStyle/esphome-panasonic-ac #21: Outside temperature outside range

sygys, 2021-08-18:
Hi,

I get the warning that the outside temperature is outside of range. When looking at the hex code receiving from the poll and comparing your data table provided in the docs i get the hex value of 7E. which when converted means the outside temperature is 126 degrees.

Im not really sure if this is a problem with my unit or with the integration. But both of my AC's have the exact same value of 7E

here is the code:
`[13:25:08][V][panasonic_ac:208]: TX: 5A.A8.10.09.00.38.01.01.30.01.11.00.80.00.00.B0.00.02.31.00.00.A0.00.00.A1.00.00.A5.00.00.A4.00.00.B2.00.02.35.00.02.33.00.02.34.00.02.32.00.00.BB.00.00.BE.00.02.20.00.02.21.00.00.86.00.B0 (63)`

`[13:25:08][V][panasonic_ac:210]: RX: 5A.A8.10.89.00.76.00.01.30.01.11.00.80.01.31.00.B0.01.42.02.31.01.22.00.A0.01.36.00.A1.01.42.00.A5.01.43.00.A4.01.44.00.B2.01.41.02.35.01.41.02.33.01.41.02.34.01.41.02.32.01.42.00.BB.01.13.00.BE.01.7E.02.20.01.41.02.21.01.41.00.86.2E.2A.00.00.0B.01.01.48.30.30.30.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.B9 (125)`

DomiStyle, 2021-08-18:
There is no calculation taking place for the temperature atm, all units I have seen so far are celsius and are 1:1.

Does your unit provide you with a value via the official Panasonic app? Not all units support outside temperature readings.
You could switch back the official adapter, take note of the temperature and plug back in the ESP. If I know what the app reports I can find the value in the packet.

sygys, 2021-08-18:
I have seen the outside temperature work with one of your earlier builds. Maybe the AC has an error in it. All hex data you provided in the docs is exactly the same like mine apart from a few values which are 41 instead of 42 which could be due to the difference in models.

Ill unplug the AC and see if the readings change. I hope i dont need to solder back the stock wifi module back on. i dont think its worth the effort of that.

sygys, 2021-08-18:
Ok so i know what happens. The AC only communicates the outside temperature when its on. I guess when its off it reports the max temp of the sensor.

closed 2021-08-18 by sygys