DomiStyle/esphome-panasonic-ac #25: Is 'nanoeX' the same as 'nanoe-G'?

dixonte, 2021-08-29:
Sorry if this should be obvious. I'm assuming the X might be a placeholder for a number of different options, but the possibility of causing damage to the unit has me hesitant to enable it.

DomiStyle, 2021-08-30:
No, nanoeX is a different feature. So far I haven't seen nanoe-G being reported via the wifi adapter so it's currently not possible to toggle it.

You can try enabling it via the IR remote and posting a verbose log with it on and off.

dixonte, 2021-08-30:
I'll see if I can do that once I actually get this thing installed. Might be a long wait for the right cable to arrive. Feel free to close this in the meantime.

blacklizard, 2021-09-09:
@dixonte Even from the official app, there is no option to toggle nanoe-G. The verbose log does not change when toggling  nanoe-G from IR. This is true for my AC(CS-S13TKH)

dixonte, 2021-09-10:
@blacklizard Good to know. Just in case I might do my own tests when I am finally able to build this thing, but thanks for the heads up.

closed 2023-01-31 by DomiStyle