DomiStyle/esphome-panasonic-ac #26: Compatibility with Panasonic  CZ-CAPWFC1 WiFi interface module ?

mr-sneezy, 2021-09-10:


uplore, 2021-09-10:
I came here to ask this exact question. I would love to disconnect comfort cloud from our central AC unit. Here is an image of the internals of the CZ-CAPWFC1
[image]

HA-TB303, 2021-09-10:
That bottom module looks identical to the DNSK-P11

DomiStyle, 2021-09-10:
Yep, that's it. You probably can replace the bottom module with an ESP. Make sure to measure voltages on the connector first to be sure. There's a pinout in the DNSK-P11 readme.

DomiStyle, 2021-10-13:
@vjmaz Take a look at the readme for the DNSK-P11: https://github.com/DomiStyle/esphome-panasonic-ac/blob/master/README.DNSKP11.md

There's a pinout there for the 8 pin connector.