DomiStyle/esphome-panasonic-ac #27: Compatible with heishamon board

IgorYbema, 2021-09-20:
Hi,
Someone came to me to ask if your project can work on my ready-made PCB which is used for Panasonic heatpumps (Aquaread H and J types):  https://www.tindie.com/products/thehognl/heishamon-communication-pcb/

I think it can work on the board. It has the right level shifters and I notice you even link to our github page for the CZ-TACG1 cable connectors so the cables should fit also.

But I am not an ESPhome expert (yet). And our software uses the hardware uart of the esp8266. ESPhome uses software serial I guess? We swap the serial port to gpio13 and gpio15 but that can also be done on software serial. But the only thing we do extra is using gpio5 to enable the level shifting after booting of the ESP (high = enable). This helps us to prevent noise data from the ESP during boot to the heatpump. So, if we use your panasonic AC ESPhome plugin on the heishamon board, the plugin should enable GPIO5 also on boot. Can you tell me how that can be inserted?

IgorYbema, 2021-09-20:
I have been checking the ESPhome documentation and believe that the changes below should work. Do you agree?
First I switch logging to UART1 (we do the same, only tx available but that is ok)
Then I put the uart for communcation on GPIO15/GPIO13. This should use the hardware uart if possible.
And last, I enable the GPIO5 on boot (which enables the levelshifting on heishamon board).

```
logger:
  level: DEBUG
  hardware_uart: UART1

uart:
  tx_pin: GPIO15
  rx_pin: GPIO13
  baud_rate: 9600
  parity: EVEN

output:
 - platform: gpio
   pin: 5
   id: enable_levelshifting
   on_boot:
     priority: 600
     then:
      - output.turn_on: enable_levelshifting

```

DomiStyle, 2021-09-20:
I believe the on_boot section only works in the esphome context. But the GPIO switch should work:

```
logger:
  level: DEBUG
  hardware_uart: UART1

uart:
  tx_pin: GPIO15
  rx_pin: GPIO13
  baud_rate: 9600
  parity: EVEN

switch:
  - platform: gpio
    pin: GPIO5
    id: enable_levelshifting
    restore_mode: ALWAYS_ON
    internal: true
```

I usually only test with ESP32s but plenty of people had success with ESP8266 so far so I don't see why your PCB shouldn't work as well.

If you have trouble with the config above you can also try disabling logging via serial altogether:

```
logger:
  level: DEBUG
  baud_rate: 0
```

gschmidl, 2021-09-21:
Some info about the 8266 seems to be here: https://github.com/DomiStyle/esphome-panasonic-ac/issues/5