DomiStyle/esphome-panasonic-ac #28: CN-NET?

ClasseM, 2021-10-03:
Dear all
Got some newer Panasonic AC's all which came with a CN-NET connector (to connect a CF-TA9 adapter). It's a 4 pin connector so I assume it is an UART connector (will measure shortly), however just wonder if anyone came across this connector and know if it is compatible with CN-CNT or CN-WLAN?
Thanks!

DomiStyle, 2021-10-03:
Can you take a picture?

ClasseM, 2021-10-05:
Hello Dominik

Thanks for looking into this. Please find picture below (from a site and not my unit however same).
[image]

Also below link refer to the installation of the CF-TA9 adapter on an AC showing the CN-NET connector.
https://kawagoe-setsubi.com/wireless-adapter/

Thanks!

DomiStyle, 2021-10-06:
Interesting, don't think I have ever seen that connector. My units don't have that connector.

You probably will have to measure the voltages and use a logic analyzer to check if the protocol is the same as for the CNT or WLAN.