DomiStyle/esphome-panasonic-ac #35: ESP32 crashes after extended use

dixonte, 2021-10-30:
This has happened maybe 4-5 times now in the last few days. The ESP32 will show as unavailable in Home Assistant and offline in MQTT Explorer until I reset it. It's never happened while I've been watching and I'm unsure how to debug the issue. Do I just need to leave the log going in case it happens?

Aside from enabling MQTT and a small interval that checks the current temperature and sets cooling mode if it goes over 30C, my YAML file is pretty standard. I'm happy to cut my modifications for testing purposes, though I am pretty sure this started happening before I made them.

DomiStyle, 2021-10-30:
Can you try disconnecting your AC and see if it still happens?

dixonte, 2021-10-30:
So connect it to USB power instead? Sure I can give that a go for a few days.

dixonte, 2021-11-02:
It happened again this morning, so I've disconnected the board from the AC and have it powered via a USB wall plug to see if it happens like this.

dixonte, 2021-11-08:
Well it's been nearly a week with the board getting power via a USB wall plug, and it's still showing as available in ESPHome. I'd say it only happens when connected to the AC.

I have eco, econavi and mild dry enabled. Should I try disabling them and reconnecting it to the AC? Are there logs I could get that might help you?

DomiStyle, 2021-11-08:
Yes, restore the default config without any of your changes and disable all sensors except the main AC and try again.

The crash logs are only available via serial so you would have to stay connected constantly while it runs.

dixonte, 2021-11-08:
Alright, I'll see how that goes.

dixonte, 2021-11-11:
And it failed again, though I just realised I never commented out the MQTT lines. Doing that now and will see if it fails again.

dixonte, 2021-11-18:
Sadly, it has failed again. Here's my config:

`esphome:
  name: panasonic-ac-lounge
  platform: ESP32
  board: esp32dev

wifi:
  ssid: "redacted"
  password: "redacted"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Panasonic-Ac Fallback Hotspot"
    password: "redacted"

captive_portal:

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:

# Enable MQTT
#mqtt:
#  broker: "redacted.lan"
#  username: "mqtt"
#  password: "redacted"

ota:
  password: "redacted"

# Requires ESPHome 1.15.0+ for the even parity option
uart:
  tx_pin: GPIO17
  rx_pin: GPIO16
  baud_rate: 9600
  parity: EVEN

external_components:
  source: github://DomiStyle/esphome-panasonic-ac
  components: [panasonic_ac]

climate:
  - platform: panasonic_ac
    id: panasonic_ac_id
    # For CZ-TACG1
    type: cnt

    # For DNSK-P11
    # type: wlan

    name: Panasonic AC
    horizontal_swing_select:
      name: Panasonic AC Horizontal Swing Mode
    vertical_swing_select:
    #  id: vertical_swing_select_id
      name: Panasonic AC Vertical Swing Mode
    outside_temperature:
      name: Panasonic AC Outside Temperature`

dixonte, 2021-11-18:
I might see if I can build a serial port I can hook up to an old netbook to see what I can get out of it.

dixonte, 2021-12-27:
This hasn't happened in forever. Maybe it was fixed by a ESPHome update? In any case, I'll close this for now, sorry for the bother.

closed 2021-12-27 by dixonte