DomiStyle/esphome-panasonic-ac #37: Outside temperatures below zero are out of range

wailashi, 2021-11-09:
Winter is coming and the outside temperature has fallen below zero. 
I am getting the following message in the logs: `Received out of range outside temperature: 254`
As the current outside temperature is about -2, it looks like something is an uint8 when it should be a int8.

Looking at the source of the error the function [update_outside_temperature](https://github.com/DomiStyle/esphome-panasonic-ac/blob/bfca2bab82eef813895b60628789d053d0bfb038/components/panasonic_ac/esppac.cpp#L62) takes the temperature as an uint8. I have not yet had time to investigate if the value is passed as an uint8 elsewhere in the code.

DomiStyle, 2021-11-12:
Hmm, I was sure I fixed that last winter.

Anyway, I pushed some changes to master, changing the remaining function signatures. Couldn't test it yet since there were no temperatures below zero the last few days.

wailashi, 2021-11-14:
Seems to work now. Thanks for the quick fix!

closed 2021-11-14 by wailashi