DomiStyle/esphome-panasonic-ac #38: Custom designed board - Open Source Hardware

jnicolson, 2021-11-11:
Hi,

Firstly thank you so much for taking the time to document and write this library - i had wanted to control my Panasonic AC units via Home Assistant but was about to give up and buy the (overpriced) panasonic modules instead.

This is not really an issue but short of a discussion board here was the best place I could find to post.  I enjoy electronics design as a hobby and this seemed like a fun project to build a custom board for.  It's all open source hardware created in KiCAD and the files are available here on GitHub:

https://github.com/jnicolson/panasonic_ac_wifi

Right now i have the module just sitting in the spot where the CZ-TACG1 unit would sit.  When i get a chance i'll design a 3d printable case for it but it's hidden under the AC cover so out of sight out of mind.

Happy for anyone to do what they want with it and thanks again!

DomiStyle, 2021-11-12:
Pretty cool!

This issue is probably also interesting for you if you didn't see it yet: https://github.com/DomiStyle/esphome-panasonic-ac/issues/12

jnicolson, 2021-11-15:
Hi,

Thanks - and yes i did see that other post but the conversation seemed to switch from a general conversation to mainly discussing the module HA-TB303 created (which looks great but wasn't available to me in Australia and wasn't open so i couldn't make my own).

If it'd be better for this to be combined into that issue please let me know and i'll close this one and post in there!  Thanks again for the great project.

DomiStyle, 2021-11-17:
> If it'd be better for this to be combined into that issue please let me know and i'll close this one and post in there! Thanks again for the great project.

No worries, just wanted to link it. :)

HA-TB303, 2023-09-09:
We have a DIY version now. Https://espthings.io/p11

HA-TB303, 2025-03-08:
New version online as of today: [https://www.ingeniousmakers.com/p11](https://www.ingeniousmakers.com/p11)