DomiStyle/esphome-panasonic-ac #39: 8/10 degree heat mode

trsqr, 2021-11-14:
Does the esphome-panasonic-ac support the activation of the 8/10 degree heat mode? In newer devices it's the 8/15 degree mode (you can choose the temperature between 8 and 15 degrees). If not, how could I help? 

[image]

I can activate this mode over the Intesis WiFi controller connected to CN-CNT, so I assume it should be possible to implement.

DomiStyle, 2021-11-14:
It is currently not supported.

In order to add support for it an ESP connected to CN-CNT or CN-WLAN is necessary in order to log the incoming packets.

trsqr, 2021-11-17:
I'll try to help with this once I'm home again in December.

agrauballe, 2021-11-17:
Currently using the panasonic_cc integration by sockless-coding using the Panasonic Wifi module (will switch to this integration as soon as I get the HW needed).
In the panasonic_cc setting the hvac_mode to "heat", temperature to "8" and fan_mode to "High" is implicitly the same as the 8/15 mode. The important thing is the fan speeed, even though I don't understand why you cannot use "Low" fan speed for 8 degrees.
Guess it will be the same here, but the 8/15 mode might be nice.

DomiStyle, 2021-11-17:
@agrauballe Thanks. Currently the temperature is locked from 16 to 30°C so a separate switch to enable the mode will need to be added I assume.

Will be interesting to try this out on ACs that don't support it though, maybe those functions are just hidden on them.

Sundius, 2021-12-15:
I have "limited" (or read old) coding skills but would like to help if I can. I have one new unit up and running with built in Wi-Fi and 2 slightly older that I'm now upgrading with this project. I'd like to have the 8-10 and/or the 8-15 feature "aka" Summerhouse" as it is called in the Panasonic app working for my garage unit. 

If it helps I can test and describe the flow from the Comfort Cloud app on this mode and I can also provide logs from my garage unit via the ESP with some guidance. 

It looks like the unit remains in "Heat" mode when you enable it in the app but the UI restricts the temperature range to 8-10 or 8-15 depending on what you selected in application settings menu. As stated above the fan runs at 100% as well. According to the installer this is to create a more consistent airflow over the built in temperature sensor and the room making the unit regulate more efficiently -> saving energy. I take that with a pinch of salt but that was the statement and that is how the "mode" works.

Let me know if I can pitch in?

DomiStyle, 2021-12-15:
@Sundius You can turn on verbose logging in ESPHome and post your logs in different modes.
That should show what is being set by the Comfort Cloud app. You will have to connect an ESP to one of the ports and the other port to the official Panasonic wifi adapter.

If you can toggle the modes from the IR remote you can use that as well instead of the official wifi adapter.

Sundius, 2021-12-20:
@DomiStyle Sorry for the delay, life happened and for some reason I've had some unifi/ESPHome issues after some needed "Patch activity". All sorted now though so ready to pitch in! First quick log capture below. 
1. Start "state": Had the unit running with auto fan, heatmode set to 22C
2. Enabled the "+8/15C HEAT" mode with the IR remote, I had done one time before and set it to 8C which it remembers, you can alter temp between 8-15 when in this "Mode".
3. Pressed "MODE" on the remote again which turned the unit back to start state (step 1).

LOG:
[22:31:36][V][panasonic_ac.cz_tacg1:248]: Polling AC
[22:31:36][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[22:31:36][V][panasonic_ac:247]: RX: 70.20.44.2C.80.A0.FC.00.00.00.00.00.48.32.00.00.20.11.80.80.FF.19.00.FF.80.80.FF.80.0E.00.00.00.00.00.95 (35)
[22:31:36][D][climate:381]: 'Panasonic AC' - Sending state:
[22:31:36][D][climate:384]:   Mode: HEAT
[22:31:36][D][climate:392]:   Custom Fan Mode: Automatic
[22:31:36][D][climate:398]:   Custom Preset: Normal
[22:31:36][D][climate:401]:   Swing Mode: VERTICAL
[22:31:36][D][climate:404]:   Current Temperature: 25.00°C
[22:31:36][D][climate:410]:   Target Temperature: 22.00°C
[22:31:41][V][panasonic_ac.cz_tacg1:248]: Polling AC
[22:31:41][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[22:31:41][V][panasonic_ac:247]: RX: 70.20.44.2C.80.A0.FC.00.00.00.00.00.48.32.00.00.20.11.80.80.FF.19.00.FF.80.80.FF.80.0E.00.00.00.00.00.95 (35)
[22:31:41][D][climate:381]: 'Panasonic AC' - Sending state:
[22:31:41][D][climate:384]:   Mode: HEAT
[22:31:41][D][climate:392]:   Custom Fan Mode: Automatic
[22:31:41][D][climate:398]:   Custom Preset: Normal
[22:31:41][D][climate:401]:   Swing Mode: VERTICAL
[22:31:41][D][climate:404]:   Current Temperature: 25.00°C
[22:31:41][D][climate:410]:   Target Temperature: 22.00°C
[22:31:46][V][panasonic_ac.cz_tacg1:248]: Polling AC
[22:31:46][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[22:31:46][V][panasonic_ac:247]: RX: 70.20.44.2C.80.A0.FC.00.00.00.00.00.4C.32.00.00.20.11.80.80.FF.19.00.FF.80.80.FF.80.30.00.02.00.00.00.6D (35)
[22:31:46][V][sensor:062]: 'Panasonic AC Power Consumption': Received new state 46.000000
[22:31:46][D][sensor:113]: 'Panasonic AC Power Consumption': Sending state 46.00000 W with 0 decimals of accuracy
[22:31:46][D][climate:381]: 'Panasonic AC' - Sending state:
[22:31:46][D][climate:384]:   Mode: HEAT
[22:31:46][D][climate:392]:   Custom Fan Mode: Automatic
[22:31:46][D][climate:398]:   Custom Preset: Normal
[22:31:46][D][climate:401]:   Swing Mode: VERTICAL
[22:31:46][D][climate:404]:   Current Temperature: 25.00°C
[22:31:51][V][panasonic_ac.cz_tacg1:248]: Polling AC
[22:31:51][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[22:31:51][V][panasonic_ac:247]: RX: 70.20.44.2C.80.A0.FC.00.00.00.00.00.4C.32.00.00.20.11.80.80.FF.19.00.FF.80.80.FF.80.41.00.02.00.00.00.5C (35)
[22:31:51][V][sensor:062]: 'Panasonic AC Power Consumption': Received new state 63.000000
[22:31:51][D][sensor:113]: 'Panasonic AC Power Consumption': Sending state 63.00000 W with 0 decimals of accuracy
[22:31:51][D][climate:381]: 'Panasonic AC' - Sending state:
[22:31:51][D][climate:384]:   Mode: HEAT
[22:31:51][D][climate:392]:   Custom Fan Mode: Automatic
[22:31:51][D][climate:398]:   Custom Preset: Normal
[22:31:51][D][climate:401]:   Swing Mode: VERTICAL
[22:31:51][D][climate:404]:   Current Temperature: 25.00°C
[22:31:56][V][panasonic_ac.cz_tacg1:248]: Polling AC
[22:31:56][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[22:31:56][V][panasonic_ac:247]: RX: 70.20.44.2C.80.A0.FC.00.00.00.00.00.4C.32.00.00.20.11.80.80.FF.19.00.FF.80.80.FF.80.53.00.03.00.00.00.49 (35)
[22:31:56][V][sensor:062]: 'Panasonic AC Power Consumption': Received new state 80.000000
[22:31:56][D][sensor:113]: 'Panasonic AC Power Consumption': Sending state 80.00000 W with 0 decimals of accuracy
[22:31:56][D][climate:381]: 'Panasonic AC' - Sending state:
[22:31:56][D][climate:384]:   Mode: HEAT
[22:31:56][D][climate:392]:   Custom Fan Mode: Automatic
[22:31:56][D][climate:398]:   Custom Preset: Normal
[22:31:56][D][climate:401]:   Swing Mode: VERTICAL
[22:31:56][D][climate:404]:   Current Temperature: 25.00°C
[22:32:01][V][panasonic_ac.cz_tacg1:248]: Polling AC
[22:32:01][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[22:32:01][V][panasonic_ac:247]: RX: 70.20.44.2C.80.A0.FC.00.00.00.00.00.4C.32.00.00.20.11.80.80.FF.19.00.FF.80.80.FF.80.66.00.03.00.00.00.36 (35)
[22:32:01][V][sensor:062]: 'Panasonic AC Power Consumption': Received new state 99.000000
[22:32:01][D][sensor:113]: 'Panasonic AC Power Consumption': Sending state 99.00000 W with 0 decimals of accuracy
[22:32:01][D][climate:381]: 'Panasonic AC' - Sending state:
[22:32:01][D][climate:384]:   Mode: HEAT
[22:32:01][D][climate:392]:   Custom Fan Mode: Automatic
[22:32:01][D][climate:398]:   Custom Preset: Normal
[22:32:01][D][climate:401]:   Swing Mode: VERTICAL
[22:32:01][D][climate:404]:   Current Temperature: 25.00°C
[... 114 lines cut ...]
[22:32:56][D][climate:392]:   Custom Fan Mode: Automatic
[22:32:56][D][climate:398]:   Custom Preset: Normal
[22:32:56][D][climate:401]:   Swing Mode: VERTICAL
[22:32:56][D][climate:404]:   Current Temperature: 25.00°C
[22:33:01][V][panasonic_ac.cz_tacg1:248]: Polling AC
[22:33:01][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[... cut ...]
[22:33:06][V][panasonic_ac:247]: RX: 70.20.40.2C.80.A0.FC.00.00.00.00.00.00.2C.00.00.20.11.80.80.FF.19.00.FF.80.80.FF.80.06.00.00.00.00.00.EF (35)
[22:33:06][D][climate:381]: 'Panasonic AC' - Sending state:
[22:33:06][D][climate:384]:   Mode: OFF
[22:33:06][D][climate:392]:   Custom Fan Mode: Automatic
[22:33:06][D][climate:398]:   Custom Preset: Normal
[22:33:06][D][climate:401]:   Swing Mode: VERTICAL
[22:33:06][D][climate:404]:   Current Temperature: 25.00°C
[22:33:06][D][climate:410]:   Target Temperature: 22.00°C

DomiStyle, 2021-12-20:
Thanks, from the known fields the following changed:

Temperature set to 10 (5°C, was 22°C)
Fan speed is set to 70 (level 5, was auto)

Is the fan speed on the highest level when this mode is active?

Can you try going from Heat/cool mode to "+8/15C HEAT" mode? Would be interesting to see if it changes to heat mode automatically as well.

I added a testing branch with the minimum temperature reduced to 5°C, you can try it by changing this line in the config:

`source: github://DomiStyle/esphome-panasonic-ac@low-temperature-mode`

After that you can try manually setting the temperature to 5°C and see if it works. In fact you can even set the temperature outside of that and see what happens.

DomiStyle, 2021-12-20:
Played around with the target temperature on my AC and it goes down as low as 0°C, not sure if it will actually do it but it will accept the value.

Sundius, 2021-12-21:
It is basically "just" the "HEAT" mode with some options blocked like fan speed and a smaller temperature range from a UI input/output point of view, right?

DomiStyle, 2021-12-23:
Yes, from what I have seen so far it seems to just switch to HEAT mode, locks the temperature range to 8-15 or 8-10 and the fan speed is locked to max.

All of that should be possible via a Home Assistant script for now. Once somebody tested it I could add a switch for it.
Of course, having both 8-15 and 8-10 seems sort of useless to me?

Sundius, 2021-12-26:
@DomiStyle Agree, having the legacy mode seems unnecessary. The units seem to be forgiving enough and I assume the low temp is enough as it is the least common denominator (8C). 
I haven't had time to test your fork yet but I did try out the logic a bit more using a Sensibo remote. It lacks the "Sunmerhouse" mode but allows you to set temp as low as 8C. So by looking at the physical LED on the Panasonic I've concluded that it takes 2 conditions to enable the mode "for real".

1. A temp set below 16C 
2. Full fan speed 

When both are true the 8/15C LED on the unit lights up. The unit can run at temps below 16C without the mode being enabled but I don't know what that means in reality, at least it shows no signs of weird behavior. 

Just wanted to share... 

Merry Christmas and I truly appreciate all the hard work that went in to this git!

trsqr, 2021-12-26:
Thank you @Sundius for the work. I still have my ESP32 board on my desk and wires not connected to the unit. Perhaps the temperature ranges could be configured. 8-15 for the newer devices and 8-10 C for older ones? Although it's interesting that the unit actual runs fine on temperature ranges that you cannot choose on the remote.

Sundius, 2022-01-03:
So, finally.... 
I have created a quick and dirty script for HA that confirms that you can enable this mode (with a few bugs). If the HVAC is off you have to run the script a few times for the temperature below 16C to take effect. At least that is my quick analysis. 
The script is not pretty at all, I did it on my phone over the Holiday's using HA's script UI, could be made better for sure. 

When it finally accepts the scrip, the 8/15C LED is turned on on the unit = success! 

I had to run the script 2 times, the first time it turned on the HVAC unit but used the old set temperature, second run it changed to 10C as expected and the LED came on on the unit. 

[image]

Script:

sequence:
  - service: climate.set_hvac_mode
    target:
      entity_id: climate.panasonic_ac
    data:
      hvac_mode: heat
  - service: climate.set_fan_mode
    target:
      entity_id: climate.panasonic_ac
    data:
      fan_mode: '5'
  - service: climate.set_temperature
    data:
      temperature: '{{ states.input_number.temperature.state | int }}'
    target:
      entity_id: climate.panasonic_ac
mode: single
alias: Summerhouse mode
icon: mdi:air-filter

@DomiStyle, again, thanks for all your pointers and hard work! This test was done on the official branch, not the fork.

donnib, 2022-01-11:
@Sundius  So what you are saying is that the repo here won't support it but you do it manually from HASS ?

donnib, 2022-01-11:
What would be nice would be to be able to run the unit in "Summer house mode" with temp below 16c but WITHOUT the fan at highest setting. That is not possible with the remote at least, as soon as you enable "Summer house mode" it cranks up the fan to max.

DomiStyle, 2022-01-11:
@donnib It's no problem for me to add a switch that enables "Summer house mode" or "Freeze protection". It's just a matter of knowing which parameters to set for it.

Obviously the temperature range needs to be changed to 8°C to 15°C and the mode should be locked to heating.
Putting on max fan speed or keeping the current one is up for discussion.

donnib, 2022-01-11:
@DomiStyle does this help ? https://github.com/Egyras/HeishaMon/blob/master/ProtocolByteDecrypt.md 

I think Holiday mode = "Summer house mode"

Sundius, 2022-01-11:
> @Sundius So what you are saying is that the repo here won't support it but you do it manually from HASS ?

Correct, the functionality does not exist within this integration in HA as of today thus this discussion and various tests! :) 
I did a quick script test in HA just to make sure that we can enable "Summer House mode" via a script. Just trying to help @DomiStyle by validating that the logic inside the HVAC is as "simple" as we have guessed in this thread.

agrauballe, 2022-02-08:
@Sundius and @DomiStyle I think the reason why the script needs to be run more than once is that th AC can only handle cmds at a certain rate. We saw this in another topic. It might require some queue.
Do you know if there is any handshaking going on where the AC should acknowledge the request and we could end up in the next cmd is sent before the previous one is executed?
For now I have solved the issue in HA by waiting for state change or by simple time delay between cmds.

DomiStyle, 2022-02-08:
@agrauballe There are 2 different protocols.

DNSK-P11 uses ack packets and different commands for each action and needs a queue.
CZ-TACG1 has no ack packets and only sends a single packet with all values included, it would need a small delay before executing the command.

trsqr, 2022-04-10:
I finally managed to setup my unit (crimping that CN-CNT connector was the thing that took me most time, bloody small pins and had to wait for my Aliexpress order of the tool). Here's my log, but I suppose it's nothing new anymore.

```
[08:51:43][D][climate:381]: 'Panasonic AC' - Sending state:
[08:51:43][D][climate:384]:   Mode: HEAT
[08:51:43][D][climate:392]:   Custom Fan Mode: 2
[08:51:43][D][climate:398]:   Custom Preset: Normal
[08:51:43][D][climate:401]:   Swing Mode: OFF
[08:51:43][D][climate:404]:   Current Temperature: 22.00°C
[08:51:43][D][climate:410]:   Target Temperature: 19.00°C
[08:51:48][D][sensor:113]: 'Panasonic AC Power Consumption': Sending state 54.00000 W with 0 decimals of accuracy
[08:51:48][D][select:032]: 'Panasonic AC Vertical Swing Mode': Sending state auto
[08:51:48][D][climate:381]: 'Panasonic AC' - Sending state:
[08:51:48][D][climate:384]:   Mode: HEAT
[08:51:48][D][climate:392]:   Custom Fan Mode: 5
[08:51:48][D][climate:398]:   Custom Preset: Normal
[08:51:48][D][climate:401]:   Swing Mode: VERTICAL
[08:51:48][D][climate:404]:   Current Temperature: 21.00°C
[08:51:48][D][climate:410]:   Target Temperature: 9.00°C
[08:51:53][D][climate:381]: 'Panasonic AC' - Sending state:
[08:51:53][D][climate:384]:   Mode: HEAT
[08:51:53][D][climate:392]:   Custom Fan Mode: 5
[08:51:53][D][climate:398]:   Custom Preset: Normal
[08:51:53][D][climate:401]:   Swing Mode: VERTICAL
[08:51:53][D][climate:404]:   Current Temperature: 21.00°C
[08:51:53][D][climate:410]:   Target Temperature: 9.00°C
[08:51:58][D][climate:381]: 'Panasonic AC' - Sending state:
[08:51:58][D][climate:384]:   Mode: HEAT
[08:51:58][D][climate:392]:   Custom Fan Mode: 5
[08:51:58][D][climate:398]:   Custom Preset: Normal
[08:51:58][D][climate:401]:   Swing Mode: VERTICAL
[08:51:58][D][climate:404]:   Current Temperature: 21.00°C
[08:51:58][D][climate:410]:   Target Temperature: 9.00°C
[08:52:03][D][sensor:113]: 'Panasonic AC Power Consumption': Sending state 32.00000 W with 0 decimals of accuracy
[08:52:03][D][climate:381]: 'Panasonic AC' - Sending state:
[08:52:03][D][climate:384]:   Mode: HEAT
[08:52:03][D][climate:392]:   Custom Fan Mode: 2
[08:52:03][D][climate:398]:   Custom Preset: Normal
[08:52:03][D][climate:401]:   Swing Mode: VERTICAL
[08:52:03][D][climate:404]:   Current Temperature: 21.00°C
[08:52:03][D][climate:410]:   Target Temperature: 20.50°C
```

andreas-berg, 2022-11-28:
@DomiStyle great project, this was the easiest device integration I've ever done. Got my Panasonic AC from 0 to 100% functional as part of Home assistant in less than 3h (not counting the time to procure the connector), thank you!

What is the status of this "Summer house" - 8/15-mode ? I'd like to use this feature, will it be merged in to main-branch or still in need of development. I'd be happy to help with coding or testing if needed.

jmichiel, 2023-02-08:
Now this is very interesting!
My units (CS-TZ20WKEW and CS-TZ25WKEW) don't have this mode, so 16 is their lowest setting, which sucks big time: I'd like to set them to 14° at night...
I was thinking of implementing that in the integration itself so that it would simply shut the unit off when it goes above the target temp (+ some hysteresis, of course). I already do something similar now via node-red, since the real temp ends up about 2 degrees above the target temp mostly (if set to 16°, the room ends up being 18°, so that's ever 4° over the 14° that I'd like...
Any chance this might work on my units? I guess trying it out would be the only option...

edalberg, 2024-09-07:
@DomiStyle
My Panasonic heat pump has worked fine by setting fan to 5 and temperature to 10 using automation in HA. Using «Climate.set_fan_mode» and «Climate.set_temperature». But, now it has stopped working. When I run the automation manually, the following message comes up «Provided temperature 8.0 is not valid. Accepted range is 16.0 to 30.0». Any idea why this is happening?

MEKadan, 2024-09-17:
> @DomiStyle My Panasonic heat pump has worked fine by setting fan to 5 and temperature to 10 using automation in HA. Using «Climate.set_fan_mode» and «Climate.set_temperature». But, now it has [...]

I would like to see a solution for this as well.

agrauballe, 2024-09-18:
@edalberg @MEKadan  @CSchlipp 
I can confirm that setting the code snippet below under platform makes 8 degrees possible again:

    visual:
      min_temperature: 8
      max_temperature: 30

MEKadan, 2024-09-18:
Thanks @CSchlipp , that does the trick at least UI-wise. Need to test whether my model actually obeys these commands. This, however, I need to spare for some later time.

MEKadan, 2024-09-18:
BTW, for monitoring purposes, does anyone know, how to get a hold on these variables/entities in home assistant?

[image]

I can see them OK in Simple Thermostat card that I use, but how to get their values to entities?