DomiStyle/esphome-panasonic-ac #41: Frequent crashes on ESP32 NodeMCU wroom board

jkaberg, 2021-11-30:
Hey @DomiStyle , I'm faceing lots of crashes with my ESP32 NodeMCU wroom board. I've gatherd logs and config below, and was hopeing you could give me a helping hand!

The AC unit is an Panasonic HZ35UKE with DNSK-P11 from factory.

Running config:
```
esphome:
  name: panasonic-ac
  platform: esp32
  board: nodemcu-32s

logger:
  level: VERBOSE #DEBUG #INFO

api:
ota:

wifi:
  power_save_mode: none
  fast_connect: true
  reboot_timeout: 60s
  ssid: "NGV6-IoT"
  password: "<redacted>"
  output_power: 17
  manual_ip:
    static_ip: 10.0.40.9
    gateway: 10.0.40.1
    subnet: 255.255.255.0
    dns1: 10.0.40.1

mdns:
  disabled: true

uart:
  tx_pin: GPIO17
  rx_pin: GPIO16
  baud_rate: 9600
  parity: EVEN

external_components:
  source: github://DomiStyle/esphome-panasonic-ac
  components: [panasonic_ac]

climate:
  - platform: panasonic_ac
    type: wlan

    name: Panasonic AC
    horizontal_swing_select:
      name: Panasonic AC Horizontal Swing Mode
    vertical_swing_select:
      name: Panasonic AC Vertical Swing Mode
    outside_temperature:
      name: Panasonic AC Outside Temperature

    # Enable as needed
    # eco_switch:
    #   name: Panasonic AC Eco Switch
    # econavi_switch:
    #   name: Panasonic AC Econavi Switch
    nanoex_switch:
      name: Panasonic AC NanoeX Switch
    # mild_dry_switch:
    #   name: Panasonic AC Mild Dry Switch
    # current_power_consumption:
    #   name: Panasonic AC Power Consumption

    # Useful when the ac does not report a current temperature (CZ-TACG1 only)
    # current_temperature_sensor: temperature_sensor_id
```
[... 808 lines cut ...]
ښ▒ke▒▒Y.▒idf▒▒▒▒▒J▒(3471J▒▒▒▒▒▒[0m
[V][esp-idf:000]: state: assoc -> run )▒J▒[0▒j
                                              ▒▒j▒eյU▒▒idf:000]:

Guu edtaio Ero: or  1paiced(Iterut dttieot n PU
A2    : x3ff0058 3     : x3f4183a0 4      0x00000014 5     : x3ffff0
A       0x3fbf58 7     : x0000001 8     : x0000001 9     : xf131
A1    : 00000a5 11     0x3ff1ae  A12     0x8008b58b A3     0x3fbee
14    : 0000008 15    : x0000001 AR    : x000001a ECCUSE 0x00000006
                                                                    ECVADR: x000000 LEG    0x                                                                                                Baktrace0x4000c2710xfb1590 0x40ad:x3fb50 0x008c57c:x3fb5d0 0x40c3:0xfb1584

Coe 0 rgisterdump
P     : x0086ff  S      0x00060c3  0      0x800fc0 A     : xfafe0
                                                                  A     : x000000 3     : x3f                                                                                                fca4  4      0x00000007 5     : xfcc16
                                       A     : xfccb4 7      0x0000001a A      0x800fbac A                                                                                                      0x3ff2cc
            A0    : x000001 11    : x3ffca4 12    : x0000001 13    : x3ffcb0
A4      0x3fc29  15    : xfc2b9  SAR     0x00000018 XCCASE: x0000006
                                                                     ECVADR: x000000 BEG   :                                                                                                 x400c4  LEND    0x4000c36b LCOUN  : x0000000

LF fil SHA56: 000000000000

Bcktrce:0x08f20xffa2e0 x40c1fd:0x3fa40 x4153d90xffa440 04164:03fa480 0x408da:x3fa40

Reboting..
```

jkaberg, 2021-11-30:
Another one for good measure:

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
[I][logger:214]: Log initialized
[C][ota:452]: There have been 1 suspected unsuccessful boot attempts.
[D][esp32.preferences:114]: Saving preferences to flash...
[I][app:029]: Running through setup()...
[V][app:030]: Sorting components by setup priority...
[C][uart.arduino_esp32:075]: Setting up UART...
[I][panasonic_ac:039]: Panasonic AC component v2.2.1 starting...
[D][panasonic_ac.dnskp11:013]: Using DNSK-P11 protocol via CN-WLAN
[C][wifi:037]: Setting up WiFi...
[V][esp32.preferences:057]: nvs_get_blob('-1446447864'): ESP_ERR_NVS_NOT_FOUND - the key might not be set yet
[V][wifi_esp32:039]: Enabling STA.
[V][esp-idf:000]: I (88) wifi:
[V][esp-idf:000]: wifi driver task: 3ffb5bfc, prio:23, stack:3584, core=0
[V][esp-idf:000]:

[V][esp-idf:000]: I (369) wifi:
[V][esp-idf:000]: wifi firmware version: dc30037
[V][esp-idf:000]:

[V][esp-idf:000]: I (370) wifi:
[V][esp-idf:000]: config NVS flash: enabled
[V][esp-idf:000]:

[V][esp-idf:000]: I (381) wifi:
[V][esp-idf:000]: config nano formating: disabled
[V][esp-idf:000]:

[V][esp-idf:000]: I (394) wifi:
[V][esp-idf:000]: Init data frame dynamic rx buffer num: 32
[V][esp-idf:000]:

[V][esp-idf:000]: I (406) wifi:
[V][esp-idf:000]: Init management frame dynamic rx buffer num: 32
[V][esp-idf:000]:

[V][esp-idf:000]: I (420) wifi:
[V][esp-idf:000]: Init management short buffer num: 32
[V][esp-idf:000]:

[V][esp-idf:000]: I (432) wifi:
[V][esp-idf:000]: Init dynamic tx buffer num: 32
[V][esp-idf:000]:

[V][esp-idf:000]: I (444) wifi:
[V][esp-idf:000]: Init static rx buffer size: 1600
[V][esp-idf:000]:

[V][esp-idf:000]: I (456) wifi:
[V][esp-idf:000]: Init static rx buffer num: 16
[V][esp-idf:000]:

[V][esp-idf:000]: I (468) wifi:
[V][esp-idf:000]: Init dynamic rx buffer num: 32
[V][esp-idf:000]:

[V][esp-idf:000]: I (577) wifi:
[... 359 lines cut ...]
PC       0x40086f2  PS       0x00060234  A0       0x8012a3c A1       0x3fbf0
A2       0x3fcf0  A3       0x00000006  A4       0x00000017  A5       0x0000bf
A6       0x00000000  A7       0x3fb540  A8       0x00000020  A9       0x80000020
A10      0x00000003  A11      0x00000017  A12      0x00000005  A13      0x00000020
A14      0x00000020  A15      0x00000000  SAR      0x00000010  EXCCAUSE 0x00000005
EXCVADDR 0x00000000  LBEG     0x4000ce  LEND     0x4000cf  LCOUNT   0xffff

ELF file SHA256: 000000000000000

                                Backtrace:0x40086f2:0x3fbf00x4012a39:0x3fbf0 0x4012ab:0x3fbf0 0x401291f:0x3fb010 0x4012fc:0x3fb030 0x401344d:0x3fb060 0x40122063:0x3fb080 0x40089da0x3fb0b

ore 1 register dump:
PC       0x40157886  PS       0x00060e4  A0       0x80142d2  A1       0x3fb6c
A2       0x00000000  A3       0x80000001  A4       0x00000000  A5       0x00000001
A6       0x00060a0  A7       0x00000000  A8       0x00060023  A9       0x3fb230
A10      0x00000000  A11      0x00000001  A12      0x80089fc A13      0x3fb980
A14      0x00000000  A15      0x3fb3c  SAR      0x00000000  EXCCAUSE 0x00000005
EXCVADDR 0x00000000  LBEG     0x00000000  LEND     0x00000000  LCOUNT   0x00000000

ELF file SHA256: 000000000000000

                                Backtrace:0x40157886:0x3fb6c 0x40142df0x3fb6e 0x4008b79:0x3fb700 0x40089da0x3fb720

Rebooting...
```

DomiStyle, 2021-12-01:
@jesserockz Any ideas on how to diagnose something like that?