DomiStyle/esphome-panasonic-ac #42: Feature request: Long Term Statistics on Energy

brywithawhy, 2021-12-02:
Would like to add my units as individual devices in HA. The help article refers:

https://developers.home-assistant.io/docs/core/entity/sensor/#long-term-statistics

Can this change be made to the energy measuring sensor?

DomiStyle, 2021-12-03:
ESPHome already supports state_class and device_class, you can just add them to any sensor in your config.

```
current_power_consumption:
  name: Panasonic AC Power Consumption
  unit_of_measurement: W
  accuracy_decimals: 0
  device_class: "power"
  state_class: "measurement"
```

However, there is no total power consumption sensor so I think you can't add them to the energy statistics in Home Assistant.

brywithawhy, 2021-12-06:
According to the docs it should work without the power consumption sensor but I couldn't get it to. Instead I created a total daily one with esphome which works great. 
here's my config which adds this sensor to HA which can be added to the energy ecosystem for those who might want to use it

```
esphome:
  name: bedroom_ac
  platform: ESP32
  board: esp32thing

wifi:
  networks:
    - ssid: "xxxx"
      password: "xxxx"
  use_address: bedroom_ac.iot
  power_save_mode: none

logger:
  level: DEBUG

api:
  password: "xxxx"

ota:
  password: "xxxx"

uart:
  tx_pin: GPIO17
  rx_pin: GPIO16
  baud_rate: 9600
  parity: EVEN

external_components:
  source: github://DomiStyle/esphome-panasonic-ac
  components: [panasonic_ac]

climate:
  - platform: panasonic_ac
    type: cnt
    name: Bedroom AC
    vertical_swing_select:
      name: Bedroom AC Vertical Swing Mode
    outside_temperature:
      name: Bedroom AC Outside Temperature
    current_power_consumption:
      id: bedroom_ac_power_consumption
      name: Bedroom AC Power Consumption
      unit_of_measurement: W
      accuracy_decimals: 0
      device_class: power
      state_class: measurement
    eco_switch:
      name: Bedroom AC Eco Switch

sensor:
  - platform: total_daily_energy
    name: "Bedroom AC Total Daily Energy"
    power_id: bedroom_ac_power_consumption
    filters:
        - multiply: 0.001
    unit_of_measurement: kWh
    device_class: energy

time:
  - platform: sntp
    id: my_time
```

sunsetdk, 2021-12-15:
This can also be done using the integration sensor

  - platform: integration
    name: "Total Power Consumption"
    min_save_interval: 5min
    sensor: instantPower
    time_unit: h
    state_class: total_increasing
    device_class: energy
    accuracy_decimals: 3
    unit_of_measurement: 'kWh'
    filters:
        - multiply: 0.001

danielcherubini, 2024-06-21:
An update, there was some changes that change the config above

First give the `current_power_consumption` block a new field `id` call it whatever you want

```yaml
climate:
    current_power_consumption:
      name: Panasonic AC Power Consumption
      id: panasonic_ac_4th_panasonic_ac_power_consumption
```

```yaml
sensor:
  - platform: integration
    name: "Total Daily Energy"
    sensor: panasonic_ac_4th_panasonic_ac_power_consumption
    time_unit: h
    state_class: total_increasing
    device_class: energy
    accuracy_decimals: 3
    unit_of_measurement: kWh
    filters:
      - multiply: 0.001
preferences:
  flash_write_interval: 180s  
```