DomiStyle/esphome-panasonic-ac #43: Cooling Only models

maeneak, 2021-12-05:
Is it possible to provide a config entry to remove the 'Heating' mode? Although its not a major drama it would be great to be able to disabe this so it doesnt appear in the UI

DomiStyle, 2021-12-06:
Yes, the supported modes are set in the climate traits and it should be possible to set them dynamically:
https://github.com/DomiStyle/esphome-panasonic-ac/blob/master/components/panasonic_ac/esppac.cpp#L21

> Also with cooling only models the HA HVAC_MODE_HEAT_COOL mode is redundant and HVAC_MODE_AUTO would be more relevant

HVAC_MODE_AUTO  is only allowed for ACs which provide some form of automation, no temperature can be set in that mode. 
ACs which just switch between heat and cool depending on the temperature are supposed to use HVAC_MODE_HEAT_COOL.

See: https://github.com/home-assistant/core/blob/3efdf29dfa178838002201eaea258db78e6fa2ba/homeassistant/components/climate/const.py#L16

hpirila, 2022-04-04:
In Midea climate component it can be set like this ([link](https://esphome.io/components/climate/midea.html))

```
    supported_modes:
      - FAN_ONLY
      - HEAT_COOL
      - COOL
      - HEAT
      - DRY
 ```
This would be useful in Panasonic AC too.