DomiStyle/esphome-panasonic-ac #44: Panasonic nz25vke giving error in log

sunsetdk, 2021-12-12:
Everything works except for the eco/Navi button.

If I enable it on the remote i get the following error in the log
_Received unknown nanoex value_
I then thought mine wasn't a eco/navi, but just a Eco, if i attemp that i get no function from the button.

If I enable the button in HASS the indicator on the unit changes state, so data are send correctly. But some times i disables straight away again (on next data from unit).

Looking at the code, it looks like nanoex and eco/navi button use the same byte and same bits, which doesn't make sense.

I'm willing to debug/look at it. Do you have some info on what is contained in which bytes (as far as you know)

DomiStyle, 2021-12-12:
There's a protocol description for both wifi adapters here: https://github.com/DomiStyle/esphome-panasonic-ac/tree/master/protocol

sunsetdk, 2021-12-12:
I made a PR, which fix the problem for me (it shouldn't break anything)

closed 2021-12-12 by sunsetdk