DomiStyle/esphome-panasonic-ac #47: Current temp higher than set temp

sunsetdk, 2021-12-12:
Set temperature vs current temperature.
Im using cnt mode, and i seem to have a offset of 4-5 degrees in temperature. 

The current temperature is about 4-5 degrees  higher than the set, this seems to be pretty constant.

DomiStyle, 2021-12-13:
The temperature is read directly from the AC, so any incorrect measurements come directly from the AC. Out of experience the current temperature measurement is pretty bad when the AC is running.

I assume you are heating with your AC? My measurements were always too low when cooling.

sunsetdk, 2021-12-13:
Yes i use the AC to heat the room. But i have place a exernal temp sensor next to the AC, and it seems to report the same temperatue, which is also too high.
I'm closing this as it is not related to this project

closed 2021-12-13 by sunsetdk

DomiStyle, 2021-12-13:
The most ideal way to measure temperature would be while the AC has been off for a while or with an external sensor on the other side of the room.

Everything else is just guessing.