DomiStyle/esphome-panasonic-ac #49: Delay in commands

sygys, 2021-12-20:
A few times now i experienced that when the module gets 2 commands in a short time not all commands are executed. could there be a delay added to the code in which when more then one command is pushed the next one will be delayed. I had a few times now that when Home assistant pushed a temperature change and in the same second a turn off command the AC wouldnt turn off. this also happens when swingmode is changed and the temperature. in some cases the AC even turns from cool mode to heat.

I see the code hasnt been updated for over 4 months now. I hope this project is still active...

DomiStyle, 2021-12-20:
The last update was just 5 days ago.

A delay in commands would require a queue to be added, which is currently not the case. For me the project is feature complete but of course I will accept pull requests if anyone wants to implement this.

sygys, 2021-12-21:
I wonder how the official wifi modules handle multiple commands at the same time

agrauballe, 2022-02-05:
I also noticed this issue. In this case the the cmds: heat, temp=8 and fan_mode=5 is set  but last cmd fan_mode is not applied. For now I will play with delays in HA to make sure the cmds are executed.

[21:49:16][D][climate:381]: 'Panasonic AC' - Sending state:
[21:49:16][D][climate:384]:   Mode: OFF
[21:49:16][D][climate:392]:   Custom Fan Mode: 4
[21:49:16][D][climate:398]:   Custom Preset: Normal
[21:49:16][D][climate:401]:   Swing Mode: OFF
[21:49:16][D][climate:404]:   Current Temperature: 9.00°C
[21:49:16][D][climate:410]:   Target Temperature: 9.00°C
[21:49:20][D][climate:009]: 'Panasonic AC' - Setting
[21:49:20][D][climate:013]:   Mode: HEAT
[21:49:20][D][climate:038]:   Target Temperature: 8.00
[21:49:20][D][climate:009]: 'Panasonic AC' - Setting
[21:49:20][D][climate:017]:  Custom Fan: 5
[21:49:20][D][climate:381]: 'Panasonic AC' - Sending state:
[21:49:20][D][climate:384]:   Mode: HEAT
[21:49:20][D][climate:392]:   Custom Fan Mode: 4
[21:49:20][D][climate:398]:   Custom Preset: Normal
[21:49:20][D][climate:401]:   Swing Mode: OFF
[21:49:20][D][climate:404]:   Current Temperature: 9.00°C
[21:49:20][D][climate:410]:   Target Temperature: 8.00°C

jmichiel, 2023-04-07:
If you're using CN-CNT, then my pull-request should solve that:
https://github.com/DomiStyle/esphome-panasonic-ac/pull/88

DomiStyle, 2023-06-21:
Closed via #88

closed 2023-06-21 by DomiStyle