DomiStyle/esphome-panasonic-ac #50: External temperature sensor

Driekes, 2022-01-09:
Is it possible with this board to use an external temperature sensor?

I currently have two TZ models. They have a theoretical lowest power consumption of 200W. However, I have only seen it when the outside temperature is low enough (oddly enough).

E.g. outside 4 degrees, inside 17 degrees, it is able to work on 200W for a sustained period of time. However, when I increase to 18 degrees inside, it shoots up to 380W and starts going on/off continuously and will never get back to 200W consistent use, while the on/off usage results in 300W+ average use. Very strange.

So I was looking into ways to prevent this, not sure if your system has any options in this area?
- External temperature sensor?
- Some power up script to gradually increase set temperature to 'help' the system stay within the boundaries?

DomiStyle, 2022-01-10:
There is no pre-made board for this repo, it's just a normal ESP8266 or ESP32 dev board. As such it can do everything an ESP can do through ESPHome: https://esphome.io/

If you want a dedicated board you can check out this: https://github.com/jnicolson/panasonic_ac_wifi

As for the software side:

> External temperature sensor

It's possible already but needs to be added either via ESPHome (see https://esphome.io/#sensor-components) or via Home Assistant.

> Some power up script to gradually increase set temperature to 'help' the system stay within the boundaries?

Not sure what exactly you need but you can add automation either via ESPHome directly or via Home Assistant, so pretty much anything is possible.

> However, when I increase to 18 degrees inside, it shoots up to 380W and starts going on/off continuously

How do you measure the energy usage? The energy measurement inside of the unit is not very precise and is just an estimate.

Driekes, 2022-01-10:
Thanks for the feedback, I don't have too much ESP experience yet, but it is cool to see that this is possible, and remove the dependency of the cloud.

- External software temperature sensor, I mean using an external sensor to have the air conditioning decide to turn off/on. Now it is using an internal one which is very sensitive to temperature change of the unit itself of course. So not just an external sensor for additional information, but one that is actively used. I know an external thermostat is possible, so thought maybe with this ESP board you can feed an external source to control the AC
- I will have to dive into ESP possibilities, so that will be another couple of weeks of testing and learning I think :)
- I measure power with Shelly 2.5PM, so there I see the continuous up and down spikes, which are indeed averaged out in the Pana reporting.

donnib, 2022-01-25:
@DomiStyle does is it mean i can in the esphome yaml file point to a sensor i have in HASS that reads the temperature ? The internal temperature sensor quickly reaches the temperature but the room temperature is not there yet so it would be quite usefull that when the climate device comes into HASS that it can use a different temperature sensor.

Driekes, 2022-01-25:
@donnib I did get an advice online to get into the service menu and set parameter 58 (heatshift) to +3. As far as I understand under normal circumstances when you set the thermostat to 20 degrees, it can go up to 22 and shut down.

With this setting it has a bit more range, up to +5 in total if you set the above setting to +3.

It seems the Panasonic can only run in 200W mode when it is set between 16-18 degrees. So with this setting I can set it to 18 and the effective room temperature will be around 21/22. Due to the longer range it has time to reach 200W consistently.

Your experience may be different, but I hope this helps you as well!

donnib, 2022-01-25:
@Driekes ok that sounds interesting, can you link to where and how you access the service menu ?

Driekes, 2022-01-25:
https://www.manualslib.com/manual/1803115/Panasonic-Cs-Tz20wkew.html?page=123#manual code 58, see the manual for instructions how to operate with the remote.

donnib, 2022-01-25:
@Driekes Thx, i see 3C is max to set and i think it's not enough in my case so using an external sensor would yield better results so i am still insterested in getting an answer on my question from @DomiStyle

Driekes, 2022-01-25:
@donnib default is +2, this gives you +3 extra, so +5. You can try, the behavior of the device on my attic has been great, runs of 200W for hours, instead of continuous on/off (and consuming average of over 300W). Temperature recorded it around 20/21 degrees in the room while set at 16 or 17. Silent ánd energy efficient.

donnib, 2022-01-26:
@Driekes Are you running the "summer house mode" or just normal heat and set to 18C ?

Driekes, 2022-01-26:
Normal heat at 16/17 degrees.

DomiStyle, 2022-01-26:
@donnib Yes, you can use the Home Assistant sensor to get an external temperature sensor into ESPHome:
https://esphome.io/components/sensor/homeassistant.html

Of course you have to write the temperature logic yourself.

Driekes, 2022-05-02:
@donnib My current approach is to use a digital potentiometer and put it in parallel (or possibly as replacement at some point of the temperature sensor.). It seems this is a NTC, I just have to find out the curve, so I can set it to a level I need it to be. My skill level is not yet there, so this might take a while. If I succeed, I'll make sure to share the approach.

Driekes, 2022-05-27:
@DomiStyle Just to come back to this topic, as I now understand it a bit better. Is it possible to use an external temp sensor to be used by the Panasonic unit? E.g. the pansonic will no longer use its internal temp sensor, but the external one?
This would be simimlar to using a remote controller I guess. This would allow me to control the target temperature and hopefully also prevent on/off behaviour due to build up of heat around the AC.

DomiStyle, 2022-05-27:
You would have to switch between max and min temperature and heat and cool mode. That should make the AC ignore any internal temperature reading and always heat/cool forever.

Driekes, 2022-05-27:
How would the max/min work? I currently use manual heat or cool selection.

My main goal is to allow the AC to work in the 200W range continuously and currently that is only possible when heating is set between 16 and 18 degrees, but that is not warm enough. 

A bit more detail:
- E.g. I want to have it 21 degrees, I can set the AC to heat to 21, but if the outside temp is high enough it will continuously  go on and off and the minimum power consumption is 400W. I would like to find a situation in which i can have a continuous 200W mode. The only way I currently know is to interfere with the internal sensor creating an offset with a potentiometer, but it would be even better if we can use a (fake) external sensor as input for the AC directly. A lot more elegant :)

And I read that MHI allows this, using the interface that is otherwise use for external (thermostat) control, so that is why I checked here again! I hope my story makes sense.

HA-TB303, 2022-07-23:
We have changed our hardware design a little to fit a DS18B20 1-Wire Temperature Sensor like this:
[image]

If enough people are interested we could run a small batch with this version.

[image]

This would fit the same small case we shared here: https://espthings.ip/p11

Cheers,

Michel

Driekes, 2022-07-26:
I already bought the existing version, which works excellent, so no need for this, appreciate the effort!

My goal was mainly to have the AC use a different source for temperature, because due to the AC location the difference in room and AC can be a couple degrees. I don't think it is resolved with the above solution.

So my alternative approach (that I haven't tested yet) is to add a programmable resistence and put it in a serial way to the AC temperature measurement so I can offset it by a couple degrees. I have seen one do it with a manual resistance so a programmable one I just still need to test.

HA-TB303, 2022-07-26:
Why don't you just get the correct temperature from an external HA temperature sensor en feed it to the correct esphome entity?

Driekes, 2022-07-26:
I need to feed this temperature to the AC to use as reference. I do not think it is possible to do that currently.

It would be perfect if I could use a HA temperature as source for the AC. I know I can get it onto the ESP, but not as external override for the AC. E.g. it would be possible to use the current AC internal temperature sensor and just extend it with a cable to the position I need it. But if possible I would like to prevent it, as I can't hide the cable.

PS: I know that there are Panasonic/other brands that have the option to use wired external temperature sources to use as reference, but I'm not aware if they exist from my TZ model and/or if this is possible through the ESP board.

PS2: I feel I'm explaining it slightly incorrect because the answers are always a bit our of line with my expectations, I hope this one clarifies it a bit more :)

marnixt, 2022-11-07:
is there any news about using an external temperature sensor for steering the airco. I have some serious issues with the location of my unit. In no-time the temperature reported is 25 or more causing the unit to stop before the room temperature is reached.

Wired remote controllers can be attached via the CN-RMT port. I expect this port should be able to receive a temperature but I have not confirmed this yet. 

alternatively I wonder if we could replace the internal temperature sensor using a long cable (I would manage to get it somewhere else with the use of some drilling). Would that be something like this?
https://www.ebay.com/itm/284409985888

best option of course would be if we can somehow feed this external temperature via the CN-CNT connector but I'm not sure this is even possible. If the hardware supports it at least we could maybe try making it in software

marnixt, 2022-11-13:
I did some more investigations (way too much in fact). It seems like the RAC (domestic) units only support the simple CZ-RD514C wired remote. This one does not measure the temperature. 
Only the more advanced CZ-RTC4, CZ-RTC5 etc. wired remotes which are only for non-residential (ECOi & PACi / VRF) seem to be able to be used as an external temperature sensor. They are connected via the RC1/RC2 port apparently:
[image]

anyway, now it looks like I'm royally f*cked. I will contact Panasonic and my installer again.

hppi-johnsson, 2022-11-22:
@**Marnixt**: The temperature sensor is right under the hood in front of the heat exchanger. Why don't you just cut the cable, extend it and place it somewhere else?

I also want my unit to run at 200W constant (to utilize solar energy for heating as much as possible on cloudy days). The on/off behavior drives me crazy.

I think the best way to completely control the unit is to use an analog output of the ESP32 and feed it right into the sensor input of the indoor unit. The resistance curve of the NTC sensor is published in the service manual for the indoor unit.

A simpler approach is to use a fix resistor to let the unit think it is always somewhere between 16-18 degrees in the room (based on your statement above in the thread to allow 200W operation). Then you can use the temperature setting to directly control the power of the unit.

Driekes, 2022-11-22:
@hppi-johnsson I have ordered a digital pot meter so I can control the resistance and thus the temperature that the AC is measuring. My inspiration comes from here: https://community.home-assistant.io/t/control-digital-potentiometer-through-home-assistant-via-esphome/142704/48

However, I have no idea if this works, but will get back here as soon as I have it working. Basically the AC will be set to 16 always and I can manage the 'measured temperature' to control the AC. So a bit reverse :p

hppi-johnsson, 2022-11-22:
Off course it will work!

The digital potentiometer approach is not necessary in our case. It is enough to set a voltage on one of the wires of the NTC sensor. The ESP32 can do this without a potentiometer.
The voltage over R26 needs to be set:
[image]

hppi-johnsson, 2022-11-22:
Yes, R62 off course.

ESP32 has a built in DAC:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/dac.html

I would completely remove the NTC thermistor, but it can stay in the circuit since it has a relatively large resistance. The ESP32 analog output would simply force the voltage to the set value.

The service manual specifies the NTC resistor to have 20k at 20 deg. That means the voltage over R62 is 2.14 V (5V*15k/(15k+20k)).
At 15 deg. the value is 25k --> 1.875V.
[image]

NTC thermistor are usually modeled with the Steinhart–Hart Approximation. So you can write code to perfectly mimic the behavior of the NTC thermistor. Here is some further info: https://makersportal.com/blog/2019/1/15/arduino-thermistor-theory-calibration-and-experiment

marnixt, 2022-11-22:
> @**Marnixt**: The temperature sensor is right under the hood in front of the heat exchanger. Why don't you just cut the cable, extend it and place it somewhere else?
>
> I also want my unit to run at [...]

my 2nd unit (7.1kW) behaves better now, but even with the lower temperatures in the Netherlands now the 3.5kW in the small room still jumps into on/off cycles if you set it even at 20 degrees or lower. And thank god we have parameter 58 (thermo shift adjustment) otherwise the unit would be practically unusabIe. 
So I did discuss this with my installer and they in turn talked with Panasonic technical people. The result was something like this "we discussed moving the intake air temp sensor but they advise against it because incorrect or too long running times because you would have to put the sensor in a cold corner".

I don't know but that seems exactly what we need.
My esp32 board should be arriving any moment now (https://www.espthings.io/index.php/2021/08/03/panasonic-climate-wifi-interface-for-esphome-and-home-assistant/) and using the DAC for directly steering the temperature seems like the perfect solution. 

So now I'm wondering if I should 
- push my installer to move the temp sensor, 
- move the temp sensor myself
- wait / try the ESP32 solution.
Is there any risk blowing up the unit using the ESP32 DAC?

hppi-johnsson, 2022-11-23:
The ESP32 DAC delivers 0-3.3V, which is within the standard range of the temperature sensor. I will try this myself in the next days.

However, the DAC outputs GPIO25 & 26 don't seem to be wired out on the board you have ordered. You would need to get the signal directly from the ESP32 IC by soldering a wire to it.

Thanks for the tip with parameter 58 by the way. Exactly what I needed.

redbullterror, 2022-11-25:
what is the reason that panasonic has son arrangement.
I don't think the 2.5-2.0 kw has this. only the 3.5 kw.
I put a 5k potentiometer between 1 wire of the ntc. and can rotate back 5 degrees. it works well because the original ntc varies with it.

i follow this thread

redbullterror, 2022-11-25:
if you need pictures of the mainboard i have them.
indoor unit and outdoor unit

Git-lop, 2022-12-01:
I recommend not to change anything in relation with the internal NTCs. They are used by the AC processor as they are designed. For example : an overheating condition may be detected internally because the sensors are implemented that way.
I use a Fritzdect440 wireless sensor in conjunction with a Fritzbox router and Node-Red to get the temperature at any place in the room.

StefanTrethan, 2022-12-12:
> I recommend not to change anything in relation with the internal NTCs. They are used by the AC processor as they are designed. For example : an overheating condition may be detected internally [...]

We do not want to measure the room temperature, this is easily done with some other device as you say. What we want is to simulate the Panasonic wired remote, which is normally connected to the AC-CNT connector, and sends a temperature measured somewhere else in the room to the AC so it regulates the temperature better. This is important to get the AC into the correct operating point (power draw) and stop it from cycling.

profigriller, 2022-12-15:
It would be nice to use a MQTT value as a temperature sensor value instead of the build in sensor in the indoor unit. Then we would be very flexible what to use as the source of that value.

StefanTrethan, 2022-12-18:
I have now connected the signal side of the thermistor connection (uppermost pin of CH-TH, the one not connected to GND) to GPIO 25 and added the following code to AC.yaml, this gives me a light control in HA which allows me to set the sensed room temperature from 0-38°C. 

I'm sure this can be made more elegant by someone who knows what they are doing (scaling to a nice numeric input in degree C ect.) but the hardware is perfectly suited to just simulate the temperature directly with the DAC, requiring only a wire and no other parts. If you do improve the code, please let me know.

output:
  - platform: esp32_dac
    pin: GPIO25
    id: dac_indoor_temperature_1
    min_power: 0.4
    
light:
  - platform: monochromatic
    output: dac_indoor_temperature_1
    name: "Panasonic DAC"
    id: dac_indoor_temperature

marnixt, 2022-12-18:
> I have now connected the signal side of the thermistor connection (uppermost pin of CH-TH, the one not connected to GND) to GPIO 25 and added the following code to AC.yaml, this gives me a light [...]

that's brilliant! What did you do with the existing thermistor? Do you care taking a picture?

StefanTrethan, 2022-12-18:
The thermistor is easily driven by the DAC, so I just left it in.
Someone who isn't me could probably change the code to switch GPIO to an input (high-Z), this way the thermistor could be used as factory, or optionally driven by the DAC.

There isn't much to see, but here you go:
https://i.ibb.co/y6qGxBq/pana-dac-side.jpg
https://i.ibb.co/hKLGhXv/pana-dac-front.jpg

Instead of soldering to the pin you could of course tap into the wire whichever way you prefer, or cut it and just connect the ESP in place of the thermistor.

But do turn off the mains power, lest you get bitten by the angry pixies near that area.

Driekes, 2022-12-19:
Super interesting, thanks for the feedback! What do you mean with high-Z?

StefanTrethan, 2022-12-19:
I mean switch it to an input so it does not drive the output pin and instead presents a high impedance.

But guys, this is an ugly hack to make it work, nothing more.

We _know_ you can connect a wired remote to the CN-CNT connector, we know this remote has a temperature sensor, and we know the indoor unit then uses this remote temperature sensor. We also know there is no spare pin (or rather the spare carries +12V), so the remote indoor temperature sensor value is definitely sent digitally via the same RX/TX lines already connected to the ESP32. 

All it takes is for someone to figure out the right header and value to send, probably one of those with a question mark in the description files.

Unfortunately the code in this repository is way too complicated for me, I can't even figure out where the serial packet is put together, or I would change a few bytes and just see what happens. I don't see how all of this is necessary to transmit a few bytes over a serial port, but hey, that's probably because I am not a programmer, I just try stuff until it works.

redbullterror, 2022-12-19:
`ID
18 | 16 | Current temperature
19 | 19 | Outside temperature
20 | FF | Marker
21 | 16 | Current temperature

22 | 19 | Outside temperature
`

why would it send 2 times that value.
or is there one for an external connection

marnixt, 2022-12-19:
> I mean switch it to an input so it does not drive the output pin and instead presents a high impedance.
>
> But guys, this is an ugly hack to make it work, nothing more.
>
> We _know_ you can connect a [...]

as far as my research goes (and what has been confirmed by my installer, although that doesn't mean everything) the consumer etherea models do not support wired remotes with internal temperature sensor (https://github.com/DomiStyle/esphome-panasonic-ac/issues/50#issuecomment-1312679592) I hope to be proven wrong of course.

Driekes, 2022-12-22:
@StefanTrethan how is the behaviour of the AC? To what temperature do you set the value now? Does it work steady and can you somewhat control the AC if it works in low power mode or high power?

StefanTrethan, 2022-12-22:
If you set the AC to 16.5 and 17°C I can get it to run on 220W minimum power and not cycle, which was what I wanted. I can also set it to higher power levels with higher temperatures and/or a greater differential.

However there are many caveats. 

The AC always starts at higher power (600-800W) when you switch it on, and decreases to minimum power over the course of half an hour or so. It can then run at minimum power continuously, but only until it reaches the defrost cycle, after which the whole nonsense starts again. At current conditions around 0°C this means about 1/3 of the time defrost, 1/3 operation at decreasing power, 1/3 steady at minimum, repeating every hour or so. Suffice to say, this sucks, but it is more useful at higher outdoor temps with longer defrost cycles.

Also, indoor temperature is manipulated by the Panasonic with many factors, including outdoor temperature. So it is not a set-and-forget of a specific value and you always get the same operating point, you have to adjust it to find that point.

Overall the Panasonic control algorithm really sucks, doing useless things like defrost with a completely clean and dry condenser. Also the oil pan heater draws 48W below -2°C, which is crazy if you want to switch off your AC in the winter. 

None of that nonsense gets any better with the indoor temperature simulated by ESP, but at least the wild bucking bronco power cycling goes away, and I can attain a lower power steady operating point for some time.

marnixt, 2022-12-22:
I am also disappointed with the performance of the Panasonic algorithm/software. It feels outdated, like something from the early 2000s. Additionally, I too have noticed that the air conditioning unit is defrosting every hour. This seems to be timer-based, as I cannot see a connection to power usage. I would expect that less frosting would occur when the air conditioner is running at a lower capacity, but the Panasonic seems to think differently. As a result it is more energy-efficient to use 60%-70% of the air conditioner's maximum power in order to reduce the relative energy losses caused by the defrosting. In very cold weather, I for sure also need more than the minimum. If 200W is truly all that is needed even when it is below freezing outside, it seems that your unit's capacity might be too large for your room (or you're lucky and have an extremely good isolation).

Regarding the oil pan heating, what would happen if the plug was removed from the outlet? Would it freeze and break down?

StefanTrethan, 2022-12-22:
I just want to use excess solar power and not heat the whole house.

The oil sump heater is to avoid compressor damage by loss of lubrication from absorbed refrigerant that suddenly boils out as the compressor starts up and produces a low pressure. You can remove power, but you should wait for a long time after power is restored before you turn it on, or better yet just wait until it warms up outside. Most if not all AC units have an oil sump heater, some even worse than the Panasonic (some are also on in summer!), but others (Mitsu) claim they automatically wait for the heat to build up after a power outage before they start the compressor, which is smart.

Most customers simply don't care that the software is total crap, or won't even notice. Panasonic has slapped on a wifi module with an atrocius cloud based app and called it good, the rest of the effort went into building it cheaper. SCOP is still good, but it could work _so_ _much_ better overall if they cared a little. 

As a product developer myself I would find it hard to sleep at night with so much compromise, but it does what 80% of the customers look for, and that fairly cheap.

Driekes, 2022-12-22:
On advice of a technician I added a potmeter between the ESP and Pana instead of just a single cable. Reasoning, this will mix the signal from the ESP with the original signal, allowing for the pana to influence a little (or more). I'm currently testing with a 20kOhm potmeter which I have set at 10kOhm and I do see the original values influencing the result, and probably the AC is reacting on that a bit and it slows down to 200W from start within 10 minutes.

Otherwise the AC might think that with 800W power usage temperature is constant, so it will need to work at this level to maintain that temperature (just an assumption of some logic they might have developed.). Maybe that helps others. I'll let you know what my final best working arrangement will be.

My approach as it currently stand is that I will monitor my room thermostat. If it moves in the direction of the desired setpoint I'll leave the AC to it, if it moves away, i'll make sure to correct the AC and increase or decrease the ESP output as needed. Let's see if that gives me an arrangement that works. This way at least I'm able to give it a much more stable measurement and hopefully reduce the on/off behavior.

Git-lop, 2022-12-22:
more information here:
https://www.hrv.co.nz/content/images/uploads/2021/07/Panasonic-Heat-Pump-Defrost-Operation-Explained.pdf

StefanTrethan, 2022-12-23:
Thanks.
This explains what the unit should do, but what it doesn't say is that they don't actually detect blocked airflow or ice, instead they simply estimate the need for defrost by some temperature and time based algorithm that is mistaken more often than not.

hppi-johnsson, 2022-12-23:
Has anyone though about overriding the outside temperature sensor as well?
It probably has a strong influence on the whole process, so the machine could start behaving irregularly.
But at least one could avoid heating the oil sump when the machine is turned off.

Driekes, 2022-12-29:
Well I'm quite happy with it! Have a 5kOhm resistor between the ESP and Panasonic, which provides a bit of feedback from the machine next to the signal I'm sending.

I found out that around 82% the unit will run stable at 200W. So now I use a room thermostat, and use Home Assistant to steer it. What it will do is with a temperature between setpoint and setpoint +0,5, it will run at 200W. Anything above will be set at 90% to turn the unit effectively off. Anything below will be reducing this percentage with 10x the temp difference. So if setpoint is 20.5 and room temp is 20, it wil set it to 77%. This has provided me with a very stable room temperature and quiet operation, and almost no on/off behaviour. So I'm happy!

marnixt, 2022-12-30:
Hi, sounds great but I don't completely get it. You still have the original
thermistor in place but got a connection from esp to Panasonic as well. Why
is the resistor between the esp and Panasonic? What voltage is the esp
sending?

Driekes, 2022-12-30:
@marnixt I built it this way on advice of an electrician, I guess to prevent unwanted currents or so.
However, the advantage is that the ESP is providing the main signal, and the Panasonic still has some influence as well, which might be better for its internal automated function. E.g. if the device starts at 800W, and does not detect a reduction in temperature (due to direct ESP signal) it might continue because it thinks it needs to to keep the room temp stable.

But I don't know, I have tested a bit and this works for me, so I'll not touch it anymore ;)

The voltage (not measured) is 82% of I believe 3,3V, so I guess it is around 2.7V.

hAMMERlized, 2023-04-29:
> Thanks.
> This explains what the unit should do, but what it doesn't say is that they don't actually detect blocked airflow or ice, instead they simply estimate the need for defrost by some temperature and time based algorithm that is mistaken more often than not.

Hi @StefanTrethan 

would you mind getting in contact with me via mail? I have several questions regarding the Pana AC and ESP as I have several Pana AC's myself. I don't want to spam here with several stupid questions :)

Not sure but I guess you are German too? So we could even talk in german via mail. 

Anyway, my mail adress is "s.forys@icloud.com" and I would really appreciate it if you can take some time and help me out. Thanks in advance anyway.

darkfile, 2023-09-07:
Hi @StefanTrethan 

thanks for all the work you've already put into this. I've refactored your code to use a numeric input in Volts directly in HA AND to switch to high impedance mode as soon the voltage drops below a certain value (1V in my case). 

#Enables DAC on GPIO25 to simulate a steady inside temperature. 
number:
  - platform: template
    id: dac_value
    name: "DAC Voltage"
    optimistic: true
    min_value: 0
    max_value: 3.3
    step: 0.01
    on_value:
      then:
        - lambda: |-
            if (x < 1){
              dacWrite(25, 0);
              pinMode(25, INPUT_PULLUP);
            } else {
              pinMode(25, OUTPUT);
              dacWrite(25, (255/3.3)*x); 
            }

Next step would be to not control DACs Voltage in the frontend but the target simulated temperature. This might require some sort of controller in the code. What was your take on that?

mmib, 2024-05-20:
Found this nice thread! Solved this with pure software approach with HA:

1. Add virtual AC switch and temperature regulator (you need HACS virtual component). I don't want to operate with real AC switch, we are going to set temp gently with PID controller.
`switch:
  - platform: virtual
    name: ac_bedroom

input_number:
  ac_bedroom_regulator:
    name: Regulator
    initial: 18
    min: 16
    max: 22
    step: 0.5
`

2. you need to install [SmartThermostat](https://github.com/hacker-cb/hassio-component-smart-thermostat) via HACS
 and odd something like this into your config

`climate:
  - platform: smart_thermostat
    name: bedroom_thermostat
    unique_id: bedroom_thermostat
    target_sensor: sensor.livingroom_temperature
    min_temp: 16
    max_temp: 22
    precision: 0.5
    cooler:
      - entity_id: input_number.ac_bedroom_regulator
        inverted: true
        switch_entity_id: switch.virtual_ac_bedroom
         pid_params: 1.0, 0.002, 0.0
`
where sensor.livingroom_temperature is your external temperature sensor (I use ESPhome with SDC41 CO2 sensor)

3. you need to add 2 automations. First is to set AC target temperature (I use panasonic cpmfort cloud via HACS, can be this esphome-panasonic-ac of course)
`alias: AC-Bedroom
description: ""
trigger:
  - platform: state
    entity_id:
      - input_number.ac_bedroom_regulator
condition: []
action:
  - service: climate.set_temperature
    metadata: {}
    data:
      temperature: "{{ states.input_number.ac_bedroom_regulator.state | float }}"
    target:
      device_id: 9caacLOOOONG_AC_ID_HERE
mode: single`

second is to set initial temp according to regulator
`alias: AC-Bedroom-initial-temp
description: ""
trigger:
  - platform: state
    entity_id:
      - climate.bedroom
    to: cool
    for:
      hours: 0
      minutes: 0
      seconds: 5
condition:
  - condition: numeric_state
    entity_id: input_number.ac_bedroom_regulator
    below: 16.5
action:
  - service: climate.set_temperature
    metadata: {}
    data:
      temperature: "{{ states.input_number.ac_bedroom_regulator.state | float }}"
    target:
      device_id: 9caacLOOOONG_AC_ID_HERE
mode: single`

That's it. You can set temp via homeassitant and virtual PID regulater will controll it
First graph is regulator (16-25 C with 0.5C step, second is real room temperature. AC set to 20C at night)
[image]

marnixt, 2024-05-22:
graph looks nice!

I also steer it using software but I need to take precautions to avoid the AC to shut off prematurely. How is this working in your case as it doesn't seem to use the internally measured temperature of the AC?

some code snippet of the node-RED function node I'm using:

`const hvacMinSetpoint = 16;
const hvacMaxSetpoint = 30;
const hvacKeepOnSetpoint = hvacTemperature - 3.5;
let setpoint;

function calculateSetpointOld() {
    if (hvacState === 'heat') {
        if (tempDiff <= -0.5) return hvacMinSetpoint;
        if (tempDiff <= 0) return hvacKeepOnSetpoint;
        if (tempDiff <= 0.5) return Math.max(desiredSetpoint, hvacKeepOnSetpoint);
        if (tempDiff <= 1.0) return Math.max(desiredSetpoint + tempDiff, hvacKeepOnSetpoint);
        return Math.max(desiredSetpoint + 3 * tempDiff, hvacKeepOnSetpoint);
    } else if (hvacState === 'cool') {
        return desiredSetpoint;
    } else { return hvacMinSetpoint; 
    }
}
function calculateSetpointNew() {
    const adjustmentFactor = 5;
    const responseDelay = 0.5;
    const expectedChange = Math.max(0, derivedTemperature * responseDelay);
    let adjustedSetpoint = desiredSetpoint + (tempDiff - expectedChange) * adjustmentFactor;

    // Ensure the adjusted setpoint is within HVAC operational limits and not below hvacKeepOnSetpoint when heating
    if (hvacState === 'heat') {
        if (tempDiff <= -0.5) { 
            adjustedSetpoint = hvacMinSetpoint;
        } else {
            adjustedSetpoint = Math.max(hvacMinSetpoint, Math.min(adjustedSetpoint, hvacMaxSetpoint), hvacKeepOnSetpoint) }
    } else {
        adjustedSetpoint = desiredSetpoint;
    }

    return adjustedSetpoint;
}
`

willie013, 2024-11-20:
> https://www.manualslib.com/manual/1803115/Panasonic-Cs-Tz20wkew.html?page=123#manual code 58, see the manual for instructions how to operate with the remote.

Can this by done for any model? I have a different model, but when I try to enter the service menu with the set button on the remote nothing happens... I also cannot find any info in the manual about this

franksch50, 2026-10-04:
...or you cut the internal sensor and conect it to a ESP32 and x9c104 and controll the AC with an external wierless sensor.

willie013, 2026-11-04:
I found this product, which claims it can override the internal temp sensor via the CN-HT port:
https://protoart.net/nl/product/panasonic-air-to-air-cncnt-interface-wifi/

https://protoart.net/knowledgebase/pana_cncnt_ata_wifi_installation/#2-toc-title
https://protoart.net/nl/knowledgebase/pana_cncnt_ata_wifi_http_api_mqtt_topics/

Maybe we could reverse engineer this and built it into this integration?