DomiStyle/esphome-panasonic-ac #54: Doesn't work with WEMOS D1 mini

Sashik259, 2022-03-15:
Hi!
Thanks a lot for your great job!
I've took WEMOS D1 mini and done all the instructions, then connected it to CN-CNT port, but it doesn't work. HA integration is OK, but it doesn't control my AC and doesn't receive any info from it when I use its RC. My ac.yaml, photos of soldered parts and logs are attached. Will you help me please to find out what's wrong with it? I'm newbie in such things. I would really appreciate it if you would help me. Thank you.

ac.yaml
````````
esphome:
  name: bedroom-ac
  platform: esp8266
  board: d1_mini

wifi:
  ssid: '*****'
  password: '*****'
  manual_ip:
    static_ip: 192.168.1.154
    gateway: 192.168.1.1
    subnet: 255.255.255.0

logger:
  level: DEBUG
  baud_rate: 0 # ESP8266 only

api:
  password: 'api_password'

ota:
  password: 'api_password'
# Requires ESPHome 1.15.0+ for the even parity option
uart:
  tx_pin: GPIO1
  rx_pin: GPIO3
  baud_rate: 9600
  parity: EVEN

external_components:
  source: github://DomiStyle/esphome-panasonic-ac
  components: [panasonic_ac]

climate:
  - platform: panasonic_ac
    # For CZ-TACG1
    type: cnt

    # For DNSK-P11
    # type: wlan

    name: Panasonic AC
    # horizontal_swing_select:
    #  name: Panasonic AC Horizontal Swing Mode
    vertical_swing_select:
      name: Panasonic AC Vertical Swing Mode
    outside_temperature:
      name: Panasonic AC Outside Temperature

    # Enable as needed
    # eco_switch:
    #   name: Panasonic AC Eco Switch
    # econavi_switch:
    #   name: Panasonic AC Econavi Switch
    # nanoex_switch:
    #   name: Panasonic AC NanoeX Switch
    # mild_dry_switch:
    #   name: Panasonic AC Mild Dry Switch
    # current_power_consumption:
    #   name: Panasonic AC Power Consumption

    # Useful when the ac does not report a current temperature (CZ-TACG1 only)
    # current_temperature_sensor: temperature_sensor_id
````````

logs
````````
INFO Reading configuration /config/esphome/bedroom-ac.yaml...
INFO Updating https://github.com/DomiStyle/esphome-panasonic-ac.git@None
INFO Starting log output from 192.168.1.154 using esphome API
INFO Successfully connected to 192.168.1.154
[19:34:02][I][app:102]: ESPHome version 2022.2.6 compiled on Mar 12 2022, 11:31:42
[19:34:02][C][wifi:491]: WiFi:
[19:34:02][C][wifi:353]:   Local MAC: **:**:**:**:**:**
[19:34:02][C][wifi:354]:   SSID: '*******'[redacted]
[19:34:02][C][wifi:355]:   IP Address: 192.168.1.154
[19:34:02][C][wifi:356]:   BSSID: **:**:**:**:**:**[redacted]
[19:34:02][C][wifi:358]:   Hostname: 'bedroom-ac'
[19:34:02][C][wifi:360]:   Signal strength: -69 dB ▂▄▆█
[19:34:02][C][wifi:364]:   Channel: 1
[19:34:02][C][wifi:365]:   Subnet: 255.255.255.0
[19:34:02][C][wifi:366]:   Gateway: 192.168.1.1
[19:34:02][C][wifi:367]:   DNS1: 192.168.1.1
[19:34:02][C][wifi:368]:   DNS2: 0.0.0.0
[19:34:02][C][logger:233]: Logger:
[19:34:02][C][logger:234]:   Level: DEBUG
[19:34:02][C][logger:235]:   Log Baud Rate: 0
[19:34:02][C][logger:236]:   Hardware UART: UART0
[19:34:02][C][uart.arduino_esp8266:102]: UART Bus:
[19:34:02][C][uart.arduino_esp8266:103]:   TX Pin: GPIO1
[19:34:02][C][uart.arduino_esp8266:104]:   RX Pin: GPIO3
[19:34:02][C][uart.arduino_esp8266:106]:   RX Buffer Size: 256
[19:34:02][C][uart.arduino_esp8266:108]:   Baud Rate: 9600 baud
[19:34:02][C][uart.arduino_esp8266:109]:   Data Bits: 8
[19:34:02][C][uart.arduino_esp8266:110]:   Parity: EVEN
[19:34:02][C][uart.arduino_esp8266:111]:   Stop bits: 1
[19:34:02][C][uart.arduino_esp8266:113]:   Using hardware serial interface.
[19:34:02][C][mdns:084]: mDNS:
[19:34:02][C][mdns:085]:   Hostname: bedroom-ac
[19:34:02][C][ota:085]: Over-The-Air Updates:
[19:34:02][C][ota:086]:   Address: 192.168.1.154:8266
[19:34:02][C][ota:089]:   Using Password.
[19:34:02][C][api:138]: API Server:
[19:34:02][C][api:139]:   Address: 192.168.1.154:6053
[19:34:02][C][api:143]:   Using noise encryption: NO
[19:34:31][D][climate:010]: 'Panasonic AC' - Setting
[19:34:31][D][climate:014]:   Mode: COOL
[19:34:53][D][climate:010]: 'Panasonic AC' - Setting
[19:34:53][D][climate:014]:   Mode: HEAT
[19:35:11][D][climate:010]: 'Panasonic AC' - Setting
[19:35:11][D][climate:014]:   Mode: HEAT_COOL
[19:35:39][D][climate:010]: 'Panasonic AC' - Setting
[19:35:39][D][climate:039]:   Target Temperature: 23.00
[19:35:44][D][climate:010]: 'Panasonic AC' - Setting
[19:35:44][D][climate:039]:   Target Temperature: 22.00
[19:36:52][D][climate:010]: 'Panasonic AC' - Setting
[19:36:52][D][climate:014]:   Mode: COOL
[19:37:01][D][climate:010]: 'Panasonic AC' - Setting
[19:37:01][D][climate:014]:   Mode: COOL
[19:37:19][I][ota:105]: Boot seems successful, resetting boot loop counter.
[19:37:47][D][climate:010]: 'Panasonic AC' - Setting
[19:37:47][D][climate:014]:   Mode: COOL
````````

[image]
[image]
[image]

zioCristia, 2022-03-17:
I had the same problem with wemos d1 mini. 
Try to add this code as mentioned in [the other issue](https://github.com/DomiStyle/esphome-panasonic-ac/issues/5):
```
# Enable logging
logger:
  hardware_uart: UART1 #Uses D4 (GPIO2) as LOG output!
```
Then check also if all the solderings are done properly. However the connections seem to be fine. 

If you don't find the problem I'd suggest to try my solution to it. I've used GPIO13 as Rx and GPIO2 as Tx and it works fine without any problem. You'll have just to desolder and then resolder the rx and the tx cables in esp side. In this way you'll leave the debug to the uart1.

Hope to have helped, let me know if you finally succeed!

This is the code that I used:
```
esphome:
  name: condizionatore-sala

esp8266:
  board: esp01_1m

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:

ota:
  password: "faa26a"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: !secret api_wifi_ssid
    password: !secret api_wifi_password

captive_portal:

uart:
  rx_pin: GPIO13
  tx_pin: GPIO2
  baud_rate: 9600
  parity: EVEN

external_components:
  source: github://DomiStyle/esphome-panasonic-ac
  components: [panasonic_ac]

climate:
  - platform: panasonic_ac
    # For CZ-TACG1
    type: cnt

    name: Panasonic sala
    #horizontal_swing_select:
      #name: Panasonic AC Horizontal Swing Mode
    vertical_swing_select:
      name: Panasonic AC Vertical Swing Mode
    outside_temperature:
      name: Panasonic AC Outside Temperature
```

Sashik259, 2022-03-18:
@zioCristia, I tried it, but unsuccessful. I took another board to exclude its faulty, but no result. Maybe I chose wrong board type in my code?

DomiStyle, 2022-03-19:
I don't have any ESP8266 to test right now but looking at your logs I can't see a single line being logged by the panasonic-ac component, which is not possible if the component is being setup. So there might be something else going on.

There should be a `Panasonic AC component vX.X.X starting...` line at the very beginning followed by the handshake for the CNT interface.

Sashik259, 2022-03-20:
@DomiStyle, It's very sadly to wait boards for a long time, but nothing works then. Can I run your tests? Will you explain me shortly how I can find and fix this bug, please?

Sashik259, 2022-03-29:
Can anyone tell me what board types actually work via CNT-connection?

hpirila, 2022-03-30:
I can confirm this configuration works fine with ESP32CAM module. Still no luck with ESP8266.

```
substitutions:
  devicename:       Living Room AC
  wifi_ip:          192.168.1.210

esphome:
  name: living-room-ac

esp32:
  board: esp32cam

logger:
  level: DEBUG
  baud_rate: 0

api:

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: $wifi_ip
    gateway:   192.168.1.1
    subnet:    255.255.255.0
    dns1:      8.8.8.8
    dns2:      8.8.4.4

uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 9600
  parity: EVEN

external_components:
  source: github://DomiStyle/esphome-panasonic-ac
  components: [panasonic_ac]

climate:
  - platform: panasonic_ac
    type: cnt
    name: $devicename
    vertical_swing_select:
      name: $devicename Vertical Swing Mode
    outside_temperature:
      name: Living Room AC Outside Temperature

    # Enable as needed
    eco_switch:
      name: $devicename Eco Switch
    current_power_consumption:
      name: $devicename AC Power Consumption
```

Sashik259, 2022-03-31:
@hpirila , is it possible to use ESP32Cam with Panasonic-AC component and camera at the same time?
@DomiStyle , should I buy ESP32 оr is there any way to make esp8266 work?

hpirila, 2022-03-31:
I am not using the camera together with Panasonic AC. Overall ESP32CAM gets quite hot when using camera, therefore not ideal inside AC unit.

DomiStyle, 2022-03-31:
@Sashik259 I haven't had time yet to try with a ESP8266, I can only say that ESP32s definitely work.

zioCristia, 2022-03-31:
> @hpirila , is it possible to use ESP32Cam with Panasonic-AC component and camera at the same time?

Until you have a free rx and tx pin you can use them both, but think about the heating problem @hpirila said

> @DomiStyle , should I buy ESP32 оr is there any way to make esp8266 work?

As I already said and as @Expaso said in a [previous issue](https://github.com/DomiStyle/esphome-panasonic-ac/issues/5) you can make it works in an ESP8266, however you may run into some problems. So use an ESP32 if you want to be sure.

I personally bought an ESP8266 11 pins module from [amazon](https://www.amazon.it/gp/product/B093GR7MJM/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1).

hpirila, 2022-04-11:
Try to touch pins with your hand when logging. It should cause enough interference to GPIO03 to get some logging with `[panasonic_ac.cz_tacg xxx]`.

Sashik259, 2022-04-16:
@DomiStyle @hpirila Hi, guys! I've changed logger level to VERY_VERBOSE. With ESP32 chip still have no response from air conditioner. Look at my log, please
```
INFO Reading configuration /config/esphome/bedroom-ac.yaml...
INFO Updating https://github.com/DomiStyle/esphome-panasonic-ac.git@None
INFO Starting log output from 192.168.1.154 using esphome API
INFO Successfully connected to 192.168.1.154
[09:39:22][I][app:102]: ESPHome version 2022.3.2 compiled on Apr 12 2022, 21:38:47
[09:39:22][C][wifi:491]: WiFi:
[09:39:22][C][wifi:353]:   Local MAC: XX:XX:XX:XX:XX:XX
[09:39:22][C][wifi:354]:   SSID: 'myssid'[redacted]
[09:39:22][C][wifi:355]:   IP Address: 192.168.1.154
[09:39:22][C][wifi:357]:   BSSID: XX:XX:XX:XX:XX:5XX[redacted]
[09:39:22][C][wifi:358]:   Hostname: 'bedroom-ac'
[09:39:22][C][wifi:360]:   Signal strength: -55 dB ▂▄▆█
[09:39:22][V][wifi:362]:   Priority: 0.0
[09:39:22][C][wifi:364]:   Channel: 11
[09:39:22][C][wifi:365]:   Subnet: 255.255.255.0
[09:39:22][C][wifi:366]:   Gateway: 192.168.1.1
[09:39:22][C][wifi:367]:   DNS1: 0.0.0.0
[09:39:22][C][logger:233]: Logger:
[09:39:22][C][logger:234]:   Level: VERY_VERBOSE
[09:39:22][C][logger:235]:   Log Baud Rate: 0
[09:39:22][C][logger:236]:   Hardware UART: UART0
[09:39:22][C][uart.arduino_esp32:107]: UART Bus:
[09:39:22][C][uart.arduino_esp32:108]:   TX Pin: GPIO1
[09:39:22][C][uart.arduino_esp32:109]:   RX Pin: GPIO3
[09:39:22][C][uart.arduino_esp32:111]:   RX Buffer Size: 256
[09:39:22][C][uart.arduino_esp32:113]:   Baud Rate: 9600 baud
[09:39:22][C][uart.arduino_esp32:114]:   Data Bits: 8
[09:39:22][C][uart.arduino_esp32:115]:   Parity: EVEN
[09:39:22][C][uart.arduino_esp32:116]:   Stop bits: 1
[09:39:22][C][mdns:084]: mDNS:
[09:39:22][C][mdns:085]:   Hostname: bedroom-ac
[09:39:22][V][mdns:086]:   Services:
[09:39:22][V][mdns:088]:   - _esphomelib, _tcp, 6053
[09:39:22][V][mdns:090]:     TXT: version = 2022.3.2
[09:39:22][V][mdns:090]:     TXT: mac = 7821848da530
[09:39:22][C][ota:086]:   Address: 192.168.1.154:3232
[09:39:22][C][api:138]: API Server:
[09:39:22][C][api:139]:   Address: 192.168.1.154:6053
[09:39:22][C][api:143]:   Using noise encryption: NO
[09:39:23][V][panasonic_ac.cz_tacg1:248]: Polling AC
[09:39:23][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[09:39:28][V][panasonic_ac.cz_tacg1:248]: Polling AC
[09:39:28][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[09:39:28][VV][api.service:355]: on_ping_request: PingRequest {}
[09:39:28][VV][api.service:043]: send_ping_response: PingResponse {}
[09:39:33][V][panasonic_ac.cz_tacg1:248]: Polling AC
[09:39:33][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[09:39:37][VV][api.service:355]: on_ping_request: PingRequest {}
[09:39:37][VV][api.service:043]: send_ping_response: PingResponse {}
[09:39:38][V][panasonic_ac.cz_tacg1:248]: Polling AC
[09:39:38][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[09:39:43][V][panasonic_ac.cz_tacg1:248]: Polling AC
[09:39:43][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[09:39:43][VV][api.service:355]: on_ping_request: PingRequest {}
[09:39:43][VV][api.service:043]: send_ping_response: PingResponse {}
[09:39:48][V][panasonic_ac.cz_tacg1:248]: Polling AC
[09:39:48][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[09:39:48][VV][scheduler:186]: Running interval '' with interval=60000 last_execution=115906 (now=175906)
[09:39:52][VV][api.service:355]: on_ping_request: PingRequest {}
[09:39:52][VV][api.service:043]: send_ping_response: PingResponse {}
[09:39:53][V][panasonic_ac.cz_tacg1:248]: Polling AC
[09:39:53][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[09:39:58][V][panasonic_ac.cz_tacg1:248]: Polling AC
[09:39:58][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[09:39:58][VV][api.service:355]: on_ping_request: PingRequest {}
[09:39:58][VV][api.service:043]: send_ping_response: PingResponse {}
[09:40:03][V][panasonic_ac.cz_tacg1:248]: Polling AC
[09:40:03][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[... 79 lines cut ...]
  has_mode: YES
  mode: CLIMATE_MODE_COOL
  has_target_temperature: NO
  target_temperature: 0
  has_target_temperature_low: NO
  target_temperature_low: 0
  has_target_temperature_high: NO
  target_temperature_high: 0
  has_legacy_away: NO
  legacy_away: NO
  has_fan_mode: NO
  fan_mode: CLIMATE_FAN_ON
  has_swing_mode: NO
  swing_mode: CLIMATE_SWING_OFF
  has_custom_fan_mode: NO
  custom_fan_mode: ''
  has_
[09:40:58][D][climate:010]: 'Panasonic AC' - Setting
[09:40:58][D][climate:014]:   Mode: COOL
[09:40:58][V][panasonic_ac.cz_tacg1:248]: Polling AC
[09:40:58][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[09:40:58][VV][api.service:355]: on_ping_request: PingRequest {}
[09:40:58][VV][api.service:043]: send_ping_response: PingResponse {}

```

Sashik259, 2022-04-16:
And here is OTA upload log. There are some yellow marked warnings - maybe it's something important to make debugging easier?
```

INFO Reading configuration /config/esphome/bedroom-ac.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing bedroom-ac (board: wemos_d1_mini32; framework: arduino; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <WiFi> 1.0
|-- <ESPmDNS> 1.0
|-- <Update> 1.0
Compiling /data/bedroom-ac/.pioenvs/bedroom-ac/src/esphome/components/api/api_connection.cpp.o
Compiling /data/bedroom-ac/.pioenvs/bedroom-ac/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/bedroom-ac/.pioenvs/bedroom-ac/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/bedroom-ac/.pioenvs/bedroom-ac/src/esphome/components/api/api_pb2_service.cpp.o
Compiling /data/bedroom-ac/.pioenvs/bedroom-ac/src/esphome/components/api/api_server.cpp.o
Compiling /data/bedroom-ac/.pioenvs/bedroom-ac/src/esphome/components/api/list_entities.cpp.o
Compiling /data/bedroom-ac/.pioenvs/bedroom-ac/src/esphome/components/api/proto.cpp.o
Compiling /data/bedroom-ac/.pioenvs/bedroom-ac/src/esphome/components/api/subscribe_state.cpp.o
Compiling /data/bedroom-ac/.pioenvs/bedroom-ac/src/esphome/components/api/user_services.cpp.o
Compiling /data/bedroom-ac/.pioenvs/bedroom-ac/src/esphome/components/climate/climate.cpp.o
Compiling /data/bedroom-ac/.pioenvs/bedroom-ac/src/esphome/components/climate/climate_mode.cpp.o
Compiling /data/bedroom-ac/.pioenvs/bedroom-ac/src/esphome/components/climate/climate_traits.cpp.o
Compiling /data/bedroom-ac/.pioenvs/bedroom-ac/src/esphome/components/esp32/core.cpp.o
Compiling /data/bedroom-ac/.pioenvs/bedroom-ac/src/esphome/components/esp32/gpio_arduino.cpp.o
Compiling /data/bedroom-ac/.pioenvs/bedroom-ac/src/esphome/components/esp32/gpio_idf.cpp.o
Compiling /data/bedroom-ac/.pioenvs/bedroom-ac/src/esphome/components/esp32/preferences.cpp.o
Compiling /data/bedroom-ac/.pioenvs/bedroom-ac/src/esphome/components/logger/logger.cpp.o
Compiling /data/bedroom-ac/.pioenvs/bedroom-ac/src/esphome/components/md5/md5.cpp.o
Compiling /data/bedroom-ac/.pioenvs/bedroom-ac/src/esphome/components/mdns/mdns_component.cpp.o
Compiling /data/bedroom-ac/.pioenvs/bedroom-ac/src/esphome/components/mdns/mdns_esp32_arduino.cpp.o
Compiling /data/bedroom-ac/.pioenvs/bedroom-ac/src/esphome/components/mdns/mdns_esp8266.cpp.o
Compiling /data/bedroom-ac/.pioenvs/bedroom-ac/src/esphome/components/mdns/mdns_esp_idf.cpp.o
Compiling /data/bedroom-ac/.pioenvs/bedroom-ac/src/esphome/components/network/util.cpp.o
Compiling /data/bedroom-ac/.pioenvs/bedroom-ac/src/esphome/components/ota/ota_backend_arduino_esp32.cpp.o
Compiling /data/bedroom-ac/.pioenvs/bedroom-ac/src/esphome/components/ota/ota_backend_arduino_esp8266.cpp.o
Compiling /data/bedroom-ac/.pioenvs/bedroom-ac/src/esphome/components/ota/ota_backend_esp_idf.cpp.o
Compiling /data/bedroom-ac/.pioenvs/bedroom-ac/src/esphome/components/ota/ota_component.cpp.o
Compiling /data/bedroom-ac/.pioenvs/bedroom-ac/src/esphome/components/panasonic_ac/esppac.cpp.o
Compiling /data/bedroom-ac/.pioenvs/bedroom-ac/src/esphome/components/panasonic_ac/esppac_cnt.cpp.o
Compiling /data/bedroom-ac/.pioenvs/bedroom-ac/src/esphome/components/panasonic_ac/esppac_wlan.cpp.o
In file included from src/esphome/components/climate/climate.h:7:0,
                 from src/esphome/components/panasonic_ac/esppac.h:3,
                 from src/esphome/components/panasonic_ac/esppac.cpp:1:
src/esphome/components/panasonic_ac/esppac.cpp: In member function 'void esphome::panasonic_ac::PanasonicAC::log_packet(std::vector<unsigned char>, bool)':
src/esphome/components/panasonic_ac/esppac.cpp:245:43: warning: 'std::__cxx11::string esphome::hexencode(const T&) [with T = std::vector<unsigned char>; std::__cxx11::string = std::__cxx11::basic_string<char>]' is deprecated: hexencode() is deprecated, use format_hex_pretty() instead. [-Wdeprecated-declarations]
     ESP_LOGV(TAG, "TX: %s", hexencode(data).c_str());
                                           ^
src/esphome/core/log.h:83:91: note: in definition of macro 'esph_log_v'
   esp_log_printf_(ESPHOME_LOG_LEVEL_VERBOSE, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
                                                                                           ^
src/esphome/components/panasonic_ac/esppac.cpp:245:5: note: in expansion of macro 'ESP_LOGV'
     ESP_LOGV(TAG, "TX: %s", hexencode(data).c_str());
     ^
In file included from src/esphome/components/climate/climate.h:5:0,
                 from src/esphome/components/panasonic_ac/esppac.h:3,
                 from src/esphome/components/panasonic_ac/esppac.cpp:1:
src/esphome/core/helpers.h:648:13: note: declared here
 std::string hexencode(const T &data) {
             ^
In file included from src/esphome/components/climate/climate.h:7:0,
                 from src/esphome/components/panasonic_ac/esppac.h:3,
                 from src/esphome/components/panasonic_ac/esppac.cpp:1:
src/esphome/components/panasonic_ac/esppac.cpp:247:43: warning: 'std::__cxx11::string esphome::hexencode(const T&) [with T = std::vector<unsigned char>; std::__cxx11::string = std::__cxx11::basic_string<char>]' is deprecated: hexencode() is deprecated, use format_hex_pretty() instead. [-Wdeprecated-declarati [... cut ...]
[10:09:52][V][panasonic_ac.cz_tacg1:248]: Polling AC
[10:09:52][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[10:09:56][VV][api.service:355]: on_ping_request: PingRequest {}
[10:09:56][VV][api.service:043]: send_ping_response: PingResponse {}
[10:09:57][V][panasonic_ac.cz_tacg1:248]: Polling AC
[10:09:57][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[10:09:57][VV][api.service:355]: on_ping_request: PingRequest {}
[10:09:57][VV][api.service:043]: send_ping_response: PingResponse {}

```

hpirila, 2022-04-16:
Hi,

Just now I tried with
```
logger:
  level: VERY_VERBOSE
```
But it didn't work any more with this setting. In homeassistant it showed device unavailable.

```
[14:42:40][C][template.sensor:024]:   Update Interval: 60.0s
[14:42:40][V][panasonic_ac.cz_tacg1:248]: Polling AC
[14:42:40][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[14:42:40][V][component:199]: Component panasonic_ac.climate took a long time for an operation (0.17 s).
[14:42:40][V][component:200]: Components should block for at most 20-30ms.
[14:42:40][VV][api.service:337]: on_disconnect_request: DisconnectRequest {}
[14:42:41][D][api.connection:159]: Home Assistant 2022.4.3 (::FFFF:C0A8:1C8) requested disconnected
[14:42:41][VV][api.service:031]: send_disconnect_response: DisconnectResponse {}
[14:42:41][VV][api.service:132]: send_list_entities_sensor_response: ListEntitiesSensorResponse {
  object_id: 'bedroom_ac_power_consumption'
  key: 2250203900
```

I am using same pins as you. So change to this setting. Remember baud_rate: 0
```
uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 9600
  parity: EVEN
```
```
logger:
  level: DEBUG
  baud_rate: 0
```

One easy troubleshooting is to change to:

```
uart:
  rx_pin: GPIO1
  tx_pin: GPIO3
  baud_rate: 9600
  parity: EVEN
```
Just in case tx / rx wires are crossed.

Sashik259, 2022-04-16:
> Just in case tx / rx wires are crossed.
@hpirila Soldering is OK. If I cross tx / rx wires then tx-tx pins will connect each other. I think it will damage ESP32 or AC board

Sashik259, 2022-04-16:
@blacklizard @bobwho69 @jesserockz @sunsetdk @trsqr Hi, guys! I found you forking this repository. Can anyone of you help me to fix this bug, please? I newbie in this.

sunsetdk, 2022-04-16:
I ended giving up with the WEMOS D1 mini and went with a generic ESP32. (not really  helpfull)

Sashik259, 2022-04-16:
@sunsetdk I've changed my **WEMOS D1 mini** to **WEMOS D1 mini ESP32**. But it doesn't work too. What type of board do you use?

zioCristia, 2022-04-16:
I don't know if this could be helpful but this is my log output with a working esp8266 mini:

```
INFO Reading configuration /config/esphome/condizionatore-sala.yaml...
INFO Starting log output from condizionatore-sala.local using esphome API
INFO Successfully connected to condizionatore-sala.local
[14:29:19][I][app:102]: ESPHome version 2022.3.1 compiled on Apr 15 2022, 22:29:03
[14:29:19][C][wifi:491]: WiFi:
[14:29:19][C][wifi:353]:   Local MAC: -----
[14:29:19][C][wifi:354]:   SSID: -----
[14:29:19][C][wifi:355]:   IP Address: -----
[14:29:19][C][wifi:356]:   BSSID: -----
[14:29:19][C][wifi:358]:   Hostname: 'condizionatore-sala'
[14:29:19][C][wifi:360]:   Signal strength: -51 dB ▂▄▆█
[14:29:19][C][wifi:364]:   Channel: 5
[14:29:19][C][wifi:365]:   Subnet: 255.255.255.0
[14:29:19][C][wifi:366]:   Gateway: 192.168.1.1
[14:29:19][C][wifi:367]:   DNS1: 192.168.1.1
[14:29:19][C][wifi:368]:   DNS2: 0.0.0.0
[14:29:19][C][logger:233]: Logger:
[14:29:19][C][logger:234]:   Level: DEBUG
[14:29:19][C][logger:235]:   Log Baud Rate: 115200
[14:29:19][C][logger:236]:   Hardware UART: UART0
[14:29:19][C][uart.arduino_esp8266:102]: UART Bus:
[14:29:19][C][uart.arduino_esp8266:103]:   TX Pin: GPIO2
[14:29:19][C][uart.arduino_esp8266:104]:   RX Pin: GPIO13
[14:29:19][C][uart.arduino_esp8266:106]:   RX Buffer Size: 256
[14:29:19][C][uart.arduino_esp8266:108]:   Baud Rate: 9600 baud
[14:29:19][C][uart.arduino_esp8266:109]:   Data Bits: 8
[14:29:19][C][uart.arduino_esp8266:110]:   Parity: EVEN
[14:29:19][C][uart.arduino_esp8266:111]:   Stop bits: 1
[14:29:19][C][uart.arduino_esp8266:115]:   Using software serial
[14:29:19][C][captive_portal:088]: Captive Portal:
[14:29:19][C][mdns:084]: mDNS:
[14:29:20][C][mdns:085]:   Hostname: condizionatore-sala
[14:29:20][C][ota:085]: Over-The-Air Updates:
[14:29:20][C][ota:086]:   Address: condizionatore-sala.local:8266
[14:29:20][C][ota:089]:   Using Password.
[14:29:20][C][api:138]: API Server:
[14:29:20][C][api:139]:   Address: condizionatore-sala.local:6053
[14:29:20][C][api:143]:   Using noise encryption: NO
[14:29:21][D][climate:385]: 'Panasonic sala' - Sending state:
[14:29:22][D][climate:388]:   Mode: OFF
[14:29:22][D][climate:396]:   Custom Fan Mode: Automatic
[14:29:22][D][climate:402]:   Custom Preset: Normal
[14:29:22][D][climate:405]:   Swing Mode: OFF
[14:29:22][D][climate:408]:   Current Temperature: 22.00°C
[14:29:22][D][climate:414]:   Target Temperature: 27.50°C
[14:29:26][D][climate:385]: 'Panasonic sala' - Sending state:
[14:29:26][D][climate:388]:   Mode: OFF
[14:29:26][D][climate:396]:   Custom Fan Mode: Automatic
[14:29:26][D][climate:402]:   Custom Preset: Normal
[14:29:26][D][climate:405]:   Swing Mode: OFF
[14:29:26][D][climate:408]:   Current Temperature: 22.00°C
[14:29:26][D][climate:414]:   Target Temperature: 27.50°C
[14:30:42][D][climate:010]: 'Panasonic sala' - Setting
[14:30:42][D][climate:014]:   Mode: HEAT
```

As you can see every 5s I receive the status informations from the conditioner and if I click in a button of the climate card I send the regarding information to the conditioner.
However I don't have any `Panasonic AC component vX.X.X starting...` as @DomiStyle said, I'm not sure if we should see that line with CZ-TACG.

The code in the esp is the one in my [previous post](https://github.com/DomiStyle/esphome-panasonic-ac/issues/54#issuecomment-1071213800).

sunsetdk, 2022-04-16:
> @DomiStyle @zioCristia I've got Wemos D1 mini ESP32. Here's yaml
> 
> ```
> esphome:
>   name: bedroom-ac
> esp32:
>   board: wemos_d1_mini32
> 
> wifi:
>   ssid: 'myssid'
>   password: 'mypass'
>   manual_ip:
>     static_ip: 192.168.1.154
>     gateway: 192.168.1.1
>     subnet: 255.255.255.0
> 
> logger:
>   level: DEBUG
>   baud_rate: 0 # ESP8266 only
> 
> api:
>   password: 'api_password'
> 
> ota:
>   password: 'api_password'
> # Requires ESPHome 1.15.0+ for the even parity option
> uart:
>   tx_pin: GPIO1
>   rx_pin: GPIO3
>   baud_rate: 9600
>   parity: EVEN

Are these the right pins?

sunsetdk, 2022-04-16:
Do you have a multimeter? If so i would suggest you test the converter interface.

Power up the converter with 3 and 5 volt. 
Then put the TX from EPS to 3 volt and see that the corosponding pin on the HV side goes to 5 volt. 
Put it to gnd and see that HV side goes low

same with the TX on HV side.
put it to 5 volt, check that LV side goes to 3 volt
put to gnd and check that LV side goes low.

Sashik259, 2022-04-16:
@DomiStyle @sunsetdk Got something. What does such logs mean?
```
[17:22:19][W][panasonic_ac.cz_tacg1:259]: Dropping invalid packet (length)
[17:22:34][W][panasonic_ac.cz_tacg1:267]: Dropping invalid packet (header)
[17:23:29][W][panasonic_ac.cz_tacg1:259]: Dropping invalid packet (length)
[17:23:34][W][panasonic_ac.cz_tacg1:267]: Dropping invalid packet (header)
[17:24:04][W][panasonic_ac.cz_tacg1:259]: Dropping invalid packet (length)
[17:24:14][W][panasonic_ac.cz_tacg1:259]: Dropping invalid packet (length)
```

sunsetdk, 2022-04-16:
If i recall this looks like when tried using a WEMOS D1 mini.
Is this with the ESP32?
is this with very_verbose logging?

Sashik259, 2022-04-16:
@sunsetdk Wemos D1 mini and mini esp32 didn't write panasonic_ac.cz_tacg1 logs. This is ESP8266 (ESP01_1m). Logger level DEBUG

Sashik259, 2022-04-16:
> I personally bought an ESP8266 11 pins module from [amazon](https://www.amazon.it/gp/product/B093GR7MJM/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1).
@zioCristia Do you use this board via cnt or wlan connection? If it's cnt - show me your config please.

sunsetdk, 2022-04-16:
> @DomiStyle @sunsetdk Got something. What does such logs mean?
>
> ```
> [17:22:19][W][panasonic_ac.cz_tacg1:259]: Dropping invalid packet (length)
> [17:22:34][W][panasonic_ac.cz_tacg1:267]: Dropping [...]

Short version. There is data, but no good data. Did you check your config on the esp32 with the pins (i think you used the wrong pins). 
Could you check the hardware like i suggested?

Sashik259, 2022-04-16:
> Short version. There is data, but no good data. Did you check your config on the esp32 with the pins (i think you used the wrong pins). Could you check the hardware like i suggested?

@sunsetdk With  [ESP8266 (ESP01_1m)](https://ae04.alicdn.com/kf/HTB16dKFaVzsK1Rjy1Xbq6xOaFXah/CH340-USB-ESP8266-ESP-01-ESP-01S-ESP01-ESP01S-Wi-Fi.jpg_640x640.jpg)+ [ESP01 adapter](https://ae04.alicdn.com/kf/HTB1a7FcSpXXXXcAXXXXq6xXFXXXT/Wi-Fi-ESP8266-ESP.jpg) I use GPIO1 as Tx and GPIO3 as Rx. And I have no multimeter. I can use another board and adapter

jesserockz, 2022-04-20:
I actually use a D1 mini directly connected to the unit cables. No logic level converter in between with no issues.

Here is my config: https://github.com/jesserockz/esphome-configs/blob/main/modules/devices/panasonic-ac.yaml

Sashik259, 2022-04-20:
@jesserockz Thanks a lot for answering. I see you use esp01_1m instead of d1_mini as board type - I didn't know it was possible. And you use tx_pin: 14 and rx_pin: 12 - is it physical number or GPIO14 and GPIO12 ? Can you send a picture of wires soldered to d1mini, please?

jesserockz, 2022-04-20:
[image]

Sashik259, 2022-04-24:
> I actually use a D1 mini directly connected to the unit cables. No logic level converter in between with no issues.
> 
> Here is my config: https://github.com/jesserockz/esphome-configs/blob/main/modules/devices/panasonic-ac.yaml

@jesserockz THANKS A LOT!!!! Your config works for me with Wemos D1 mini. I really don't understand why tx_pin: 14 and rx_pin: 12 work instead of 1 and 3 pins and board type esp01_1m instead of d1_mini, but IT WORKS!!! Thank you very much once again for your help!

Sashik259, 2022-04-24:
@DomiStyle The outside temperature sensor always shows 0 for both of my ACs, but the real outside temperature is about 15-17 degrees above 0. How can I use data from temperature sensor of outdoor AC-unit?
And indoor unit has Eco-Navi presence sensor - is it possible to read its state?

closed 2022-04-24 by Sashik259

Schaussi, 2024-06-17:
> I actually use a D1 mini directly connected to the unit cables. No logic level converter in between with no issues.
> 
> Here is my config: https://github.com/jesserockz/esphome-configs/blob/main/modules/devices/panasonic-ac.yaml

Panasonic changed auth mechanism today, which currently causes cloud access to fail, e.g. when using HA/HACS Panasonic Comfort Cloud project.
So I searched for a cloudless solution and found this - set it up with a Wemos D1 Mini - worked flawlessly in less than 2 hours! (Just had to swap TX/RX wires)
Thank you guys for this project and your work.