DomiStyle/esphome-panasonic-ac #56: ECO mode not functioning CZ-TACG1

magnocapite, 2022-04-24:
I have just replaced a CZ-TACG1 with a ESP32 mini and am loving getting away from the cloud. Everything is working well apart from the eco mode selection. Each time I try and turn eco mode on the switch toggles on then straight back off again with no change in mode on the aircon unit itself.

Extract from log:
```
[12:57:19][D][switch:013]: 'Panasonic AC Eco Switch' Turning ON.
[12:57:19][D][switch:037]: 'Panasonic AC Eco Switch': Sending state ON
[12:57:20][D][switch:037]: 'Panasonic AC Eco Switch': Sending state OFF
```

DomiStyle, 2022-04-24:
Can you enable verbose logging (not very verbose) and toggle eco mode back and forth a few times via the IR remote?

Make sure to take not which log entries correspond to which eco state on the AC.

magnocapite, 2022-04-25:
Ok so I started by turning eco on at 10:37:00 and toggling every 10sec for a minute. Seems to be associated with RX#13 although it does also seem to depend what mode it's in and the temperature target. While in heat mode and set at 24 eco on is 31 and eco off is 37.

As far as I can tell:
cool mode eco off = target * 2+ 2
cool mode eco on = target * 2 + 6
heat mode eco off = target * 2 + 7
heat mode eco on = target * 2 + 1

When in heat/cool mode it changes depending on whether it's actively heating or cooling and dry mode seems to be acting the same as cooling.

```
[10:36:59][V][panasonic_ac.cz_tacg1:248]: Polling AC
[10:36:59][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[10:36:59][V][panasonic_ac:247]: RX: 70.20.44.30.80.A0.FD.00.00.40.00.00.4C.37.00.00.30.90.14.17.FF.18.17.FF.80.80.FF.80.C5.01.0C.80.41.57.9B (35)
[10:36:59][V][sensor:074]: 'Panasonic AC Power Consumption': Received new state 441.000000
[10:36:59][D][sensor:125]: 'Panasonic AC Power Consumption': Sending state 441.00000 W with 0 decimals of accuracy
[10:36:59][D][climate:385]: 'Panasonic AC' - Sending state:
[10:36:59][D][climate:388]:   Mode: HEAT
[10:36:59][D][climate:396]:   Custom Fan Mode: Automatic
[10:36:59][D][climate:402]:   Custom Preset: Normal
[10:36:59][D][climate:405]:   Swing Mode: BOTH
[10:36:59][D][climate:408]:   Current Temperature: 20.00°C
[10:36:59][D][climate:414]:   Target Temperature: 24.00°C
[10:36:59][V][component:199]: Component panasonic_ac.climate took a long time for an operation (0.09 s).
[10:36:59][V][component:200]: Components should block for at most 20-30ms.
[10:37:04][V][panasonic_ac.cz_tacg1:248]: Polling AC
[10:37:04][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[10:37:04][V][panasonic_ac:247]: RX: 70.20.44.30.80.A0.FD.00.00.40.00.00.4C.31.00.00.30.90.14.17.FF.18.17.FF.80.80.FF.80.DB.01.0D.C0.00.00.E2 (35)
[10:37:04][V][sensor:074]: 'Panasonic AC Power Consumption': Received new state 462.000000
[10:37:04][D][sensor:125]: 'Panasonic AC Power Consumption': Sending state 462.00000 W with 0 decimals of accuracy
[10:37:04][D][climate:385]: 'Panasonic AC' - Sending state:
[10:37:04][D][climate:388]:   Mode: HEAT
[10:37:04][D][climate:396]:   Custom Fan Mode: Automatic
[10:37:04][D][climate:402]:   Custom Preset: Normal
[10:37:04][D][climate:405]:   Swing Mode: BOTH
[10:37:04][D][climate:408]:   Current Temperature: 20.00°C
[10:37:04][D][climate:414]:   Target Temperature: 24.00°C
[10:37:04][V][component:199]: Component panasonic_ac.climate took a long time for an operation (0.09 s).
[10:37:04][V][component:200]: Components should block for at most 20-30ms.
[10:37:09][V][panasonic_ac.cz_tacg1:248]: Polling AC
[10:37:09][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[10:37:09][V][panasonic_ac:247]: RX: 70.20.44.30.80.A0.FD.00.00.40.00.00.4C.31.00.00.30.90.14.17.FF.18.17.FF.80.80.FF.80.AF.01.0C.80.41.57.B7 (35)
[10:37:09][V][sensor:074]: 'Panasonic AC Power Consumption': Received new state 419.000000
[10:37:09][D][sensor:125]: 'Panasonic AC Power Consumption': Sending state 419.00000 W with 0 decimals of accuracy
[10:37:09][D][climate:385]: 'Panasonic AC' - Sending state:
[10:37:09][D][climate:388]:   Mode: HEAT
[10:37:09][D][climate:396]:   Custom Fan Mode: Automatic
[10:37:09][D][climate:402]:   Custom Preset: Normal
[10:37:09][D][climate:405]:   Swing Mode: BOTH
[10:37:09][D][climate:408]:   Current Temperature: 20.00°C
[10:37:09][D][climate:414]:   Target Temperature: 24.00°C
[10:37:09][V][component:199]: Component panasonic_ac.climate took a long time for an operation (0.09 s).
[10:37:09][V][component:200]: Components should block for at most 20-30ms.
[10:37:14][V][panasonic_ac.cz_tacg1:248]: Polling AC
[10:37:14][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[10:37:14][V][panasonic_ac:247]: RX: 70.20.44.30.80.A0.FD.00.00.40.00.00.4C.37.00.00.30.90.14.17.FF.18.17.FF.80.80.FF.80.9A.01.0B.C0.00.00.1F (35)
[10:37:14][V][sensor:074]: 'Panasonic AC Power Consumption': Received new state 399.000000
[10:37:14][D][sensor:125]: 'Panasonic AC Power Consumption': Sending state 399.00000 W with 0 decimals of accuracy
[10:37:14][D][climate:385]: 'Panasonic AC' - Sending state:
[10:37:14][D][climate:388]:   Mode: HEAT
[10:37:14][D][climate:396]:   Custom Fan Mode: Automatic
[10:37:14][D][climate:402]:   Custom Preset: Normal
[10:37:14][D][climate:405]:   Swing Mode: BOTH
[10:37:14][D][climate:408]:   Current Temperature: 20.00°C
[10:37:14][D][climate:414]:   Target Temperature: 24.00°C
[10:37:14][V][component:199]: Component panasonic_ac.climate took a long time for an operation (0.09 s).
[10:37:14][V][component:200]: Components should block for at most 20-30ms.
[10:37:19][V][panasonic_ac.cz_tacg1:248]: Polling AC
[10:37:19][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[10:37:19][V][panasonic_ac:247]: RX: 70.20.44.30.80.A0.FD.00.00.40.00.00.4C.37.00.00.30.90.14.17.FF.18.17.FF.80.80.FF.80.AF.01.0C.80.41.57.B1 (35)
[... 135 lines cut ...]
[10:38:09][V][panasonic_ac.cz_tacg1:248]: Polling AC
[10:38:09][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[10:38:09][V][panasonic_ac:247]: RX: 70.20.44.30.80.A0.FD.00.00.40.00.00.4C.31.00.00.30.90.13.17.FF.17.17.FF.80.80.FF.80.43.02.0F.80.41.57.21 (35)
[10:38:09][D][climate:385]: 'Panasonic AC' - Sending state:
[10:38:09][D][climate:388]:   Mode: HEAT
[10:38:09][D][climate:396]:   Custom Fan Mode: Automatic
[... cut ...]
[10:38:14][D][climate:385]: 'Panasonic AC' - Sending state:
[10:38:14][D][climate:388]:   Mode: HEAT
[10:38:14][D][climate:396]:   Custom Fan Mode: Automatic
[10:38:14][D][climate:402]:   Custom Preset: Normal
[10:38:14][D][climate:405]:   Swing Mode: BOTH
[10:38:14][D][climate:408]:   Current Temperature: 20.00°C
[10:38:14][D][climate:414]:   Target Temperature: 24.00°C
[10:38:14][V][component:199]: Component panasonic_ac.climate took a long time for an operation (0.07 s).
[10:38:14][V][component:200]: Components should block for at most 20-30ms.
```