DomiStyle/esphome-panasonic-ac #57: Disconnects and connects regularly

MEKadan, 2022-05-01:
Hi, I'm getting these connection based errors and cannot get any readings nor control the AC.

I have lates versions of HAOS and esphome (and latest versions of all the other components also, if that matters) installed on Raspberry Pi 4 4GB version. I have used until now the CZ-TACG1 to connect to the internet, but as we all now, it cannot be held very reliable..

ESP is regularly disconnecting/ connecting itself. Log from esphome compoinent says:

`INFO Reading configuration /config/esphome/ac-mek-garage.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing ac-mek-garage (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <WiFi> 1.0
|-- <FS> 1.0
|-- <Update> 1.0
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <AsyncTCP-esphome> 1.2.2
|-- <DNSServer> 1.1.0
|-- <ESPmDNS> 1.0
Compiling /data/ac-mek-garage/.pioenvs/ac-mek-garage/src/main.cpp.o
Linking /data/ac-mek-garage/.pioenvs/ac-mek-garage/firmware.elf
RAM:   [=         ]  12.7% (used 41764 bytes from 327680 bytes)
Flash: [=====     ]  51.2% (used 939806 bytes from 1835008 bytes)
Building /data/ac-mek-garage/.pioenvs/ac-mek-garage/firmware.bin
esp32_create_combined_bin(["/data/ac-mek-garage/.pioenvs/ac-mek-garage/firmware.bin"], ["/data/ac-mek-garage/.pioenvs/ac-mek-garage/firmware.elf"])
Wrote 0xf5790 bytes to file /data/ac-mek-garage/.pioenvs/ac-mek-garage/firmware-factory.bin, ready to flash to offset 0x0
========================= [SUCCESS] Took 37.22 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of ac-mek-garage.local
INFO  -> 192.168.1.11
INFO Uploading /data/ac-mek-garage/.pioenvs/ac-mek-garage/firmware.bin (939920 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from ac-mek-garage.local using esphome API
WARNING Can't connect to ESPHome API for ac-mek-garage.local: Error resolving IP address: [Errno -2] Name or service not known
INFO Trying to reconnect to ac-mek-garage.local in the background
INFO Successfully connected to ac-mek-garage.local
[18:13:55][I][app:102]: ESPHome version 2022.4.0 compiled on May  1 2022, 18:12:57
[18:13:55][C][wifi:491]: WiFi:
[18:13:55][C][wifi:353]:   Local MAC: 7C:9E:BD:62:5C:F4
[18:13:55][C][wifi:354]:   SSID: [redacted]
[18:13:55][C][wifi:355]:   IP Address: 192.168.1.11
[18:13:55][C][wifi:357]:   BSSID: [redacted]
[18:13:55][C][wifi:358]:   Hostname: 'ac-mek-garage'
[18:13:55][C][wifi:360]:   Signal strength: -49 dB ▂▄▆█
[18:13:55][C][wifi:364]:   Channel: 8
[18:13:55][C][wifi:365]:   Subnet: 255.255.255.0
[18:13:55][C][wifi:366]:   Gateway: 192.168.1.1
[18:13:55][C][wifi:367]:   DNS1: 192.168.1.1
[18:13:55][C][wifi:368]:   DNS2: 0.0.0.0
[18:13:55][C][logger:233]: Logger:
[18:13:55][C][logger:234]:   Level: DEBUG
[18:13:55][C][logger:235]:   Log Baud Rate: 115200
[18:13:56][C][logger:236]:   Hardware UART: UART0
[18:13:56][D][api:102]: Accepted ::FFFF:C0A8:105
[18:13:56][C][uart.arduino_esp32:107]: UART Bus:
[18:13:56][C][uart.arduino_esp32:108]:   TX Pin: GPIO17
[18:13:56][C][uart.arduino_esp32:109]:   RX Pin: GPIO16
[18:13:56][C][uart.arduino_esp32:111]:   RX Buffer Size: 256
[18:13:56][C][uart.arduino_esp32:113]:   Baud Rate: 9600 baud
[18:13:56][C][uart.arduino_esp32:114]:   Data Bits: 8
[18:13:56][C][uart.arduino_esp32:115]:   Parity: EVEN
[18:13:56][C][uart.arduino_esp32:116]:   Stop bits: 1
[18:13:56][D][api.connection:827]: Home Assistant 2022.4.7 (::FFFF:C0A8:105): Connected successfully
[... 120 lines cut ...]
    name: Panasonic AC ESP MEK
    #horizontal_swing_select:
    #  name: Panasonic AC Horizontal Swing Mode
    vertical_swing_select:
      name: Panasonic AC ESP MEK Vertical Swing Mode
    outside_temperature:
      name: Panasonic AC ESP MEK Outside Temperature

    # Enable as needed
    # eco_switch:
    #   name: Panasonic AC Eco Switch
    # econavi_switch:
    #   name: Panasonic AC Econavi Switch
    # nanoex_switch:
    #   name: Panasonic AC NanoeX Switch
    # mild_dry_switch:
    #   name: Panasonic AC Mild Dry Switch

    current_power_consumption:
      name: Panasonic AC ESP MEK Power Consumption

    # Useful when the ac does not report a current temperature (CZ-TACG1 only)
    # current_temperature_sensor: temperature_sensor_id`

What might be the problem?

DomiStyle, 2022-05-01:
Can you connect to the ESP directly via USB? It's hard to see what's wrong with all the connection issues.

MEKadan, 2022-05-01:
I can connect to the esp via usb, but after taking it of from the AC. Also, esp boards (btw, the one I'm using is ESP32 Devkit V1 VROOM32 30-pin version) as fas as I know, should not be powered from many sources simultaneously? Which oprobably means I would need to disconnect it from AC anyway.

BTW- I enabled the verbose logging and I'm getting

`[19:02:10][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[19:02:15][V][panasonic_ac.cz_tacg1:248]: Polling AC
[19:02:15][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[19:02:20][V][panasonic_ac.cz_tacg1:248]: Polling AC
[19:02:20][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[19:02:25][V][panasonic_ac.cz_tacg1:248]: Polling AC
[19:02:25][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[19:02:30][V][panasonic_ac.cz_tacg1:248]: Polling AC
[19:02:30][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[19:02:35][V][panasonic_ac.cz_tacg1:248]: Polling AC
[19:02:35][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[19:02:40][V][panasonic_ac.cz_tacg1:248]: Polling AC
[19:02:40][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[19:02:45][V][panasonic_ac.cz_tacg1:248]: Polling AC
[19:02:45][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[19:02:50][V][panasonic_ac.cz_tacg1:248]: Polling AC
[19:02:50][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[19:02:55][V][panasonic_ac.cz_tacg1:248]: Polling AC
[19:02:55][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[19:03:00][V][panasonic_ac.cz_tacg1:248]: Polling AC
[19:03:00][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[19:03:05][V][panasonic_ac.cz_tacg1:248]: Polling AC
[19:03:05][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[19:03:10][V][panasonic_ac.cz_tacg1:248]: Polling AC
[19:03:10][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[19:03:15][V][panasonic_ac.cz_tacg1:248]: Polling AC
[19:03:15][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[19:03:20][V][panasonic_ac.cz_tacg1:248]: Polling AC
[19:03:20][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)`

... seems that it is not communicating with the AC?  I think I try to swap RX and TxD next...

MEKadan, 2022-05-01:
No difference when swapping RX and TX. Looks like the AC does not get anything and vice versa.

I could tka it out and use USB, but I think there is not much point in it, because the situation would be different?

MEKadan, 2022-05-01:
Hmmm... The ESP pcb + logic converter (TXS0108E, system built like in https://components101.com/modules/txs0108e-bi-directional-logic-level-converter-module , there is an example of connecting rasp to 5V system in the end ) was built on a trial board or whatever you call them, full of holes and the backside almost full of copper plating anyway :) .

Wifi ap was only few meters away. There semed to be lot of disturbance. I turned to trial pcb so that the copper side was facing ap, and the connection vastly improved. However, I cannot get a connection to AC. ESP seems to be polling AC (see above), which never responds. 

I have board on the table now, what you wish me to do next?

MEKadan, 2022-05-01:
OK. The connection probelem was disturbance-based and is now solved by electro-mecahical structure change, together with the fact, that bi-directional TXS0108E unit seems to be very delicate in these kind of applications, when it needs to devide, which side of the converter is load/ driver side at changing situtations.

Simplifying the structure as presented in this post:

https://github.com/DomiStyle/esphome-panasonic-ac/issues/34#issuecomment-975135342

.....solved the situation. One needs to be somewhat carefull, because this involves using 5V signals from AC to ESP's 3,3V inputs (and also feeds AC's 5V inputs with ESP's 3,3V, which seems to be enough).

Time will tell, how long ESP will last, according to this https://www.ridiculously-simple.com/2021/05/19/are-the-esp32-and-esp8266-5v-tolerant-yes-they-officially-are/ there are reasons to believe it will stand it.

However, solution is very simple and elegantish. ;)

closed 2023-01-31 by DomiStyle