DomiStyle/esphome-panasonic-ac #58: Compilig with ESPHOME 2022.05.0 fails

hpirila, 2022-05-19:
Compiling fails with latest ESPHOME version.

Three compiling logs below

- 2022.04.0, success
- 2022.05.0, failed
- 2022.05.0 with clean first, failed

```
hp@hpix:~$ pip install esphome==2022.04.0
Defaulting to user installation because normal site-packages is not writeable
Collecting esphome==2022.04.0
  Using cached esphome-2022.4.0-py2.py3-none-any.whl (2.2 MB)
Requirement already satisfied: esptool==3.3 in ./.local/lib/python3.10/site-packages (from esphome==2022.04.0) (3.3)
Requirement already satisfied: pyserial==3.5 in /usr/lib/python3/dist-packages (from esphome==2022.04.0) (3.5)
Collecting click==8.1.2
  Using cached click-8.1.2-py3-none-any.whl (96 kB)
Requirement already satisfied: tzlocal==4.2 in ./.local/lib/python3.10/site-packages (from esphome==2022.04.0) (4.2)
Requirement already satisfied: paho-mqtt==1.6.1 in ./.local/lib/python3.10/site-packages (from esphome==2022.04.0) (1.6.1)
Requirement already satisfied: aioesphomeapi==10.8.2 in ./.local/lib/python3.10/site-packages (from esphome==2022.04.0) (10.8.2)
Requirement already satisfied: zeroconf==0.38.4 in ./.local/lib/python3.10/site-packages (from esphome==2022.04.0) (0.38.4)
Requirement already satisfied: kconfiglib==13.7.1 in ./.local/lib/python3.10/site-packages (from esphome==2022.04.0) (13.7.1)
Requirement already satisfied: colorama==0.4.4 in /usr/lib/python3/dist-packages (from esphome==2022.04.0) (0.4.4)
Requirement already satisfied: voluptuous==0.13.1 in ./.local/lib/python3.10/site-packages (from esphome==2022.04.0) (0.13.1)
Requirement already satisfied: tzdata>=2021.1 in ./.local/lib/python3.10/site-packages (from esphome==2022.04.0) (2022.1)
Requirement already satisfied: tornado==6.1 in ./.local/lib/python3.10/site-packages (from esphome==2022.04.0) (6.1)
Requirement already satisfied: platformio==5.2.5 in ./.local/lib/python3.10/site-packages (from esphome==2022.04.0) (5.2.5)
Collecting esphome-dashboard==20220309.0
  Using cached esphome_dashboard-20220309.0-py3-none-any.whl (593 kB)
Requirement already satisfied: PyYAML==6.0 in ./.local/lib/python3.10/site-packages (from esphome==2022.04.0) (6.0)
Requirement already satisfied: protobuf<4.0,>=3.12.2 in ./.local/lib/python3.10/site-packages (from aioesphomeapi==10.8.2->esphome==2022.04.0) (3.20.1)
Requirement already satisfied: noiseprotocol<1.0,>=0.3.1 in ./.local/lib/python3.10/site-packages (from aioesphomeapi==10.8.2->esphome==2022.04.0) (0.3.1)
Requirement already satisfied: cryptography>=2.1.4 in /usr/lib/python3/dist-packages (from esptool==3.3->esphome==2022.04.0) (3.4.8)
Requirement already satisfied: bitstring>=3.1.6 in ./.local/lib/python3.10/site-packages (from esptool==3.3->esphome==2022.04.0) (3.1.9)
Requirement already satisfied: reedsolo<=1.5.4,>=1.5.3 in ./.local/lib/python3.10/site-packages (from esptool==3.3->esphome==2022.04.0) (1.5.4)
Requirement already satisfied: ecdsa>=0.16.0 in ./.local/lib/python3.10/site-packages (from esptool==3.3->esphome==2022.04.0) (0.17.0)
Requirement already satisfied: requests==2.* in /usr/lib/python3/dist-packages (from platformio==5.2.5->esphome==2022.04.0) (2.25.1)
Requirement already satisfied: semantic-version==2.9.* in ./.local/lib/python3.10/site-packages (from platformio==5.2.5->esphome==2022.04.0) (2.9.0)
Requirement already satisfied: pyelftools<1,>=0.27 in ./.local/lib/python3.10/site-packages (from platformio==5.2.5->esphome==2022.04.0) (0.28)
Requirement already satisfied: uvicorn==0.17.* in ./.local/lib/python3.10/site-packages (from platformio==5.2.5->esphome==2022.04.0) (0.17.6)
Requirement already satisfied: ajsonrpc==1.* in ./.local/lib/python3.10/site-packages (from platformio==5.2.5->esphome==2022.04.0) (1.2.0)
Requirement already satisfied: marshmallow<4,>=2 in ./.local/lib/python3.10/site-packages (from platformio==5.2.5->esphome==2022.04.0) (3.15.0)
Requirement already satisfied: tabulate==0.8.* in ./.local/lib/python3.10/site-packages (from platformio==5.2.5->esphome==2022.04.0) (0.8.9)
Requirement already satisfied: wsproto==1.0.* in ./.local/lib/python3.10/site-packages (from platformio==5.2.5->esphome==2022.04.0) (1.0.0)
Requirement already satisfied: starlette==0.18.* in ./.local/lib/python3.10/site-packages (from platformio==5.2.5->esphome==2022.04.0) (0.18.0)
Requirement already satisfied: bottle==0.12.* in ./.local/lib/python3.10/site-packages (from platformio==5.2.5->esphome==2022.04.0) (0.12.19)
Requirement already satisfied: aiofiles==0.8.* in ./.local/lib/python3.10/site-packages (from platformio==5.2.5->esphome==2022.04.0) (0.8.0)
Requirement already satisfied: pytz-deprecation-shim in ./.local/lib/python3.10/site-packages (from tzlocal==4.2->esphome==2022.04.0) (0.1.0.post0)
Requirement already satisfied: ifaddr>=0.1.7 in ./.local/lib/python3.10/site-packages (from zeroconf==0.38.4->esphome==2022.04.0) (0.1.7)
Requirement already satisfied: anyio<4,>=3.0.0 in ./.local/lib/python3.10/site-packages (from starlette==0.18.*->platformio==5.2.5->esphome==2022.04.0) (3.6.1)
Requirement already satisfied: h11>=0.8 in ./.local/lib/python3.10/site-packages (from uvicorn==0.17.*->platformio==5.2.5->esphome==2022.04.0) (0.13.0)
Requirement already satisfied: asgiref>=3.4.0 in ./.local/lib/python3.10/site-packages (from uvicorn==0.17.*->platformio==5.2.5->esphome==2022.04.0) (3.5.2)
Requirement already satisfied: six>=1.9.0 in /usr/lib/python3/dist-packages (from ecdsa>=0.16.0->esptool==3.3->esphome==2022.04.0) (1.16.0)
Requirement already satisfied: packaging in ./.local/lib/python3.10/site-packages (from marshmallow<4,>=2->platformio==5.2.5->esphome==2022.04.0) (21.3)
Requirement already satisfied: idna>=2.8 in /usr/lib/python3/dist-packages (from anyio<4,>=3.0.0->starlette==0.18.*->platformio==5.2.5->esphome==2022.04.0) (3.3)
Requirement already satisfied: sniffio>=1.1 in ./.local/lib/python3.10/site-packages (from anyio<4,>=3.0.0->starlette==0.18.*->platformio==5.2.5->esphome==2022.04.0) (1.2.0)
Requirement already satisfied: pyparsing!=3.0.5,>=2.0.2 in /usr/lib/python3/dist-packages (from packaging->marshmallow<4,>=2->platformio==5.2.5->esphome==2022.04.0) (2.4.7)
Installing collected packages: esphome-dashboard, click, esphome
  Attempting uninstall: esphome-dashboard
    Found existing installation: esphome-dashboard 20220508.0
    Uninstalling esphome-dashboard-20220508.0:
      Successfully uninstalled esphome-dashboard-20220508.0
  Attempting uninstall: click
    Found existing installation: click 8.1.3
    Uninstalling click-8.1.3:
      Successfully uninstalled click-8.1.3
  Attempting uninstall: esphome
    Found existing installation: esphome 2022.5.0
    Uninstalling esphome-2022.5.0:
      Successfully uninstalled esphome-2022.5.0
[... 220 lines cut ...]
Compiling .pioenvs/living-room-ac/src/esphome/components/panasonic_ac/esppac.cpp.o
src/esphome/components/panasonic_ac/esppac.cpp: In member function 'void esphome::panasonic_ac::PanasonicAC::set_vertical_swing_select(esphome::select::Select*)':
src/esphome/components/panasonic_ac/esppac.cpp:187:4: error: no matching function for call to 'esphome::select::Select::add_on_state_callback(esphome::panasonic_ac::PanasonicAC::set_vertical_swing_select(esphome::select::Select*)::<lambda(const string&)>)'
   });
    ^
In file included from src/esphome/components/panasonic_ac/esppac.h:4:0,
                 from src/esphome/components/panasonic_ac/esppac.cpp:1:
src/esphome/components/select/select.h:55:8: note: candidate: void esphome::select::Select::add_on_state_callback(std::function<void(std::__cxx11::basic_string<char>, unsigned int)>&&)
   void add_on_state_callback(std::function<void(std::string, size_t)> &&callback);
        ^
src/esphome/components/select/select.h:55:8: note:   no known conversion for argument 1 from 'esphome::panasonic_ac::PanasonicAC::set_vertical_swing_select(esphome::select::Select*)::<lambda(const string&)>' to 'std::function<void(std::__cxx11::basic_string<char>, unsigned int)>&&'
src/esphome/components/panasonic_ac/esppac.cpp: In member function 'void esphome::panasonic_ac::PanasonicAC::set_horizontal_swing_select(esphome::select::Select*)':
src/esphome/components/panasonic_ac/esppac.cpp:196:4: error: no matching function for call to 'esphome::select::Select::add_on_state_callback(esphome::panasonic_ac::PanasonicAC::set_horizontal_swing_select(esphome::select::Select*)::<lambda(const string&)>)'
   });
    ^
In file included from src/esphome/components/panasonic_ac/esppac.h:4:0,
[... cut ...]
src/esphome/components/select/select.h:55:8: note:   no known conversion for argument 1 from 'esphome::panasonic_ac::PanasonicAC::set_horizontal_swing_select(esphome::select::Select*)::<lambda(const string&)>' to 'std::function<void(std::__cxx11::basic_string<char>, unsigned int)>&&'
*** [.pioenvs/living-room-ac/src/esphome/components/panasonic_ac/esppac.cpp.o] Error 1
====================================================================== [FAILED] Took 8.71 seconds ======================================================================
hp@hpix:~$
```

jnicolson, 2022-05-20:
I'm getting this too after upgrading esphome.  The pull request which changed the signature of add_on_state_callback is https://github.com/esphome/esphome/pull/3423 but after a quick look i'm not 100% sure what the new index parameter is and how it's meant to be used.. hopefully someone can figure it out!

jnicolson, 2022-05-20:
Update - i just added the index into the callback signature but didn't use it, everything seems fine but still be nice to understand what the new index parameter is for.

https://github.com/jnicolson/esphome-panasonic-ac/commit/a323d4f8488eccfc80dd5d97e16d6b573c8cfca2

Sashik259, 2022-05-20:
@DomiStyle @hpirila I have the same issue after upgrading ESPHome to 2022.5.0

```
INFO Reading configuration /config/esphome/bedroom-ac.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing bedroom-ac (board: esp01_1m; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
Compiling /data/bedroom-ac/.pioenvs/bedroom-ac/src/esphome/components/panasonic_ac/esppac.cpp.o
Compiling /data/bedroom-ac/.pioenvs/bedroom-ac/src/esphome/components/select/select.cpp.o
Compiling /data/bedroom-ac/.pioenvs/bedroom-ac/src/esphome/components/select/select_call.cpp.o
Compiling /data/bedroom-ac/.pioenvs/bedroom-ac/src/esphome/components/select/select_traits.cpp.o
Compiling /data/bedroom-ac/.pioenvs/bedroom-ac/src/esphome/components/sensor/automation.cpp.o
src/esphome/components/panasonic_ac/esppac.cpp: In member function 'void esphome::panasonic_ac::PanasonicAC::set_vertical_swing_select(esphome::select::Select*)':
src/esphome/components/panasonic_ac/esppac.cpp:187:4: error: cannot convert 'esphome::panasonic_ac::PanasonicAC::set_vertical_swing_select(esphome::select::Select*)::<lambda(const string&)>' to 'std::function<void(std::__cxx11::basic_string<char>, unsigned int)>&&'
  187 |   });
      |    ^
In file included from src/esphome/components/panasonic_ac/esppac.h:4,
                 from src/esphome/components/panasonic_ac/esppac.cpp:1:
src/esphome/components/select/select.h:55:73: note:   initializing argument 1 of 'void esphome::select::Select::add_on_state_callback(std::function<void(std::__cxx11::basic_string<char>, unsigned int)>&&)'
   55 |   void add_on_state_callback(std::function<void(std::string, size_t)> &&callback);
      |                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
src/esphome/components/panasonic_ac/esppac.cpp: In member function 'void esphome::panasonic_ac::PanasonicAC::set_horizontal_swing_select(esphome::select::Select*)':
src/esphome/components/panasonic_ac/esppac.cpp:196:4: error: cannot convert 'esphome::panasonic_ac::PanasonicAC::set_horizontal_swing_select(esphome::select::Select*)::<lambda(const string&)>' to 'std::function<void(std::__cxx11::basic_string<char>, unsigned int)>&&'
  196 |   });
      |    ^
In file included from src/esphome/components/panasonic_ac/esppac.h:4,
                 from src/esphome/components/panasonic_ac/esppac.cpp:1:
src/esphome/components/select/select.h:55:73: note:   initializing argument 1 of 'void esphome::select::Select::add_on_state_callback(std::function<void(std::__cxx11::basic_string<char>, unsigned int)>&&)'
   55 |   void add_on_state_callback(std::function<void(std::string, size_t)> &&callback);
      |                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
Compiling /data/bedroom-ac/.pioenvs/bedroom-ac/src/esphome/components/sensor/filter.cpp.o
Compiling /data/bedroom-ac/.pioenvs/bedroom-ac/src/esphome/components/sensor/sensor.cpp.o
*** [/data/bedroom-ac/.pioenvs/bedroom-ac/src/esphome/components/panasonic_ac/esppac.cpp.o] Error 1
========================= [FAILED] Took 11.06 seconds =========================

```

Sashik259, 2022-05-20:
@jnicolson It doesn't work for me
[Sashik259@dfb0d69](https://github.com/Sashik259/esphome-panasonic-ac/commit/dfb0d69f59d46a6adf925cc19e4fc150fc3c03c9)

jnicolson, 2022-05-20:
@Sashik259 just double checking that you updated your esphome config and changed the external components section to your repository?

i.e. change
`source: github://DomiStyle/esphome-panasonic-ac`
to
`source: github://Sashik259/esphome-panasonic-ac`

My 3 are all running just fine now

Sashik259, 2022-05-21:
@jnicolson I changed it a month ago when I changed temperature step from 0.5 to 1.0. Yesterday I changed esppac.cpp file as you did, but with no result (the same log as you can see above).

DomiStyle, 2022-05-21:
@jnicolson I just tested your branch and it works fine for me. Would you mind sending in a pull request?

Sashik259, 2022-05-21:
@jnicolson you branch works for me too, but my one doesn't. WTF?

jnicolson, 2022-05-21:
> @jnicolson you branch works for me too, but my one doesn't. I deleted my branch, forked your one - and it doesn't work. WTF?

I can only guess somethings being cached by esphome - definitely strange!

closed 2022-05-21 by DomiStyle

DomiStyle, 2022-05-21:
Latest version should work with 2022.5.0 now.

If it doesn't make sure ESPHome pulls the latest repo changes, it doesn't do that just by cleaning the project.

Sashik259, 2022-05-21:
@jnicolson @DomiStyle
I am stupid idiot! Just added `refresh: 0s` string to update from github.
Now works fine! Thank you guys!
```
external_components:
  source: github://Sashik259/esphome-panasonic-ac
  components: [panasonic_ac]
  refresh: 0s
```

PR #59 (2022-05-21): Add index parameter into callback function for select state change
Required because of changes in ESPHome https://github.com/esphome/esphome/pull/3423

Fixes #58