DomiStyle/esphome-panasonic-ac #60: Cannot compile v2.3.0 on ESPHome v2022.3.1

dixonte, 2022-05-22:
I believe I'm on the latest HAOS, latest ESPHome, etc. And I cannot get it to build.

YAML:
```yaml
esphome:
  name: panasonic-ac-lounge
  platform: ESP32
  board: esp32dev

wifi:
  ssid: "StelTechCor"
  password: "****"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Panasonic-Ac Fallback Hotspot"
    password: "****"

captive_portal:

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:

# Enable MQTT
#mqtt:
#  broker: "homeassistant.lan"
#  username: "mqtt"
#  password: "b0rderlund5"

ota:
  password: "****"

# Requires ESPHome 1.15.0+ for the even parity option
uart:
  tx_pin: GPIO17
  rx_pin: GPIO16
  baud_rate: 9600
  parity: EVEN

external_components:
  source: github://DomiStyle/esphome-panasonic-ac
  components: [panasonic_ac]

climate:
  - platform: panasonic_ac
    id: panasonic_ac_id
    # For CZ-TACG1
    type: cnt

    # For DNSK-P11
    # type: wlan

    name: Panasonic AC
    horizontal_swing_select:
      name: Panasonic AC Horizontal Swing Mode
    vertical_swing_select:
    #  id: vertical_swing_select_id
      name: Panasonic AC Vertical Swing Mode
    outside_temperature:
      name: Panasonic AC Outside Temperature
```

Error log:
```
Processing panasonic-ac-lounge (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <WiFi> 1.0
|-- <FS> 1.0
|-- <Update> 1.0
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <AsyncTCP-esphome> 1.2.2
|-- <DNSServer> 1.1.0
|-- <ESPmDNS> 1.0
Compiling /data/panasonic-ac-lounge/.pioenvs/panasonic-ac-lounge/src/esphome/components/panasonic_ac/esppac.cpp.o
Compiling /data/panasonic-ac-lounge/.pioenvs/panasonic-ac-lounge/src/esphome/components/select/select.cpp.o
Compiling /data/panasonic-ac-lounge/.pioenvs/panasonic-ac-lounge/src/esphome/components/sensor/automation.cpp.o
Compiling /data/panasonic-ac-lounge/.pioenvs/panasonic-ac-lounge/src/esphome/components/sensor/filter.cpp.o
src/esphome/components/panasonic_ac/esppac.cpp: In member function 'void esphome::panasonic_ac::PanasonicAC::set_vertical_swing_select(esphome::select::Select*)':
src/esphome/components/panasonic_ac/esppac.cpp:187:4: error: no matching function for call to 'esphome::select::Select::add_on_state_callback(esphome::panasonic_ac::PanasonicAC::set_vertical_swing_select(esphome::select::Select*)::<lambda(const string&, size_t)>)'
   });
    ^
In file included from src/esphome/components/panasonic_ac/esppac.h:4:0,
                 from src/esphome/components/panasonic_ac/esppac.cpp:1:
src/esphome/components/select/select.h:60:8: note: candidate: void esphome::select::Select::add_on_state_callback(std::function<void(std::__cxx11::basic_string<char>)>&&)
   void add_on_state_callback(std::function<void(std::string)> &&callback);
        ^
src/esphome/components/select/select.h:60:8: note:   no known conversion for argument 1 from 'esphome::panasonic_ac::PanasonicAC::set_vertical_swing_select(esphome::select::Select*)::<lambda(const string&, size_t)>' to 'std::function<void(std::__cxx11::basic_string<char>)>&&'
src/esphome/components/panasonic_ac/esppac.cpp: In member function 'void esphome::panasonic_ac::PanasonicAC::set_horizontal_swing_select(esphome::select::Select*)':
src/esphome/components/panasonic_ac/esppac.cpp:196:4: error: no matching function for call to 'esphome::select::Select::add_on_state_callback(esphome::panasonic_ac::PanasonicAC::set_horizontal_swing_select(esphome::select::Select*)::<lambda(const string&, size_t)>)'
   });
    ^
In file included from src/esphome/components/panasonic_ac/esppac.h:4:0,
                 from src/esphome/components/panasonic_ac/esppac.cpp:1:
src/esphome/components/select/select.h:60:8: note: candidate: void esphome::select::Select::add_on_state_callback(std::function<void(std::__cxx11::basic_string<char>)>&&)
   void add_on_state_callback(std::function<void(std::string)> &&callback);
        ^
src/esphome/components/select/select.h:60:8: note:   no known conversion for argument 1 from 'esphome::panasonic_ac::PanasonicAC::set_horizontal_swing_select(esphome::select::Select*)::<lambda(const string&, size_t)>' to 'std::function<void(std::__cxx11::basic_string<char>)>&&'
Compiling /data/panasonic-ac-lounge/.pioenvs/panasonic-ac-lounge/src/esphome/components/sensor/sensor.cpp.o
Compiling /data/panasonic-ac-lounge/.pioenvs/panasonic-ac-lounge/src/esphome/components/socket/bsd_sockets_impl.cpp.o
*** [/data/panasonic-ac-lounge/.pioenvs/panasonic-ac-lounge/src/esphome/components/panasonic_ac/esppac.cpp.o] Error 1
========================= [FAILED] Took 10.01 seconds =========================
```

jnicolson, 2022-05-23:
Sorry this was my fault - when i fixed it for 2022.5.0 it broke it for older versions of ESPHome. 

Can you try with my fork?  This checks if the version is 2022.5.0 or greater and changes a function signature selectively

Update

`  source: github://DomiStyle/esphome-panasonic-ac`
to
`  source: github://jnicolson/esphome-panasonic-ac`

If it works i'll submit a pull request for DomiStyle to review

DomiStyle, 2022-05-23:
I'm personally fine with the component only working on the latest version of ESPHome. I did add the new minimum version to the readme though.

People running older versions can always pull one of the older releases: https://github.com/DomiStyle/esphome-panasonic-ac/releases/tag/2.2.1

dixonte, 2022-05-23:
Ah, and this is what I get for assuming Home Assitant just had the latest ESPHome. I've installed 2022.5 now and the adaptor code has compiled just fine. Sorry for the bother!

closed 2022-05-23 by dixonte

jnicolson, 2022-05-23:
> I'm personally fine with the component only working on the latest version of ESPHome. I did add the new minimum version to the readme though.
> 
> People running older versions can always pull one of the older releases: https://github.com/DomiStyle/esphome-panasonic-ac/releases/tag/2.2.1

That definitely makes things easier! And now I know a bit more about esphome so not wasted effort either way.  Thanks again for the work you've done!