DomiStyle/esphome-panasonic-ac #62: MQTT

ss4344, 2022-06-25:
A question.

I've successfully set up ESPHome with this Panasonic code, which is brilliant by the way. But I can't wrap my head around how to get it talking via MQTT, though. (I've got HomeAssistant integrated with CBus via MQTT, and also want to trigger A/C messages to this code from a CBus automation controller via MQTT.)

Got any pointers / tips / examples / doco to read?

redbullterror, 2022-06-25:
hello.

i use mqtt with node-red flow. and domoticz
for the topic's you need.
   
 `ac/climate/panasonic_sala/mode/command` 
`ac/climate/panasonic_sala/mode/state`
`ac/climate/panasonic_sala/target_temperature/command`
`ac/climate/panasonic_sala/fan_mode/command`

everything works only presets-mode doesn't work.
i don't now why.

ss4344, 2022-06-26:
Looks like I'm on the right path now.

Added this to my yaml and esphome began populating status topics via mqtt. (Or it was, before I blew up the board... LOLs.) Presumably the commands work, based on your hint @redbullterror, although I've not tried those yet.

`mqtt:
  broker: 192.168.xxx.xxx
  username: mqtt
  password: password
  
climate:
  - platform: panasonic_ac
    type: cnt
    name: Store AC

    mode_command_topic: storeac/climate/store_ac/mode/command
    swing_mode_command_topic: storeac/climate/store_ac/swing_mode/command
    fan_mode_command_topic: storeac/climate/store_ac/fan_mode/command
    target_temperature_command_topic: storeac/climate/store_ac/target_temperature/command
`

DomiStyle, 2022-06-27:
As mentioned, the AC unit and all sensors/switches will automatically be exposed over MQTT as soon as you enable MQTT, you shouldn't have to define any topics manually unless you need to somehow redirect them somewhere else.

redbullterror, 2022-06-30:
I can see all topics through mqtt explorer.
everything works with mqtt too.
only preset doesn't work,
only with home assistant it works to change preset.
looks like he doesn't make topic for preset

DomiStyle, 2022-07-01:
Looking at the ESPHome source it seems like the MQTT climate has no support for custom presets, so there's nothing I can do for now.

ss4344, 2022-07-03:
@DomiStyle it strikes me as inefficent that every climate status update received from the Panasonic results in an MQTT publish. This appears to occur every five seconds, regardless of previous state, and only for the 'climate' topics ('select' topics are not). Is it possible to prevent climate status publish where the value is already known to the ESPHome code?

My CBus SHAC will be responding to every status update, and running code consuming CPU but doing nothing as there is almost always no status change for the A/C (except if something is genuinely changed). Persistent messages means I get current status on topic subscribe, so frequent updates are irrelevant.

DomiStyle, 2022-07-26:
@ss4344 The integration currently does not save the previous state or compare to it when a new one arrives.

The polling interval is 5s for the CZ-TACG1 and 30s for the DNSK-P11. This mimics the behavior of the original wifi units.

> Is it possible to prevent climate status publish where the value is already known to the ESPHome code?

Sure it is, but is has to be implemented [here](https://github.com/DomiStyle/esphome-panasonic-ac/blob/e0719bf790f157ae0d5a6946d446bff726ad1327/components/panasonic_ac/esppac_cnt.cpp#L301) before parsing the packet.

I'm not planning to do that right now but feel free to open a separate issue for it so it can be tracked.

closed 2023-01-31 by DomiStyle

jmichiel, 2023-02-02:
> Looking at the ESPHome source it seems like the MQTT climate has no support for custom presets, so there's nothing I can do for now.

I'm working on a pull request for ESPHome for exactly that. Seems to work
You can use this as external component in the meantime:
```
external_components:
  - source: github://jmichiel/esphome@mqtt_climate_presets
    components: [ climate, mqtt ]
```
(if you don't also pull in the climate component, the preset won't get posted for this integration, as it only has custom presets)