DomiStyle/esphome-panasonic-ac #63: Help debugging interface for Panasonic CZ-CAPWFC1

Aidan-Chey, 2022-07-09:
Hello there Dominik,

I was pointed here by Michel as a possibility to help figure out if we can get the [Panasonic Climate Interface](https://www.espthings.io/index.php/2021/08/03/panasonic-climate-wifi-interface-for-esphome-and-home-assistant/) to work with the [CZ-CAPWFC1](https://www.logicool-ac.com/wp-content/uploads/2019/07/CZ-CAPWFC1-1.pdf) WLAN adaptor supplied by Panasonic for my Ducted Heat Pump.

Michel was nice enough to allow me to order 1 of the Panasonic Climate Interface with the hope it may work with the above adaptor. So I have one now and it's installed in the adaptor, replacing it's original Panasonic Climate Interface. It seems to work well enough, it powers on, has the web server and connects to home assistant through ESPHome. 

Only issue is the Panasonic Climate Interface doesn't seem to understand the packets from the adaptor and I was 
hoping you could cast your eye over the readings/logs and figure out if anything can be done to get it working.

Here is the config I have installed on the Panasonic Climate Interface at the moment. I've slightly adjusted the recommended to exclude the fin adjustment (ducted heat pumps don't have these) and add a restart button and verbose logging.
[bedroom-theatre-aircon-config.yaml.txt](https://github.com/DomiStyle/esphome-panasonic-ac/files/9075962/bedroom-theatre-aircon-config.yaml.txt)

Here is a log file from the Panasonic Climate Interface, taken from the ESPHome web interface in HA. 
[logs_bedrooms-theatre-aircon_logs.txt](https://github.com/DomiStyle/esphome-panasonic-ac/files/9075957/logs_bedrooms-theatre-aircon_logs.txt)

Here is a picture of the install in the adaptor (I'm aware the cable connection is backwards in image, I have corrected that)
[image]

Any help or suggestions you could give regarding this would be appreciated.

DomiStyle, 2022-07-09:
Looking at your log the AC responds with a valid packet but one that is not known to me: `5A.00.00.06.00.00.A0.5A.01.00.09.00.00.9C`.

After that the data is not parseable anymore (starting with `00`). It's hard to tell what's going on exactly without the expected output by a logic analyzer with the original unit in place. It might be that just the handshake is different in these units or they use a different protocol entirely.

However, the first packet is in the correct format for the usual DNSK-P11 so it might just be an error message you get back.

I never bothered actually decoding the handshake, I just replay it on the ESP as it is sent by the DNSK-P11 module, so for all I know there might be a version or device type encoded somewhere.

Is it possible for you to connect a logic analyzer to the original unit and sniff the handshake during power on?

Aidan-Chey, 2022-07-10:
Thanks a bunch for looking into this.

I don't have a logic analyzer, or heard of it before but the concept makes sense. I could probably buy one depending on how expensive it's gonna get. Looking online in NZ (where I live) stores there are options that range from $30 to $1500, with the cheaper variants being usb sticks and the more expensive being full fledged oscilloscopes.

I'm looking at this one here: https://www.phoenix-tech.co.nz/products/saleae-logic-analyser?variant=12429818036272
Says it uses a software package from https://www.saleae.com/downloads/ to read the results.

So my understanding is that when I hook this up to the adaptor instead of the _Panasonic Climate Interface_ the analyzer will record the handshake traffic sent by the adaptor. I can then use the logic recording on my PC with the Saleae software, then I'd post the results here for analysis. Does that sound about right?

DomiStyle, 2022-07-10:
> I'm looking at this one here: https://www.phoenix-tech.co.nz/products/saleae-logic-analyser?variant=12429818036272

Ideally it supports PulseView so I can open it easily. There's a list of supported devices here: https://sigrok.org/wiki/Supported_hardware

Any with 2 channels will do fine. The one you linked seems to be this one: https://sigrok.org/wiki/MCU123_Saleae_Logic_clone

> So my understanding is that when I hook this up to the adaptor instead of the Panasonic Climate Interface

You hook up the original Panasonic wifi adapter and in parallel you hook into the RX and TX pins with the logic analyzer and perform a cold start of the wifi module.

That way you will capture the entire handshake between the two boards.

DomiStyle, 2022-07-11:
> Looks like it needs to be plugged into a device for power and to do the reading

Yes, you will have to have a laptop connected with PulseView running to take the readings. Android might work too, never used the app before.

So laptop to logic analyzer and logic analyzer to TX, RX and GND on the DNSK-P11 module. That should do it. In case you have difficulties attaching the probes directly to the board a soldering iron might come in handy as well, there are pads you can connect to on the back.

Aidan-Chey, 2022-11-20:
Well, that was a hassel and a half.

For some reason suppliers weren't sending me the probes I ordered or they kept getting lost in shipping... I ended up soldering wires to the T2-T8 (T1 is 5v, looking at diagram) pads on the back of the board.

PulseView on Android is unusable, I can capture the sessions but I can't save the data anywhere cause it's seemed blackboxed by the OS (permissions issue maybe?). Anyway installed it on a windows tablet thing and captured the startup session. Of note the 1 minute cutoff for recording in pulseview stoped the recording as the card completed connection with the cloud (I can tell by the LED output of the wireless module) so not sure if you needed that bit.

In this attechement you will find the .sr & .pvs files
[immediateStart.zip](https://github.com/DomiStyle/esphome-panasonic-ac/files/10048953/immediateStart.zip)

Please let me know if this is what you needed. If you need to know which channel is which, im goning to need to spend more time doing another recording becaue I didn't take note of that. I know on the analyzer pin (channel?) 7 & 8 were marked as ground.

Here is an image I found from a [PDF for the DNSK-p11 board](https://fccid.io/NKR-P11/User-Manual/Users-Manual-revised0617-4320334.pdf) about it's layout.
[image]

Aidan-Chey, 2023-05-13:
No, no further developments so far. I'm not sure what's happening. As far as I am aware I've gotten the information @DomiStyle needed to attempt some sort of integration. Maybe they just weren't aware because I did finally get the readout around year end (holidays), and just needs a "poke" 😁

skellycode, 2023-07-21:
I'd be keen to see this progressed as I'll be getting a `CZ-CAPWFC1` installed in the next few weeks.  @DomiStyle, just checking that this is still on the radar?  Happy to help with testing when I get mine installed.

DomiStyle, 2023-12-12:
I didn't have time to take a detailed look into the captures.

There's an ongoing PR for newer devices that might work on this one as well: https://github.com/DomiStyle/esphome-panasonic-ac/pull/95

mchacher, 2023-12-12:
Thanks a lot. If you need help to test I can help

DomiStyle, 2023-12-12:
@mchacher You can try running the branch from the PR and see if it works.

mchacher, 2023-12-12:
@DomiStyle I am very familiar with ESP32 and hardware / software dev but not to esphome or homesssistant. Any simple way to do testing with C / C++ code and basic mqtt integration ?

DomiStyle, 2023-12-12:
Yes, setup the normal repo like described in the readme and then replace https://github.com/DomiStyle/esphome-panasonic-ac/blob/master/ac.yaml.example#L26 with `source: github://jonasf21/esphome-panasonic-ac`.

Agent6-6-6, 2024-05-31:
Can anyone provide an update on whether or not they got a solution working with an esp32 replacing the module in the CZ-CAPWFC1 wifi adaptor? Thanks.

wjcloudy, 2024-08-03:
Hi, also interested in current status here - I presume the wlan adapter mainboard the esphome module is connected to is just a PSU and level shifter for the RC lines (I don't even have that, they are super expensive)  Does anyone know the pinout for CN3 in the bottom left of the picture in the original post?

wjcloudy, 2024-08-27:
Have now also got an adapter connected to the CZ-CAPWFC1.  FYI I tried the jonasf21 fork @DomiStyle , but still getting similar errors:

`[11:11:07][V][panasonic_ac:247]: RX: 80.80.80.80.80 (5)
[11:11:07][W][panasonic_ac.cz_tacg1:271]: Dropping invalid packet (length)
[11:11:08][V][panasonic_ac.cz_tacg1:252]: Polling AC
[11:11:08][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[11:11:09][V][panasonic_ac:247]: RX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[11:11:09][W][panasonic_ac.cz_tacg1:362]: Received unknown fan speed
[11:11:09][W][panasonic_ac.cz_tacg1:476]: Received unknown mild dry value
[11:11:09][W][panasonic_ac:075]: Received out of range inside temperature: 110
[11:11:09][W][panasonic_ac:064]: Received out of range outside temperature: 101
[11:11:09][E][select:018]: 'House A/C Vertical Swing Mode': invalid state for publish_state(): unsupported
[11:11:09][D][climate:396]: 'House A/C' - Sending state:
[11:11:09][D][climate:399]:   Mode: OFF
[11:11:09][D][climate:407]:   Custom Fan Mode: Unknown
[11:11:09][D][climate:413]:   Custom Preset: Normal
[11:11:09][D][climate:416]:   Swing Mode: OFF
[11:11:09][D][climate:419]:   Current Temperature: nan°C
[11:11:09][D][climate:425]:   Target Temperature: 0.00°C
[11:11:11][VV][scheduler:226]: Running interval '' with interval=10000 last_execution=3328 (now=13332)
[11:11:12][V][panasonic_ac:247]: RX: 80.80.80.80.80 (5)`

My next step will be to capture a handshake using the original module.

dzungpv, 2024-09-21:
The CZ-CAPWFC1 using two wire Homebus standard, It can connect to R1R2 for the wall remote adapter.  
Document https://www.logicool-ac.com/wp-content/uploads/2019/07/CZ-CAPWFC1-1.pdf
You must have compatible hardware to read from R1R2 bus. If the WIFI CN03 does not work, so you can try connect direct to R1R2 bus.
[image]

I study about the Homebus too and have hardware running on ESP32 for Daikin, but the hardware quite complicate. Check project P1P2MQTT for more information. 
I can help decode the R1R2 data, it nearly the same with Daikin. But I am not own any Panasonic device.

chriswiggins, 2025-05-09:
I’m now looking to wire up our ducted unit into home assistant too. Is there any reason we can’t dump the firmware of the official adaptor and then reverse engineer it if it’s also an ESP?

B52-de, 2026-06-07:
Hi, https://github.com/beebop5/esphome-panasonic-ac has a solution for the new Wi-Fi adapter protocol. It works like a charm.

rndm2, 2026-07-04:
Hi,

I’m glad it works!

Dropped packets are normal as long as there are not dozens of them or a constant stream of errors.

Would you mind sending me the full ESP log (debug on) from boot? Ideally from the moment the device starts, through the cold init sequence, until the climate state, identity data, admin settings, and outdoor temperature are all populated.

That would help me check if anything unusual is happening on your unit.