DomiStyle/esphome-panasonic-ac #64: Failing to compile

Mark-Longley, 2022-07-11:
Hi,
I have 2 Panasonic units working in HA and have never had any issues previously.
I added a total daily energy meter and tried to install to the ESP, but it failed to compile.
I’ve tried reinstalling on my other ESP with the original unchanged configuration and I get the same errors.
I am running the latest version of Home assistant with all current updates.

Any guidance would be much appreciated
Thanks

INFO Reading configuration /config/esphome/master-bed-ac.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing master-bed-ac (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <WiFi> 1.0
|-- <FS> 1.0
|-- <Update> 1.0
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <AsyncTCP-esphome> 1.2.2
|-- <DNSServer> 1.1.0
|-- <ESPmDNS> 1.0
Compiling /data/master-bed-ac/.pioenvs/master-bed-ac/src/esphome/components/panasonic_ac/esppac.cpp.o
Compiling /data/master-bed-ac/.pioenvs/master-bed-ac/src/main.cpp.o
src/esphome/components/panasonic_ac/esppac.cpp: In member function 'void esphome::panasonic_ac::PanasonicAC::set_vertical_swing_select(esphome::select::Select*)':
src/esphome/components/panasonic_ac/esppac.cpp:187:4: error: no matching function for call to 'esphome::select::Select::add_on_state_callback(esphome::panasonic_ac::PanasonicAC::set_vertical_swing_select(esphome::select::Select*)::<lambda(const string&, size_t)>)'
   });
    ^
In file included from src/esphome/components/panasonic_ac/esppac.h:4:0,
                 from src/esphome/components/panasonic_ac/esppac.cpp:1:
src/esphome/components/select/select.h:60:8: note: candidate: void esphome::select::Select::add_on_state_callback(std::function<void(std::__cxx11::basic_string<char>)>&&)
   void add_on_state_callback(std::function<void(std::string)> &&callback);
        ^
src/esphome/components/select/select.h:60:8: note:   no known conversion for argument 1 from 'esphome::panasonic_ac::PanasonicAC::set_vertical_swing_select(esphome::select::Select*)::<lambda(const string&, size_t)>' to 'std::function<void(std::__cxx11::basic_string<char>)>&&'
src/esphome/components/panasonic_ac/esppac.cpp: In member function 'void esphome::panasonic_ac::PanasonicAC::set_horizontal_swing_select(esphome::select::Select*)':
src/esphome/components/panasonic_ac/esppac.cpp:196:4: error: no matching function for call to 'esphome::select::Select::add_on_state_callback(esphome::panasonic_ac::PanasonicAC::set_horizontal_swing_select(esphome::select::Select*)::<lambda(const string&, size_t)>)'
   });
    ^
In file included from src/esphome/components/panasonic_ac/esppac.h:4:0,
                 from src/esphome/components/panasonic_ac/esppac.cpp:1:
src/esphome/components/select/select.h:60:8: note: candidate: void esphome::select::Select::add_on_state_callback(std::function<void(std::__cxx11::basic_string<char>)>&&)
   void add_on_state_callback(std::function<void(std::string)> &&callback);
        ^
src/esphome/components/select/select.h:60:8: note:   no known conversion for argument 1 from 'esphome::panasonic_ac::PanasonicAC::set_horizontal_swing_select(esphome::select::Select*)::<lambda(const string&, size_t)>' to 'std::function<void(std::__cxx11::basic_string<char>)>&&'
*** [/data/master-bed-ac/.pioenvs/master-bed-ac/src/esphome/components/panasonic_ac/esppac.cpp.o] Error 1
========================== [FAILED] Took 2.67 seconds ==========================

DomiStyle, 2022-07-11:
Can you try updating your ESPHome version?

Mark-Longley, 2022-07-11:
I’m running the home assistant add on, it’s not showing any updates available.
I have a Version v2022.3.1

Mark-Longley, 2022-07-11:
Ah ok I’ll have to try and update it.
Not sure why it’s not showing any updates in HA.
Appreciate the help, Many Thanks!!

Mark-Longley, 2022-07-11:
In-install re-install (off and on again) up to date and all sorted. 
Awesome! Thanks Again

closed 2022-07-11 by Mark-Longley