DomiStyle/esphome-panasonic-ac #65: Ethernet

marcolega, 2022-07-25:
Is it possibile to use an ethernet esp board instead of wifi, like WT32-ETH01?
Thank

Marco

DomiStyle, 2022-07-26:
As long as you keep the serial pins (GPIO17 and GPIO16) free or change them and the board is supported by ESPHome, yes.

marcolega, 2022-07-28:
Unfortunately gpio16 is used for ethernet power switch, is it possible to use an other gpio? In the wt32 board uart use gpio17 and gpio5, is it possible use gpio5 istead of 16?
BR

DomiStyle, 2022-07-28:
You can use any of the following pairs:

```
UART0 - TX: GPIO1, RX: GPIO3

UART1 - TX: GPIO9, RX: GPIO10 (Only on ESP32)

UART2 - TX: GPIO16, RX: GPIO17 (Only on ESP32 but not ESP32S2, ESP32S3 or ESP32C3)
```

UART0 is used by the logger by default.

closed 2023-01-31 by DomiStyle