DomiStyle/esphome-panasonic-ac #66: DNSK-P11 Power Consumption

bigjme93, 2022-07-28:
Hi All,

Sorry if this has been covered before or i am missing something, i have just setup 2 of these on my CS-TZ25WKEW units at home and all seems to be working well

Previously i was able to get the current power consumption of the units from the comfort cloud integration i had but since switching to this i have lost this option

My config looks like this
`substitutions:
  devicename: "ac-unit-server-room"
  long_devicename: "Server Room AC"
  
esphome:
  name: "${devicename}"
  name_add_mac_suffix: false ## use different names/hostnames for each device!
  comment: ${long_devicename}
  platform: ESP32
  board: mhetesp32minikit
  project:
    name: esphome.pana-climate
    version: "220528-1"
external_components:
  source: github://DomiStyle/esphome-panasonic-ac
  components: [panasonic_ac]
  
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: "${long_devicename} Fallback Hotspot"
    password: !secret wifi_password
    
captive_portal:
    
web_server:
  port: 80
  
logger:
  level: DEBUG
api:
  
ota:
    
uart:
  tx_pin: GPIO17
  rx_pin: GPIO16
  id: ac_uart
  baud_rate: 9600
  parity: EVEN
  
status_led:
  pin: GPIO12
  
climate:
  - platform: panasonic_ac
    type: wlan
    name: "${long_devicename}"
    vertical_swing_select:
      name: "${long_devicename} Vertical Swing Mode"
    outside_temperature:
      name: "${long_devicename} Outside Temperature"`

I have tried adding this option but esphome does not like it and fails the compile. Am i missing anything or is this just not supported?
`    current_power_consumption:
      name: "${long_devicename} Power Consumption"`

DomiStyle, 2022-07-30:
Power consumption is not implemented for the DNSK-P11.

bigjme93, 2022-07-30:
Is there any way to find out this information to get it enabled? I'm very new to esphome so I'm not sure if I can program it via ota while connected to the ac to turn debug on or anything else

DomiStyle, 2022-08-01:
Sure, enable verbose logging and take a look as the fields that change as the power consumption should change. If you have the values that show in the original Panasonic cloud you can compare more easily.

There's a description of the known fields in the ODS files here: https://github.com/DomiStyle/esphome-panasonic-ac/tree/master/protocol

There's also a similar issue where someone implemented it for the CZ-TACG1: https://github.com/DomiStyle/esphome-panasonic-ac/issues/9

bigjme93, 2022-08-01:
I will have a look later and see what's returned, just to confirm, it should be 100% safe to flash the units with the new debug mode enabled without disconnecting them from the ac unit? Programming OTA shouldn't send anything down the uart for example? I don't have the comfort cloud now but i do have logs of old power usage so i will see what i can find

DomiStyle, 2022-08-01:
Yes, OTA update works perfectly fine.

bigjme93, 2022-08-01:
This is going to sound stupid but is it the tx or rx i need to monitor?
RX seems to have a single number incrementing each time after 5A at the start, TX seems to have 2 values changing on each request

DomiStyle, 2022-08-01:
RX is what's received from the AC so that's what you want to look at.

bigjme93, 2022-08-01:
I will have to keep looking at this i think as i'm not 100% sure its sent often enough to detect quickly. My cloud for exmaple only gave me a reading every 15 minutes which i assumed was a limit on their side

For example, this is my ac unit set to cool @ 30'c so basically doing nothing
`5A.4D.10.89.00.76.00.01.30.01.11.00.80.01.30.00.B0.01.42.02.31.01.3C.00.A0.01.33.00.A1.01.44.00.A5.01.43.00.A4.01.45.00.B2.01.41.02.35.01.41.02.33.01.41.02.34.01.41.02.32.01.42.00.BB.01.14.00.BE.01.1A.02.20.01.41.02.21.01.41.00.86.2E.2A.00.00.0B.01.01.48.30.30.30.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.5E`

This is it after being set to 16'C for about 4 minutes
`5A.66.10.89.00.76.00.01.30.01.11.00.80.01.30.00.B0.01.42.02.31.01.20.00.A0.01.33.00.A1.01.44.00.A5.01.43.00.A4.01.45.00.B2.01.41.02.35.01.41.02.33.01.41.02.34.01.41.02.32.01.42.00.BB.01.14.00.BE.01.1B.02.20.01.41.02.21.01.41.00.86.2E.2A.00.00.0B.01.01.48.30.30.30.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.60`

Power wise there should be several hundred watts difference between the 2 as a minimum but i see nothing obvious besides the settings i changed

bigjme93, 2022-08-01:
Ok, I'm fairly sure there must be another call Panasonic are making to these units to get power responses now. Having over an hours worth of logs now, and turning a unit off for 30 minutes and watching to see a different value change for the power, i can see nothing changing besides the inside and outside temperatures

Sadly I do not have a way of connecting my unit and the original Panasonic units up together to try and track what is happening between the 2 but its not looking promising so far

DomiStyle, 2022-08-02:
The DNSK-P11 protocol is quite more complex than the CZ-TACG1 protocol so it's indeed possible that the original wifi controller requests the power consumption when it needs it.

wailashi, 2022-09-09:
I started working on this, but got distracted. It looks like you can request the values you want from the AC in one packet. This library does not request the necessary values for calculating the power consumption.

We need to figure out the correct parameters to request. This could be done by capturing the communication between the DNSK-P11 module and the AC. I'll see if I can get any further with this before it gets too cold to tinker with the AC.

HA-TB303, 2022-09-22:
Nice hope you succeed. 

Sadly just replacing the DNSK-P11 version with a CZ-TACG1 is not a feasible solution as the units timer LED then keeps blinking, which points to a failed connection. Somehow the unit 'wants' the DNSK-P11. There is probably some jumper on the mainboard to select the installed wifi interface.

HA-TB303, 2022-09-29:
The issue I mentioned above was resolved based on a hint from Dominik: https://github.com/DomiStyle/esphome-panasonic-ac/issues/1#issuecomment-1255468588