DomiStyle/esphome-panasonic-ac #68: CS-Z (etherea) compatibility

CrashOverride-Git, 2022-08-02:
Hello, I am sorry I don't know if this is the right place to do this question, but I would like to know if my 2 split etherea (CS-Z25XKEW) are working with your sw.

The devices has the integrated wifi (I didn't find the expansion board) to work with their cloud-app. 
Opening the electrical wires cover I found an holding slot and a 5 pin socket. Can I use this connector?

[image]

HA-TB303, 2022-08-02:
Yes it will. I see the CN-CNT connector there.

For the integrated wifi check it there is a small box with DNSK-P11 on it as outlined here: https://www.espthings.io/index.php/2021/08/03/panasonic-climate-wifi-interface-for-esphome-and-home-assistant/#How_does_it_work

Michel

CrashOverride-Git, 2022-08-02:
Thank you, this is a very good news (their cloud based app sucks and I hate cloud services)

I have already saw that site but I don't know I can I buy that board.

I found these two info on the service manual, the wifi should be the one I point with the red arrow (Tomorrow I will check it)

[image]

HA-TB303, 2022-08-03:
Yes this: 
[image]
looks very much like the DNSK-P11 cable/connector, but you have to check yourself, I can't be certain.

CrashOverride-Git, 2022-08-04:
> Yes this: [image] looks very much like the DNSK-P11 cable/connector, but you have to check yourself, I can't be certain.

Hello, I just opened the case and the wifi module should be the DNSK (see the picture of what I found)
[image]

CrashOverride-Git, 2022-08-09:
Attached more pictures of the board inside of the unit

[image]

DomiStyle, 2022-08-10:
Yes, that's a DNSK-P11 and you can replace it with an ESP.

littlecake, 2022-12-26:
I connected the ESP to CN-CNT connector, and the original WiFi stays connected like it is. Everything works .
[image]

closed 2023-01-31 by DomiStyle

nicx, 2023-11-17:
@CrashOverride-Git @littlecake could you share your esp config? I have the same unit and I am wondering which "type" I should configure, because regarding the module I should use wlan, but regarding the connector I should use cnt?

littlecake, 2023-11-17:
> @CrashOverride-Git @littlecake could you share your esp config? I have the same unit and I am wondering which "type" I should configure, because regarding the module I should use wlan, but regarding the connector I should use cnt?

Hi, yes i used the CNT connector accesible from the front ..

esphome:
  name: panasonic-ac

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxxxxxxxxxxxxxx"

ota:
  password: "xxxxxxxxxxxxxxxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Panasonic-Ac Fallback Hotspot"
    password: "xxxxxxxxxxxxxxxx"

captive_portal:

external_components:
  source: github://DomiStyle/esphome-panasonic-ac
  components: [panasonic_ac]
    
web_server:
  port: 80

uart:
  tx_pin: GPIO17
  rx_pin: GPIO16
  id: ac_uart
  baud_rate: 9600
  parity: EVEN

switch:
  - platform: restart
    name: "Restart"
  
climate:
  - platform: panasonic_ac
    type: cnt   # For CZ-TACG1 use type: cnt / For DNSK-P11 use type: wlan
    name: "Kuchyne AC"
    vertical_swing_select:
      name: "Kuhyne AC Vertikání nastavení"
    horizontal_swing_select:
      name: "Kuhyne AC Horizontální nastavení"
    outside_temperature:
      name: "Kuhyne AC Teplota venku"
    nanoex_switch:
     name: "Kuhyne AC NanoeX"
    mild_dry_switch:
     name: "Kuhyne AC Mild Dry"
    current_power_consumption:
     name: "Kuhyne AC Spotřeba"