DomiStyle/esphome-panasonic-ac #69: Retrofit failed

ttsmarco, 2022-08-29:
I originally used the panasonic wifi module to work with the app
But the status update is very slowly
I recently discovered this project, let me looking forward to implementing localized access into HA

I disassemble my module
use ESP32 instead of DNSK-P11 module (only) according to the instructions
ESPHome log shows a failure message shortly after handshake
It seems AC has not provided any feedback
am I doing something wrong?

provide the information I found and my photo, there have IRQ TX and RX on my module
but i didn't use it
Is it relevant?

Any ideas might be helpful for me
Thanks

[image]
[image]
[image]

DomiStyle, 2022-08-29:
What's the name of your wifi unit that Panasonic uses? It looks similar (but not the same) to a `CZ-CAPWFC1` which is not supported (yet).

More info here: https://github.com/DomiStyle/esphome-panasonic-ac/issues/63

Please post verbose logs so we can see what's going on.

ttsmarco, 2022-08-29:
I am very rusty to ESPHOME
logs taken from ESPHOME log after upload ?
[image]

DomiStyle, 2022-08-31:
Yes, you have to enable verbose logging in your config:

```
logger:
  level: VERBOSE
```

Also, something seems wrong with your encoding since there'S so many ? in there.

ttsmarco, 2022-09-03:
> Yes, you have to enable verbose logging in your config:
> 
> ```
> logger:
>   level: VERBOSE
> ```
thx your info

Reconnected ESP32 and chang the log level to VERBOSE
[logs_ac_upload_1.txt](https://github.com/DomiStyle/esphome-panasonic-ac/files/9482119/logs_ac_upload_1.txt)

Seeing only TX, I realized maybe there is something wrong with the Logic converter board
after change it, I finally got the RX message
But still show failed 
[logs_ac_upload_2.txt](https://github.com/DomiStyle/esphome-panasonic-ac/files/9482133/logs_ac_upload_2.txt)

DomiStyle, 2022-09-03:
The received data seems to be corrupted.

Can you try connecting RX and TX directly to the ESP to check if the constant 00 disappear?

ttsmarco, 2022-09-03:
> The received data seems to be corrupted.
> 
> Can you try connecting RX and TX directly to the ESP to check if the constant 00 disappear?

Can't get RX message after connect RX and TX directly to the ESP
even I replaced a new ESP32 

[logs_ac_upload_3.txt](https://github.com/DomiStyle/esphome-panasonic-ac/files/9482213/logs_ac_upload_3.txt)

DomiStyle, 2022-09-04:
Can you check if you reversed RX and TX and they are connected to the correct pins?

ttsmarco, 2022-09-04:
> Can you check if you reversed RX and TX and they are connected to the correct pins?

yes
every time show mark failed ,i will try connect rx to pin17 ,tx to pin16  
even different ESP32
prevent me have fat finger

frankfurts88, 2024-11-07:
I know this is a very old thread but I stumbled upon it when pulling apart my Panasonic air conditioner and I have the same NKR P11 wifi module. Did you end up getting it to work?