DomiStyle/esphome-panasonic-ac #72: The temperature always change to 24 degree when change temperature to any value.

rotijohn87, 2022-11-01:
Hi,
I have some issues with the board.
When I select target temperature to 19 degrees inside my homeassistant, 
It suddenly changed the temperature back to 24 degrees.
I'm using  DSNK-P11 for my 2HP inverter aircond (CS-XPU18XKH-1)
Here is the attached photo for your reference
is there any solution for this problem ?
Here is attached log for your review.
[image]
[logs_panasonic-ac_logs (1).txt](https://github.com/DomiStyle/esphome-panasonic-ac/files/9910718/logs_panasonic-ac_logs.1.txt)

hpirila, 2022-12-02:
It is maybe related to the DRY mode seen, do you see same in COOL mode?

rotijohn87, 2022-12-03:
The same goes with COOL mode