DomiStyle/esphome-panasonic-ac #79: auto vertical swing error

marcolega, 2022-12-14:
Hi my panasonic Etherea split (model CS-Z25XKEW) have two different options for automatic vertical swing:
AUTO swing, but flap go fixed in upper position when target temperature is achieved:
[image]

Standard auto swing:
[image]

Standard auto swing mode option seem to be not implemented and when I put my split in that mode by remote I have the log:
[14:13:54][W][panasonic_ac.cz_tacg1:374]: Received unknown vertical swing mode
[14:13:54][E][select:018]: 'Studio AC Vertical Swing Mode': invalid state for publish_state(): Unknown

Is it possible to implement both vertical swing mode option?

DomiStyle, 2022-12-14:
I believe this was implemented very recently in https://github.com/DomiStyle/esphome-panasonic-ac/pull/77.

Can you reflash your ESP with the latest version?

marcolega, 2022-12-14:
Probabily not. Can you tell how can discover the installed version?

DomiStyle, 2022-12-14:
It's printed when the ESP is starting but there was no version bump yet for this feature.

ESPHome should automatically pull it when you rebuild the image. If unsure clear the cache first.

marcolega, 2022-12-14:
Yes I can reflash sorry.

closed 2023-01-31 by DomiStyle