DomiStyle/esphome-panasonic-ac #80: Unable to update

sygys, 2022-12-29:
It seems there are errors in the github repository.

I get the following error when trying to update:
[image]

im using the github repo to get the latest info:

```
uart:
  tx_pin: GPIO17
  rx_pin: GPIO16
  baud_rate: 9600
  parity: EVEN

external_components:
  source: github://DomiStyle/esphome-panasonic-ac@dev-cztacg1
  components: [panasonic_ac]

climate:
  - platform: panasonic_ac
    # For CZ-TACG1
    # type: cnt

    # For DNSK-P11
    type: wlan

    name: Panasonic AC
    horizontal_swing_select:
      name: Panasonic AC Horizontal Swing Mode
    vertical_swing_select:
      name: Panasonic AC Vertical Swing Mode
    outside_temperature:
      name: Panasonic AC Outside Temperature
```

DomiStyle, 2023-01-06:
There have been breaking changes with ESPHome 2022.12, I would recommend downgrading the ESPHome version for now.

DomiStyle, 2023-01-06:
Also, one more thing. You are still on the `dev-cztacg1`, that is no longer necessary.

DomiStyle, 2023-01-09:
> What breaking changes are you refering to?

I got some issues on my other ESPHome projects that use UART and I thought it might be the same case here. I didn't have time to investigate on those projects yet.

I assume you got things working by switching to the master branch?

sygys, 2023-01-13:
Yes I did indeed. Everything works like a charm. It's save to update :)

closed 2023-01-31 by DomiStyle