DomiStyle/esphome-panasonic-ac #82: Getting a packet with invalid checksum

nikarh, 2023-01-17:
Hi, I am trying to replace my native DNSK-P11 module with with ESP32 dev board using this project, and regardless of which board I use, which pins I use for TX/RT, and whether I use a logic converter or not, I get more or less the same output with verbose logging,

Basically I always get one of the two messages one has invalid header and another one has invalid checksum.
Any ideas on what I can do with that?

Thanks!

```
[20:41:08][D][panasonic_ac.dnskp11:721]: Resending previous packet
[20:41:08][V][panasonic_ac:245]: TX: 5A.01.00.09.00.00.9C (7)
[20:41:08][V][panasonic_ac:247]: RX: 5A.01.00.89.00.02.00.02.18 (9)
[20:41:08][D][panasonic_ac.dnskp11:555]: Answering handshake [2/16]
[20:41:08][V][panasonic_ac:245]: TX: 5A.02.00.0C.00.00.98 (7)
[20:41:08][V][panasonic_ac:247]: RX: 5A.02.00.8C.00.01.00.17 (8)
[20:41:08][D][panasonic_ac.dnskp11:559]: Answering handshake [3/16]
[20:41:08][V][panasonic_ac:245]: TX: 5A.03.00.10.00.01.20.72 (8)
[20:41:08][V][panasonic_ac:247]: RX: 5A.03.00.90.00.02.00.20.F1 (9)
[20:41:08][D][panasonic_ac.dnskp11:563]: Answering handshake [4/16]
[20:41:08][V][panasonic_ac:245]: TX: 5A.04.00.11.00.02.00.01.8E (9)
[20:41:08][V][panasonic_ac:247]: RX: 5A.04.00.91.00.36.00.00.01.00.01.0A.15.43.53.2D.5A.32.35.58.4B.45.57.2B.34.38.38.36.38.30.34.34.32.30.0B.02.01.2C.0C.14.43.53.2D.32.31.31.00.00.00.00.00.00.00.00.00.00.00.00.00.00.0F (61)
[20:41:08][D][panasonic_ac.dnskp11:567]: Answering handshake [5/16]
[20:41:08][V][panasonic_ac:245]: TX: 5A.05.00.12.00.04.01.10.11.12.57 (11)
[20:41:08][V][panasonic_ac:247]: RX: 5A.05.00.92.00.05.00.01.10.11.12.D6 (12)
[20:41:08][D][panasonic_ac.dnskp11:571]: Answering handshake [6/16]
[20:41:08][V][panasonic_ac:245]: TX: 5A.06.00.41.00.00.5F (7)
[20:41:08][V][panasonic_ac:247]: RX: 5A.06.00.C1.00.02.00.01.DC (9)
[20:41:08][D][panasonic_ac.dnskp11:575]: Answering handshake [7/16]
[20:41:08][V][panasonic_ac:245]: TX: 5A.07.01.4C.00.00.52 (7)
[20:41:08][V][panasonic_ac:247]: RX: 5A.07.01.CC.00.02.00.01.CF (9)
[20:41:08][D][panasonic_ac.dnskp11:579]: Answering handshake [8/16]
[20:41:08][V][panasonic_ac:245]: TX: 5A.08.10.00.00.00.8E (7)
[20:41:08][V][panasonic_ac:247]: RX: 5A.08.10.80.00.05.00.01.01.30.01.D6 (12)
[20:41:08][D][panasonic_ac.dnskp11:583]: Answering handshake [9/16]
[20:41:08][V][panasonic_ac:245]: TX: 5A.09.10.01.00.05.01.30.01.00.01.54 (12)
[20:41:09][V][panasonic_ac:247]: RX: 5A.09.10.81.00.D6.00.01.30.01.00.01.00.01.00.33.00.80.E2.01.00.85.42.04.00.86.62.2E.00.88.62.01.00.A0.E2.01.00.A1.E2.01.00.A4.E2.01.00.A5.E2.01.00.B0.E2.01.00.B2.E2.01.00.BB.42.01.00.BE.42.01.02.20.62.01.02.21.62.01.02.31.E2.01.02.32.62.01.02.33.E2.01.02.34.E2.01.02.35.E2.01.02.36.E2.01.02.42.82.01.02.7D.E2.01.02.7E.42.01.02.7F.42.02.02.80.42.02.02.81.42.02 (120)
[20:41:09][D][panasonic_ac.dnskp11:280]: Dropping invalid packet (checksum)
[20:41:09][V][panasonic_ac:247]: RX: 02.82.42.02.02.83.42.02.02.84.42.02.02.89.C2.0C.02.8A.C2.06.02.8B.C2.06.02.8C.C2.0E.02.91.42.02.02.92.C2.02.02.96.42.02.02.97.42.02.02.98.42.02.02.99.42.02.02.9A.42.02.02.9B.42.02.02.A0.C2.01.02.A1.C2.07.02.A2.C2.02.02.A3.C2.04.02.A4.C2.02.02.A5.C2.02.02.A8.42.02.02.A9.C2.03.02.AA.E2.02.02.FF.42.40.FC (101)
[20:41:09][W][panasonic_ac.dnskp11:249]: Dropping invalid packet (header)
[20:41:09][D][panasonic_ac.dnskp11:721]: Resending previous packet
[20:41:09][V][panasonic_ac:245]: TX: 5A.09.10.01.00.05.01.30.01.00.01.54 (12)
[20:41:09][V][panasonic_ac:247]: RX: 5A.09.10.81.00.D6.00.01.30.01.00.01.00.01.00.33.00.80.E2.01.00.85.42.04.00.86.62.2E.00.88.62.01.00.A0.E2.01.00.A1.E2.01.00.A4.E2.01.00.A5.E2.01.00.B0.E2.01.00.B2.E2.01.00.BB.42.01.00.BE.42.01.02.20.62.01.02.21.62.01.02.31.E2.01.02.32.62.01.02.33.E2.01.02.34.E2.01.02.35.E2.01.02.36.E2.01.02.42.82.01.02.7D.E2.01.02.7E.42.01.02.7F.42.02.02.80.42.02.02.81.42.02 (120)
[20:41:09][D][panasonic_ac.dnskp11:280]: Dropping invalid packet (checksum)
[20:41:09][V][panasonic_ac:247]: RX: 02.82.42.02.02.83.42.02.02.84.42.02.02.89.C2.0C.02.8A.C2.06.02.8B.C2.06.02.8C.C2.0E.02.91.42.02.02.92.C2.02.02.96.42.02.02.97.42.02.02.98.42.02.02.99.42.02.02.9A.42.02.02.9B.42.02.02.A0.C2.01.02.A1.C2.07.02.A2.C2.02.02.A3.C2.04.02.A4.C2.02.02.A5.C2.02.02.A8.42.02.02.A9.C2.03.02.AA.E2.02.02.FF.42.40.FC (101)
[20:41:09][W][panasonic_ac.dnskp11:249]: Dropping invalid packet (header)
[20:41:10][D][panasonic_ac.dnskp11:721]: Resending previous packet
[20:41:10][V][panasonic_ac:245]: TX: 5A.09.10.01.00.05.01.30.01.00.01.54 (12)
[20:41:10][V][panasonic_ac:247]: RX: 5A.09.10.81.00.D6.00.01.30.01.00.01.00.01.00.33.00.80.E2.01.00.85.42.04.00.86.62.2E.00.88.62.01.00.A0.E2.01.00.A1.E2.01.00.A4.E2.01.00.A5.E2.01.00.B0.E2.01.00.B2.E2.01.00.BB.42.01.00.BE.42.01.02.20.62.01.02.21.62.01.02.31.E2.01.02.32.62.01.02.33.E2.01.02.34.E2.01.02.35.E2.01.02.36.E2.01.02.42.82.01.02.7D.E2.01.02.7E.42.01.02.7F.42.02.02.80.42.02.02.81.42.02 (120)
[20:41:10][D][panasonic_ac.dnskp11:280]: Dropping invalid packet (checksum)
[20:41:10][V][panasonic_ac:247]: RX: 02.82.42.02.02.83.42.02.02.84.42.02.02.89.C2.0C.02.8A.C2.06.02.8B.C2.06.02.8C.C2.0E.02.91.42.02.02.92.C2.02.02.96.42.02.02.97.42.02.02.98.42.02.02.99.42.02.02.9A.42.02.02.9B.42.02.02.A0.C2.01.02.A1.C2.07.02.A2.C2.02.02.A3.C2.04.02.A4.C2.02.02.A5.C2.02.02.A8.42.02.02.A9.C2.03.02.AA.E2.02.02.FF.42.40.FC (101)
[20:41:10][W][panasonic_ac.dnskp11:249]: Dropping invalid packet (header)
[20:41:10][D][panasonic_ac.dnskp11:721]: Resending previous packet
[20:41:10][V][panasonic_ac:245]: TX: 5A.09.10.01.00.05.01.30.01.00.01.54 (12)
[20:41:11][V][panasonic_ac:247]: RX: 5A.09.10.81.00.D6.00.01.30.01.00.01.00.01.00.33.00.80.E2.01.00.85.42.04.00.86.62.2E.00.88.62.01.00.A0.E2.01.00.A1.E2.01.00.A4.E2.01.00.A5.E2.01.00.B0.E2.01.00.B2.E2.01.00.BB.42.01.00.BE.42.01.02.20.62.01.02.21.62.01.02.31.E2.01.02.32.62.01.02.33.E2.01.02.34.E2.01.02.35.E2.01.02.36.E2.01.02.42.82.01.02.7D.E2.01.02.7E.42.01.02.7F.42.02.02.80.42.02.02.81.42.02 (120)
[20:41:11][D][panasonic_ac.dnskp11:280]: Dropping invalid packet (checksum)
[20:41:11][V][panasonic_ac:247]: RX: 02.82.42.02.02.83.42.02.02.84.42.02.02.89.C2.0C.02.8A.C2.06.02.8B.C2.06.02.8C.C2.0E.02.91.42.02.02.92.C2.02.02.96.42.02.02.97.42.02.02.98.42.02.02.99.42.02.02.9A.42.02.02.9B.42.02.02.A0.C2.01.02.A1.C2.07.02.A2.C2.02.02.A3.C2.04.02.A4.C2.02.02.A5.C2.02.02.A8.42.02.02.A9.C2.03.02.AA.E2.02.02.FF.42.40.FC (101)
[20:41:11][W][panasonic_ac.dnskp11:249]: Dropping invalid packet (header)
[20:41:11][D][panasonic_ac.dnskp11:721]: Resending previous packet
[20:41:11][V][panasonic_ac:245]: TX: 5A.09.10.01.00.05.01.30.01.00.01.54 (12)
[20:41:11][V][panasonic_ac:247]: RX: 5A.09.10.81.00.D6.00.01.30.01.00.01.00.01.00.33.00.80.E2.01.00.85.42.04.00.86.62.2E.00.88.62.01.00.A0.E2.01.00.A1.E2.01.00.A4.E2.01.00.A5.E2.01.00.B0.E2.01.00.B2.E2.01.00.BB.42.01.00.BE.42.01.02.20.62.01.02.21.62.01.02.31.E2.01.02.32.62.01.02.33.E2.01.02.34.E2.01.02.35.E2.01.02.36.E2.01.02.42.82.01.02.7D.E2.01.02.7E.42.01.02.7F.42.02.02.80.42.02.02.81.42.02 (120)
[20:41:11][D][panasonic_ac.dnskp11:280]: Dropping invalid packet (checksum)
[20:41:11][V][panasonic_ac:247]: RX: 02.82.42.02.02.83.42.02.02.84.42.02.02.89.C2.0C.02.8A.C2.06.02.8B.C2.06.02.8C.C2.0E.02.91.42.02.02.92.C2.02.02.96.42.02.02.97.42.02.02.98.42.02.02.99.42.02.02.9A.42.02.02.9B.42.02.02.A0.C2.01.02.A1.C2.07.02.A2.C2.02.02.A3.C2.04.02.A4.C2.02.02.A5.C2.02.02.A8.42.02.02.A9.C2.03.02.AA.E2.02.02.FF.42.40.FC (101)
[20:41:11][W][panasonic_ac.dnskp11:249]: Dropping invalid packet (header)
```

nikarh, 2023-01-29:
Just wanted to share that I didn't figure out how to make it work with DNSK-P11, but since my HVAC also has CN-CNT port (which is much more easily accessible and this project supports more features via this connector), I found the [connectors](https://web.archive.org/web/20230129124912/https://www.aliexpress.us/item/3256804577594050.html?gatewayAdapt=glo2usa4itemAdapt&_randl_shipto=US) connected everything and it works flawlesly via CN-CNT!

Thanks for the amazing project!

DomiStyle, 2023-01-31:
Good to hear you got it working!

closed 2023-01-31 by DomiStyle

jonasf21, 2023-06-29:
@nikarh it seems you have a Ethera device, just as i do. I have the CS-Z35XKEW, so same device as you but 3,5kW.

This needs some changes to work. If you want you can checkout my fork:
https://github.com/jonasf21/esphome-panasonic-ac/tree/master