DomiStyle/esphome-panasonic-ac #83: Support for action

jmichiel, 2023-01-31:
I see in the code that 'action' is explicitly set to unsupported:
`traits.set_supports_action(false);`

Is there a specific reason for this, or didn't you get around to implementing it? Or is there is it not supported by both A/C protocols?

I'm currently using the Comfort Cloud integration and do get a valid action state, so it should be able to work (at least on DNSK-P11, as my A/C is using).

Or is there is it not supported by CZ-TACG1?

DomiStyle, 2023-01-31:
The AC does not report what it is actively doing from what I can tell, only what mode it is in.
So there is no way to know if the AC is currently idle or actively heating/cooling.

There might be a flag somewhere but I never bothered to dig deeper.

jmichiel, 2023-01-31:
OK, maybe if I ever find the time, I might have a stab at it, then.

Since [this integration](https://github.com/sockless-coding/panasonic_cc/) manages to get an accurate action state from the cloud, it should be possible. But perhaps not on both protocols.

Not essential, but still a nice-to-have.

jmichiel, 2023-02-08:
After some investigation, the CZ-TACG1 protocol doesn't seem to report current action indeed, so it probably only works with DNSK-P11. I'm still also running the Comfort Cloud over DNSK-P11 for the time being, so maybe if I decide to get rid of it, I could try to figure out how DNSK-P11 does it.

DomiStyle, 2023-02-08:
You can check the protocol description in the folder [here](https://github.com/DomiStyle/esphome-panasonic-ac/tree/master/protocol).
There's plenty of unknown fields that could hold the current action.

I might be able to take a look another look as well but only in summer since I don't use my AC for heating.

d96moe, 2025-01-21:
First, great project! I now have two units running, one over cnt interface and one over wlan. Did anyone find any time looking into this? As mentioned above, maybe not THE most important feature but definitely a nice to have.

d96moe, 2026-08-31:
To update on a very old thread. Found someone who found the cn-cnt solution and had a go at implementing it in https://github.com/DomiStyle/esphome-panasonic-ac/pull/194

Kudos to https://github.com/ssjoholm/panasonic-cn-cnt