DomiStyle/esphome-panasonic-ac #84: Custom preset seem to be missing (Normal, Quiet, Powerful)

profigriller, 2023-02-10:
It seems the custom presets seem to be missing (Normal, Quiet, Powerful).
I check source code and could not find them.
I tried preset, presets, custom_presets in yaml find but could not compile in EspHome.
Is the implementation missing?

profigriller, 2023-02-10:
hmmm .. they are missing in my EspHome setup. What do I have to include in he yaml file?
I am new to EspHome and I assume the problem is in front of the computer :-)
  
  [preset] is an invalid option for [climate.panasonic_ac]. Please check the indentation.
  preset:
    name: Custom Preset
  
climate:
  - platform: panasonic_ac
    type: cnt   # For CZ-TACG1 use type: cnt / For DNSK-P11 use type: wlan
    name: "${long_devicename}"
    vertical_swing_select:
      name: "${long_devicename} Vertical Swing Mode"
    horizontal_swing_select:
      name: "${long_devicename} Horizontal Swing Mode"
    outside_temperature:
      name: "${long_devicename} Outside Temperature"
    nanoex_switch:
      name: Panasonic AC NanoeX Switch
    current_power_consumption:
      name: Panasonic AC Power Consumption
    preset:
      name:  Custom Preset
[image]

DomiStyle, 2023-02-10:
They are not exposed in the web interface from what I know, you will have to use Home Assistant or MQTT to access them.
You don't have to add anything to the yaml, it's exposed automatically.

profigriller, 2023-02-10:
I am using MQTT fx. I can't find any topics being exposed which relate to preset. When changing to Quiet mode I would have expected that a new topic would be published.

homeassistant/climate/czegraum1/espthings_panasonic_climate_interface/config reports for MQTT topics as follows.

{"curr_temp_t":"czegraum1/climate/espthings_panasonic_climate_interface/current_temperature/state","mode_cmd_t":"czegraum1/climate/espthings_panasonic_climate_interface/mode/command","mode_stat_t":"czegraum1/climate/espthings_panasonic_climate_interface/mode/state","modes":["off","cool","heat","fan_only","dry","heat_cool"],"temp_cmd_t":"czegraum1/climate/espthings_panasonic_climate_interface/target_temperature/command","temp_stat_t":"czegraum1/climate/espthings_panasonic_climate_interface/target_temperature/state","min_temp":16,"max_temp":30,"temp_step":0.5,"temp_unit":"C","fan_mode_cmd_t":"czegraum1/climate/espthings_panasonic_climate_interface/fan_mode/command","fan_mode_stat_t":"czegraum1/climate/espthings_panasonic_climate_interface/fan_mode/state","fan_modes":["1","2","3","4","5","Automatic"],"swing_mode_cmd_t":"czegraum1/climate/espthings_panasonic_climate_interface/swing_mode/command","swing_mode_stat_t":"czegraum1/climate/espthings_panasonic_climate_interface/swing_mode/state","swing_modes":["off","both","vertical","horizontal"],"name":"ESPthings Panasonic Climate Interface","avty_t":"czegraum1/status","uniq_id":"ESPclimateespthings_panasonic_climate_interface","dev":{"ids":"XXXXXXX","name":"czegraum1","sw":"esphome v2022.12.8 Feb  9 2023, 17:34:52","mdl":"mhetesp32minikit","mf":"espressif"}}

[image]

profigriller, 2023-02-10:
ah, cool. Thank you. I search issues before but must have overlook this. I will close this issue. It is a duplicate.

jmichiel, 2023-02-15:
Should be fixed with the latest ESPHome version (2023.2.0)

jmichiel, 2023-02-16:
And it seems I broke MQTT discovery with that PR :fearful:

profigriller, 2023-02-18:
For me this is working OK now

profigriller, 2023-02-18:
and closed

closed 2023-02-18 by profigriller