DomiStyle/esphome-panasonic-ac #85: OTA installation does not work anymore

zioCristia, 2023-03-03:
Hi, I have been using this excellent project for at least a year without any problem. Recently I wanted to add the software reboot switch to the esp, just in case, and I am experiencing a problem when installing the OTA software.
It is a compilation problem and the fact is that now, even if I remove the restart switch lines, I still have the same problem.

Here are the logs:

```
INFO Reading configuration /config/esphome/condizionatore-camera.yaml...
INFO Detected timezone 'Europe/Rome'
INFO Generating C++ source...
INFO Compiling app...
Processing condizionatore-camera (board: esp01_1m; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.1
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
Compiling /data/condizionatore-camera/.pioenvs/condizionatore-camera/src/esphome/components/panasonic_ac/esppac.cpp.o
Compiling /data/condizionatore-camera/.pioenvs/condizionatore-camera/src/esphome/components/select/select.cpp.o
src/esphome/components/panasonic_ac/esppac.cpp: In member function 'void esphome::panasonic_ac::PanasonicAC::set_vertical_swing_select(esphome::select::Select*)':
src/esphome/components/panasonic_ac/esppac.cpp:187:4: error: cannot convert 'esphome::panasonic_ac::PanasonicAC::set_vertical_swing_select(esphome::select::Select*)::<lambda(const string&, size_t)>' to 'std::function<void(std::__cxx11::basic_string<char>)>&&'
  187 |   });
      |    ^
In file included from src/esphome/components/panasonic_ac/esppac.h:4,
                 from src/esphome/components/panasonic_ac/esppac.cpp:1:
src/esphome/components/select/select.h:60:65: note:   initializing argument 1 of 'void esphome::select::Select::add_on_state_callback(std::function<void(std::__cxx11::basic_string<char>)>&&)'
   60 |   void add_on_state_callback(std::function<void(std::string)> &&callback);
      |                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
src/esphome/components/panasonic_ac/esppac.cpp: In member function 'void esphome::panasonic_ac::PanasonicAC::set_horizontal_swing_select(esphome::select::Select*)':
src/esphome/components/panasonic_ac/esppac.cpp:196:4: error: cannot convert 'esphome::panasonic_ac::PanasonicAC::set_horizontal_swing_select(esphome::select::Select*)::<lambda(const string&, size_t)>' to 'std::function<void(std::__cxx11::basic_string<char>)>&&'
  196 |   });
      |    ^
In file included from src/esphome/components/panasonic_ac/esppac.h:4,
                 from src/esphome/components/panasonic_ac/esppac.cpp:1:
src/esphome/components/select/select.h:60:65: note:   initializing argument 1 of 'void esphome::select::Select::add_on_state_callback(std::function<void(std::__cxx11::basic_string<char>)>&&)'
   60 |   void add_on_state_callback(std::function<void(std::string)> &&callback);
      |                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
Compiling /data/condizionatore-camera/.pioenvs/condizionatore-camera/src/esphome/components/sensor/automation.cpp.o
*** [/data/condizionatore-camera/.pioenvs/condizionatore-camera/src/esphome/components/panasonic_ac/esppac.cpp.o] Error 1
========================= [FAILED] Took 14.69 seconds =========================
```

Am I doing something wrong or it's an EspHome problem?

The device still works properly and I can command the AC without any problems.

DomiStyle, 2023-03-03:
Can you post your config file?

zioCristia, 2023-03-03:
Here it is:

```
esphome:
  name: condizionatore-camera

esp8266:
  board: esp01_1m

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:

ota:
  password: !secret oat

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: !secret ap_ssid
    password: !secret ap_pw

captive_portal:

# Requires ESPHome 1.15.0+ for the even parity option
uart:
  tx_pin: GPIO2
  rx_pin: GPIO13
  baud_rate: 9600
  parity: EVEN

external_components:
  source: github://DomiStyle/esphome-panasonic-ac
  components: [panasonic_ac]

climate:
  - platform: panasonic_ac
    # For CZ-TACG1
    type: cnt

    name: Panasonic camera
    #horizontal_swing_select:
      #name: Panasonic AC Horizontal Swing Mode
    vertical_swing_select:
      name: Panasonic AC Vertical Swing Mode
    outside_temperature:
      name: Panasonic AC Outside Temperature

    # Enable as needed
    eco_switch:
      name: Panasonic AC Eco Switch
    # econavi_switch:
    #   name: Panasonic AC Econavi Switch
    # nanoex_switch:
    #   name: Panasonic AC NanoeX Switch
    # mild_dry_switch:
    #   name: Panasonic AC Mild Dry Switch
    current_power_consumption:
      name: ac-broom-power-consumption
      id: ac_broom_power_consumption

    # Useful when the ac does not report a current temperature (CZ-TACG1 only)
    # current_temperature_sensor: temperature_sensor_id

sensor:
    # AC kWh
  - platform: total_daily_energy
    name: "Energia totale condizionatore camera"
    power_id: ac_broom_power_consumption
    filters:
      - multiply: 0.001
    unit_of_measurement: kWh
    
time:
- platform: sntp
  id: my_time
  
switch:
- platform: restart
  name: "Ac Bedroom Restart"
```

Mrsash2020, 2024-10-06:
Change this
```
ota:
  password: !secret oat
```
to
```
ota:
  - platform: esphome
  password: !secret oat
```