DomiStyle/esphome-panasonic-ac #86: Power consumption when 2 indoor units on one outdoor unit

HA-TB303, 2023-03-07:
Hi, I have 2 indoor units CS-TZ25WKE attached to 1 outdoor unit.

I use the ESP32 CN-CNT variant in both indoor units.
I have 2 energy sensors reporting the same when idle (both 8 watt) and they also seem to be reporting the same when in use.

Is this energy consumption measurement taken from the outdoor unit for the whole system combined?

DomiStyle, 2023-03-22:
The power consumption was not implemented by me but from what I remember it's just a rough approximation, which is probably why you get the same values.

I assume if one is idle and one is on you get different values?

HA-TB303, 2023-03-22:
Nope, the values are practically always identical. Sometimes one is a little faster reporting then the other, but generally they are the same.

DomiStyle, 2023-03-22:
Strange, that makes it even more useless than it was with a single unit.

War3333, 2023-05-29:
I have the official intesys knx adapter for my knx system, i can confirm that the consumption is indeed calculated on the outdoor unit and will be reported to be the same on both units. The consumption of the inside split are few watts anyway (tested with a real amperometric sensor) so i don't think it really matters.

nicx, 2023-12-04:
I have 3 indoor units with the same behaviour. Is it confirmed that this is by design? In the cloud app I cann see slightly different power values between the 3 indoor units. Is Panasonic itself also using the outdoor unit for estimating power consumption?

qwert4712, 2024-06-19:
"For Multi-Split AC Systems, the energy consumption value displayed is for whole system."

see 
https://www.panasonic.com/global/hvac/products/comfort-cloud/how-to-use-comfort-cloud/check-air-conditioner-energy-consumption.html