DomiStyle/esphome-panasonic-ac #87: IDLE states on ducted unit

JeromeChignola, 2023-03-20:
Hello.
First of all, my English is bad (sorry, I'm French).
I got a document from panasonic that could probably interest you.
Then, I have been using your dev for a few months. The problem that I encounter is the following: on my ductable, I will need to recover the "idle" state because I have a regulation by zone. 
Do you think it's possible?
[Questions frÃ©quentes.ppt  -  Mode de compatibilitÃ©.pdf](https://github.com/DomiStyle/esphome-panasonic-ac/files/11019577/Questions.frA.quentes.ppt.-.Mode.de.compatibilitA.pdf)

DomiStyle, 2023-03-22:
Can you define what you mean with "idle state"?

Can you give an example of what you're trying to accomplish?

JeromeChignola, 2023-03-22:
By "idle state" I mean when the duct is on but not heating. When it has reached the set temperature for example. 
I am trying to optimize my regulation by zone but the duct starts far too often.
Ideally, I am looking to shunt the value of the return temperature sensor