DomiStyle/esphome-panasonic-ac #89: Mild Dry Switch keeps getting off

michas79de, 2023-04-13:
Hello!
great job @DomiStyle so far! I really appreciate being freed from the cloud! 👍

While my remote control has a MILD DRY button, the same button in the Home Assistant cannot be switched on permanently, but is immediately switched off again every time I try.

When using the remote control, the log shows that the 14th RX byte switches 2D (off) <-> 2B (on), I think:

> [22:32:31][V][panasonic_ac.cz_tacg1:248]: Polling AC
[22:32:31][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[22:32:31][V][panasonic_ac:247]: RX: 70.20.34.2B.80.A0.5A.00.00.00.00.00.30.2B.00.00.20.97.80.80.FF.12.06.FF.80.80.FF.80.18.00.01.00.00.00.D7 (35)
[22:32:31][D][climate:392]: 'Panasonic AC' - Sending state:
[22:32:31][D][climate:395]:   Mode: COOL
[22:32:31][D][climate:403]:   Custom Fan Mode: Automatic
[22:32:31][D][climate:409]:   Custom Preset: Normal
[22:32:31][D][climate:412]:   Swing Mode: OFF
[22:32:31][D][climate:415]:   Current Temperature: 18.00°C
[22:32:31][D][climate:421]:   Target Temperature: 21.50°C
[22:32:31][V][component:204]: Component panasonic_ac.climate took a long time for an operation (0.05 s).
[22:32:31][V][component:205]: Components should block for at most 20-30ms.
[22:32:36][V][panasonic_ac.cz_tacg1:248]: Polling AC
[22:32:36][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[22:32:36][V][panasonic_ac:247]: RX: 70.20.34.2B.80.A0.5A.00.00.00.00.00.30.2D.00.00.20.97.80.80.FF.12.06.FF.80.80.FF.80.18.00.01.00.00.00.D5 (35)
[22:32:36][D][climate:392]: 'Panasonic AC' - Sending state:
[22:32:36][D][climate:395]:   Mode: COOL
[22:32:36][D][climate:403]:   Custom Fan Mode: Automatic
[22:32:36][D][climate:409]:   Custom Preset: Normal
[22:32:36][D][climate:412]:   Swing Mode: OFF
[22:32:36][D][climate:415]:   Current Temperature: 18.00°C
[22:32:36][D][climate:421]:   Target Temperature: 21.50°C
[22:32:36][V][component:204]: Component panasonic_ac.climate took a long time for an operation (0.05 s).
[22:32:36][V][component:205]: Components should block for at most 20-30ms.
[22:32:41][V][panasonic_ac.cz_tacg1:248]: Polling AC
[22:32:41][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[22:32:41][V][panasonic_ac:247]: RX: 70.20.34.2B.80.A0.5A.00.00.00.00.00.30.2B.00.00.20.97.80.80.FF.12.06.FF.80.80.FF.80.18.00.01.00.00.00.D7 (35)
[22:32:41][D][climate:392]: 'Panasonic AC' - Sending state:
[22:32:41][D][climate:395]:   Mode: COOL
[22:32:41][D][climate:403]:   Custom Fan Mode: Automatic
[22:32:41][D][climate:409]:   Custom Preset: Normal
[22:32:41][D][climate:412]:   Swing Mode: OFF
[22:32:41][D][climate:415]:   Current Temperature: 18.00°C
[22:32:41][D][climate:421]:   Target Temperature: 21.50°C
[22:32:41][V][component:204]: Component panasonic_ac.climate took a long time for an operation (0.05 s).
[22:32:41][V][component:205]: Components should block for at most 20-30ms.
[22:32:46][V][panasonic_ac.cz_tacg1:248]: Polling AC
[22:32:46][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[22:32:46][V][panasonic_ac:247]: RX: 70.20.34.2B.80.A0.5A.00.00.00.00.00.30.2D.00.00.20.97.80.80.FF.12.06.FF.80.80.FF.80.18.00.01.00.00.00.D5 (35)
[22:32:46][D][climate:392]: 'Panasonic AC' - Sending state:
[22:32:46][D][climate:395]:   Mode: COOL
[22:32:46][D][climate:403]:   Custom Fan Mode: Automatic
[22:32:46][D][climate:409]:   Custom Preset: Normal
[22:32:46][D][climate:412]:   Swing Mode: OFF
[22:32:46][D][climate:415]:   Current Temperature: 18.00°C
[22:32:46][D][climate:421]:   Target Temperature: 21.50°C
[22:32:51][V][panasonic_ac.cz_tacg1:248]: Polling AC
[22:32:51][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[22:32:51][V][panasonic_ac:247]: RX: 70.20.34.2B.80.A0.5A.00.00.00.00.00.30.2B.00.00.20.97.80.80.FF.12.06.FF.80.80.FF.80.18.00.01.00.00.00.D7 (35)
[22:32:51][D][climate:392]: 'Panasonic AC' - Sending state:
[22:32:51][D][climate:395]:   Mode: COOL
[22:32:51][D][climate:403]:   Custom Fan Mode: Automatic
[22:32:51][D][climate:409]:   Custom Preset: Normal
[22:32:51][D][climate:412]:   Swing Mode: OFF
[22:32:51][D][climate:415]:   Current Temperature: 18.00°C
[22:32:51][D][climate:421]:   Target Temperature: 21.50°C
[22:32:51][V][component:204]: Component panasonic_ac.climate took a long time for an operation (0.05 s).
[22:32:51][V][component:205]: Components should block for at most 20-30ms.
[22:32:56][V][panasonic_ac.cz_tacg1:248]: Polling AC
[22:32:56][V][panasonic_ac:245]: TX: 70.0A.00.00.00.00.00.00.00.00.00.00.86 (13)
[22:32:56][V][panasonic_ac:247]: RX: 70.20.34.2B.80.A0.5A.00.00.00.00.00.30.2D.00.00.20.97.80.80.FF.12.06.FF.80.80.FF.80.18.00.01.00.00.00.D5 (35)
[22:32:56][D][climate:392]: 'Panasonic AC' - Sending state:
[22:32:56][D][climate:395]:   Mode: COOL
[22:32:56][D][climate:403]:   Custom Fan Mode: Automatic
[22:32:56][D][climate:409]:   Custom Preset: Normal
[22:32:56][D][climate:412]:   Swing Mode: OFF
[22:32:56][D][climate:415]:   Current Temperature: 18.00°C
[22:32:56][D][climate:421]:   Target Temperature: 21.50°C
[22:32:56][V][component:204]: Component panasonic_ac.climate took a long time for an operation (0.05 s).
[22:32:56][V][component:205]: Components should block for at most 20-30ms.

DomiStyle, 2023-04-25:
That's strange, byte 14 is actually used for eco mode on my unit and byte 5 is used for mild dry. Not sure how to combine those 2 different behaviors.

You can check out the current protocol documentation here: https://github.com/DomiStyle/esphome-panasonic-ac/tree/master/protocol/cztacg1

michas79de, 2023-04-25:
Here it seems to be free and shows 2D, which seems to be 'Mild Dry off' for me:
[image]
I would suggest switching both bytes, but I don't know what byte 5 will trigger at my A/C and 14 at other A/Cs.
(Switching on the Mild Dry Button in Home Assistant won't do anything for me, it seems. The switch switches immediately back to off and the log always show "80".)
I don't have any further suggestion or proposal at the moment.

michas79de, 2023-07-12:
@DomiStyle I want to try this other byte for Mild Dry myself. So I made a fork (https://github.com/michas79de/esphome-panasonic-ac).
I am not a programmer. So I am not sure what I am doing. But I hope I'm doing it right.

I think I found the values in [esppac_cnt.cpp](https://github.com/DomiStyle/esphome-panasonic-ac/commit/79ad2880b3364719fee435cb6bc19896ec361e38) and changed them.
But I can't find where to change the byte. Can you please guide me?

DomiStyle, 2023-07-12:
You can manipulate any value in the packet being sent by accessing `this->cmd[X]` where X is the index you want to access excluding the header and packet length. Any data that was received is inside of `this->data` instead. 

For example, mild dry (ID 4 in the table above) would be `this->cmd[2]`. If you want to change field 13 you can access it with `this->cmd[11]`.

As for reading, the values get passed to their own `determine_xyz(uint8_t value)` functions that evaluates the values.

For example:

```
bool PanasonicACCNT::determine_alternative_mild_dry(uint8_t value) {
  if (value == 0x2B)
    return true;
  else if (value == 0x2D)
    return false;
  else {
    ESP_LOGW(TAG, "Received unknown alternative mild dry value");
    return false;
  }
}
```

You will then want to call that function in `set_data`:

```
bool alternativeMildDry = determine_alternative_mild_dry(this->data[11]);
```